FastLED/FastLED #1201: garbage in the output because reading outside the array

board707, 2021-03-07:
Hi!
II suppose a bug in block controllers with more than 8 parallel channels (example **arm\k20** or  **arm\k66**)
The `Lines` structure in the controller is defined as:
`typedef union {
		uint8_t bytes[12];
		uint16_t shorts[6];
		uint32_t raw[3];
	} Lines;`
so the size of shorts[] array is six
But below in code, in `writeBits()` method we can see the cycle (for example line 135 of arm/k66/clockless_block_arm_k66.h)
`for(register uint32_t i = USED_LANES/2; i < 8; ++i) {
			.....
			if(USED_LANES>8) {
				*FastPin<FIRST_PIN>::cport() = ((~b2.shorts[i]) & PORT_MASK);
			} 
.....
		}`
<some lines skipped>
where the code reads from the shorts[] array up to index 7.

As the result I have parasitic illumination on the matrix.

As I think the issue can be easily fixed if change Lines definition to  
`typedef union {
		uint8_t bytes[16];
		uint16_t shorts[8];
		uint32_t raw[4];
	} Lines;`

kriegsman, 2021-03-22:
Thanks for the bug report; I think you're probably right, and I'll take a look at it.  Much appreciated.

board707, 2021-03-22:
I changed the Lines sctructure according to my message above and bug was gone

kriegsman, 2021-03-22:
Thank you; I’ll make the change!

closed 2026-08-31 by zackees