FastLED/FastLED #1203: GS8208

DavidAntonin, 2021-03-09:
Hello, what setting shoud I use with GS8208?
Timig is in this datasheet: http://addressable-led.com/UploadFile/upi/file/20171130/GS8208.pdf
Thank you

hijomojo, 2022-04-09:
Did you end up figuring out which chipset to specify?

I don't have my strips yet so can't test and am keen to find out if people had success with FastLED to GS8208.

From looking at what other people are saying about the GS8208, and results of their testing, it sounds like it's compatible with WS2811, WS2812, and/or SK6812 data signal/timing. I've read that connecting a WS2811 pixel's output pin to a GS8208 pixel's input pin does result in correct behaviour from the GS8208. Similarly, connecting the other way around reportedly worked fine, too. To me this says declaring your FastLED strip as WS2811 will work.

From my own reading of datasheets, it looks like GS8208's timing is most similar to WS2813, though I don't think FastLED has a setting specifically for these at the moment...

hijomojo, 2022-04-12:
Some further info: it seems what I found about WS2813's similarity might indeed be the answer. This reddit commenter mentions they declare their GS8208 strips as WS2813 and they "work fine". I guess there is a setting for WS2813 then, given the person commented 3 years ago.
[https://www.reddit.com/r/FastLED/comments/arqm0y/comment/egps9gm](https://www.reddit.com/r/FastLED/comments/arqm0y/comment/egps9gm)

marmilicious, 2022-04-12:
I think the GS8208 is a clone of the WS2815.

http://newstarleds.blogspot.com/2018/11/ws2815-compare-gs8208-12v-pixel-led-smd.html

hijomojo, 2023-10-09:
Just confirming, the GS8208 works with strips declared as WS2813. My strips worked perfectly with this setting and RGB colour order.