FastLED/FastLED #1206: Please remove FastLED.delay()

devarishi7, 2021-03-16:
Being active on the Arduino forum i have again come across people using FastLED.delay() which is causing them issues. FasdtLED.delay() calls show() repeatedly until the time has elapsed, which is **Pointless !!** The chips have their data and don't need updating (there is no new data) the processor could be doing something more useful or have at least have interrupts turned on. .show() turns off interrupts (even for an APA102 which is anyway not needed) which causes compatibility issues with other libraries. If this would only happen for a good cause, like actually sending new data to a LED-chip, it is needed, but just to have the program waiting is really, well i can't put this nicely so i will refrain from giving my expressed opinion. So please, either remove FastLED.delay() from the library altogether or modify it so it really is the same as delay() executing NOP's (probably easier so peoples sketches don't break with an update) As it is now we (me and others on the forum) spend time helping people that have used examples that come with this library and tried combining it with other libraries (IRremote, Servo.h even hwSerial !!) only to find that it doesn't work as they've hoped even though they have the proper hardware (with APA102's it should work just fine) simply because they've used FastLED.delay(). This is wasteful of our time.

kriegsman, 2021-03-16:
Hi @devarishi7, and thanks fo using FastLED and bringing this up.  Please allow me to explain why FastLED.delay() is there and the useful thing that it does for many people. 

As you probably know, FastLED has a built-in setBrightness() function that lets you programmatically scale the brightness of the whole project down dynamically, and without changing the values in the leds[] array.  It does this when show() is called by reading the source RGB values from the leds[] array, scaling them down numerically by 'brightness', and then transmitting the values to the led strip itself.  For example: if the leds[] array contains a pixel that is dark red, say (8, 0, 0), and the brightness is set to 50%, then the values transmitted to the led strip itself will be (4, 0, 0).

What may not be obvious is that FastLED has 'temporal dithering' built in to this process to help achieve finer-grained levels of brightness than the LED strip itself can in hardware.  Consider what happens if the led[] array contains (7, 0, 0), and the master brightness control is set to 50%.  Ideally, we'd transmit (3.5, 0, 0) to the led strip, but we can't send anything but an integer, so we have to round the value and either send (3, 0, 0) or (4, 0, 0) -- and neither is quite right.  This is where the temporal dithering comes in.  The first call to show() will transmit that pixel as (3, 0, 0), and the second call to show() will transmit (4, 0, 0).   By calling show() quickly and repeatedly, the led itself will be lit half of the time at (3, 0, 0) and half of the time at (4, 0, 0).  This results in an amortized brightness level of (3.5, 0, 0), and we've actually managed to drive the LED strip in such a way that we can do fractional brightness levels for each pixel.  

This works all the way down to very low brightness levels.  Here's a demo of the dithering in action, with the master brightness control turned down to 3 (out of 255).  Without the temporal dithering, the led animation looks rough and chunky.  With the dithering, it's as smooth as it would be even without the dimming.  https://www.youtube.com/watch?v=1-_JtRl2ks0. (This is from when we first released the dithering, about six years ago.)

So: in order for the master brightness control (with temporal dithering) to work, we need to rapidly refresh the LEDs, so that we can update them with _slightly different_ values each time.  Furthermore, the automatic power management (setMaxPowerInVoltsAndMilliamps) uses the master brightness control to keep current levels within the specified parameters, so that also benefits from the temporal dithering.

So the short version is that "FastLED.delay()" serves a very useful purpose by enabling smooth temporal dithering, which gives much smoother brightness control than the LED strips could do on their own.  BUT, and this is important, no one is required to use FastLED.delay().  If someone's project wants to use regular delay(), that works, too -- with, as you point out, less time spent with interrupts turned off. 

If you want to minimize time with interrupts turned off, use regular Arduino delay(), not FastLED.delay(). If you want to use the FastLED master brightness control with temporal dithering, use FastLED.delay().  

So, we're not going to take it out, but if people are confused about which to use, you're welcome to send them a link to this message, and hopefully it'll help explain why we went to all the work of adding FastLED.delay() in the first place, and the potential benefits and tradeoffs of using it.

closed 2021-03-16 by kriegsman

elliottkember, 2021-03-16:
@kriegsman Maybe the problem is that these two functions have the same name but do different things, which could be confusing to a new user. I remember when I didn't know the difference. 

Is it worth brainstorming an alternative name for `FastLED.delay()` - maybe `FastLED.wait()`? The original `.delay()` should still work, of course, but documentation could make reference to `FastLED.wait()`.

Or better yet, what about `dither()`, as in "to be indecisive" - dither for 100ms before deciding what to do next!

urish, 2021-03-16:
Very well explained. Thanks @kriegsman!

kriegsman, 2021-03-16:
I agree that a better name would be more clear. I’d want it to also explicitly explain that it was going to send the contents of the led[] array to the strip, too, so people didn’t write “FastLED.show(); FastLED.delay(20);”  The FastLED.show() is redundant. 

Maybe FastLED.show( milliseconds );

Of course, that function already takes a parameter- the temporary brightness.  But I bet no one is using that, and I might be willing to break the backward compatibility in this one case, since there’s such an easy work-around for the (very, very) few people it might affect.

urish, 2021-03-16:
personally, I'm more fan of `showAndWait(milliseconds)` or `showAndDelay(milliseconds)`. Bit more verbose, but it's also very clear what the number means.

kriegsman, 2021-03-16:
I do like the clarity of those.  `showAndDelay(ms)` is good, but potentially has the same ambiguity that led the original person to open this issue, namely that _this_ `delay` is unexpectedly different from the usual `delay()`.  

Doing some more brainstorming: `showForMillis(ms)`, `showForMilliseconds(ms)`, `showMs(ms)`, `showTime(ms)` (which must always be spoken in a Bob Fosse voice /j), `

I'm currently liking `showFor(milliseconds)` and `showAndHold(milliseconds)`

marmilicious, 2021-03-16:
I like showAndHold(milliseconds).  Seems nicely descriptive that is running show (which some didn't always realize) and also that it is going to "stop" after that.

devarishi7, 2021-03-17:
Right.. well i wasn't aware of the dithering, and i doubt if many people are, and if many people are using that part of the library. Very few users actually read the documentation. FastLED comes with a big assortment of 'ready-made' patterns and examples which makes it very inviting to newbies. Unfortunately it comes with some pitfalls. There is no error checking on writing beyond the size of the array, which is the first pitfall, but this is general c++ knowledge. FastLED.delay() is used in every example but nowhere in the comments of these examples is it explained that this is for dithering purposes, and that it effectively block ISR's from doing their work (If there are, which in the examples there aren't of course.) Now comes to mind that this also means that if i direct my program into another part where i for instance wait for user input, the dithering stops working, and so unless you actually focus your CPU on it, it anyway doesn't really work, but this is a different matter. Thank you for the clear explanation, now i am not really sure how to resolve this. Maybe dithering should be turned on or off explicitly using a #define or function ? The interrupts being on or off are always an issue when using addressable LED's when the signal is bit-banged and that newbies are not aware of this can hardly be helped, but i actually don't see why interrupts are turned off when the LED's make use of a clock pin ? (like the APA102 does) Now i am drifting of topic a little, but the issue remains the same. I can relate to it that once the function was created and a very neat solution at that, one would want to incorporate it in all examples, but we are talking about less than 0.5 % in brightness level, which really is not much, at the expense of hogging the processor. It would save a lot of time if this was made more clear. (Just as it would save a lot of time if there was a setPixelColor() function that does do an error check on writing within the bounds of the array) All i want to do is reflect what happens to people when they start using FastLED and things don't work the way they want and they come to the Arduino forum for help. Thank you for your time !

kriegsman, 2021-03-17:
Thanks for thinking about how new users first experience with FastLED; that's something that Dan and I talked about at great length together, and it's still a really good thing to think about.

As I said before, you can already turn dithering on and off.  One idea that might be worth looking at is that if dithering is not on, we could have FastLED.delay() call FastLED.show() exactly one time, and then just call Arduino delay().   I just opened https://github.com/FastLED/FastLED/issues/1207 to capture and track that idea.

Also you seemed to indicate that you're seeing interrupts being disabled when an APA102 (or a similar 'clocked' LED type) is used.  Following up on your comment,I just checked the code for that, and I'm not seeing that; I think that we _don't_ disable interrupts in our APA102 support.  I think that we only disable interrupts when we're writing data to a 'clockless' three-wire LED type, like Neopixels, that cannot tolerate having the timing arbitrarily interrupted.  If you are finding that interrupts _are_ being disabled when you're using an APA102 or other 'clocked' LEDs, please open a new Issue for that and we'll fix it immediately.  As you said, there's no point in disabling interrupts for clocked LEDs.

5chmidti, 2021-03-17:
@devarishi7 for color operations with bounds checking check out the [CRGBSet](https://github.com/FastLED/FastLED/wiki/RGBSet-Reference) (more complete api in docs or code). It doesn't really do bounds checking, but the member functions don't go out of bounds internally.