FastLED/FastLED #1208: Modernize the examples to help new users

kriegsman, 2021-03-17:
There are a few little modernizations we could make to the example sketches that would help new users get started faster, and develop good habits from the start.

1. Switch from `FastLED.setBrightness(...)` to `FastLED.setMaxPowerInVoltsAndMilliamps( ... )`
2. Switch `LEDS.[function]` to `FastLED.[function]`
3. Switch from `byte` to `uint8_t`
4. Use `uint16_t` for all LED index variables, so that the examples all work with > 255 LEDs
5. Use `FastLED.showAndHold( ... )` (when that exists) instead of `FastLED.show(); FastLED.delay( ...);`, with an explicit note that it will keep actively refreshing the LEDs to make temporal dithering / fractional brightnesses work.  (See discussion in https://github.com/FastLED/FastLED/issues/1206, in terms of educating new users about the effects of these choices.) 
6. Try to make all examples run at about the same visual speed regardless of microcontroller speed, e.g. use `FastLED.showAndHold(...)`, or have animations be a function of _time_, rather than a function of _number-of-iterations_ .
7. Switch from manifest constants (`#define`) to `static const` where possible and more legible.
8. Switch from a startup `delay(3000);` to perhaps a warmup function, like `FastLED.fadeIn( 3000 )`, which doesn't exist yet in the library as of 3/17/2021, but which I have code for.
9. Remove messages about minimum version of FastLED to use
10. Unify overall structure of each sketch so they're easier to read as a group

There are other larger pieces of rework we could do, e.g. make the demo animations like Fire2012 into their own classes, with proper member variables to control parameters, etc., but this Issue is not for tracking that work; this issue is just for tracking the small modernizations listed above.

yesyesuk, 2021-03-18:
>> 7. Switch from manifest constants (#define) to static const where possible and more legible.
Just for my understanding, why is that better? It is my understanding that "static const" would take up memory whereas a "#define" would not.

kriegsman, 2021-03-22:
This is a great question!  Twenty (or thirty!) ago, most C or C++ compilers would have actually created the variable for the `static const`, which _would_ take up RAM at runtime, as you suggested that it might.  However, when any modern compilers see a `static const` value, they literally forward the value of the constant into the surrounding code, as if you had just typed the constant value inline yourself.  

You can check your compiler yourself by taking some code, switching it back and forth between  a `#define` and a `static const` for the compile-time constants, and checking the size of the executable code (or by literally comparing the disassembly of the executable code).  If the code is exactly the same, then you have a modern compiler that does 'constant propagation' aka 'constant forwarding'.  If the generated code is different, then you probably have a very old compiler on your hands.  More information on constant propagation: https://en.wikipedia.org/wiki/Constant_folding 

Using a `static const` also offers type safety, which makes it a cleaner, safer choice than a `#define`, all other things being equal.

Part of why this is a great question is because the answer _used to be_ that a `#define` used less memory at runtime, because no variable was created, but _today_ virtually all compilers do excellent constant propagation, and they don't bother creating the wasteful, unused variable.  So if you have a programming book from more than ten years ago, it probably said "use #define", and it was right at the time!  But contemporary compilers are much smarter and more aggressive about code optimization than they were back in the day, and the 'right answer' to this question has changed!   

Thanks for asking this question - it's a good one.

ssilverman, 2021-04-02:
Would `constexpr` be acceptable in the examples instead of `const`? I figure that C++ 11 or even 14 is an okay target? I’ve been coding most of my own projects to at least C++ 14 and sometimes 17. What is your preference?