FastLED/FastLED #1210: Unable to compile 3.004.000 for LOLIN D32 (ESP32)

jon-bondy, 2021-03-19:
When I try to compile for the LOLIN D32, I get many errors like:

D:\Arduino\libraries\FastLED\src/fastpin.h:36:11: error: 'RoReg' does not name a type
  volatile RoReg *mInPort;
           ^
D:\Arduino\libraries\FastLED\src/fastpin.h: In member function 'void Pin::_init()':
D:\Arduino\libraries\FastLED\src/fastpin.h:43:3: error: 'mInPort' was not declared in this scope
   mInPort = (volatile RoReg*)portInputRegister(digitalPinToPort(mPin));
   ^
D:\Arduino\libraries\FastLED\src/fastpin.h:43:23: error: ISO C++ forbids declaration of 'type name' with no type [-fpermissive]
   mInPort = (volatile RoReg*)portInputRegister(digitalPinToPort(mPin));
                       ^

samguyer, 2021-03-22:
Hello @jon-bondy! I'd like to look into this problem. Can you tell me how to reproduce the error? What build environment are you using? I tried compiling for LOLIN D32 in the Arduino IDE and did not see any errors.
Thanks!

jon-bondy, 2021-03-25:
Sam:

Thank you!  I have attached my source code.  Well.  I tried to.  I dragged and dropped it.  No sign of the attachment, however.

I am running Arduino IDE 1.8.13.  I have a LOT of FastLED libraries copied in all over the place, including 3.1.6, 3.3.3, and 3.4.0.  When I go into Tools, Manage Libraries and type FastLED into the filter box, I see 3.4.0 is installed.  It seems that the IDE is finding the 3.1.6 also, although I do not understand how.  

I have no problem compiling FastLED code for an Arduino Nano, so the library situation seems benign in that configuration.

The  #include for FastLED is around line 121.  If I change it from just "FastLED.h" to specify a path to 3.3.3, I get different errors (see below)

When I compile, I get the following error messages.

In file included from D:\Arduino\White_LED_Strip_Plus_IR_ESP32\White_LED_Strip_Plus_IR_ESP32.ino:121:0:
D:\Arduino\libraries\FastLED\src/FastLED.h:14:13: error: '#pragma' is not allowed here
 #    pragma message "FastLED version 3.004.000"
             ^
In file included from D:\Arduino\libraries\FastLED\src/FastLED.h:48:0,
                 from D:\Arduino\White_LED_Strip_Plus_IR_ESP32\White_LED_Strip_Plus_IR_ESP32.ino:121:
D:\Arduino\libraries\FastLED\src/fastpin.h:36:11: error: 'RoReg' does not name a type
  volatile RoReg *mInPort;
           ^
D:\Arduino\libraries\FastLED\src/fastpin.h: In member function 'void Pin::_init()':
D:\Arduino\libraries\FastLED\src/fastpin.h:43:3: error: 'mInPort' was not declared in this scope
   mInPort = (volatile RoReg*)portInputRegister(digitalPinToPort(mPin));
   ^
D:\Arduino\libraries\FastLED\src/fastpin.h:43:23: error: ISO C++ forbids declaration of 'type name' with no type [-fpermissive]
   mInPort = (volatile RoReg*)portInputRegister(digitalPinToPort(mPin));
                       ^
D:\Arduino\libraries\FastLED\src/fastpin.h:43:14: error: expected primary-expression before 'volatile'
   mInPort = (volatile RoReg*)portInputRegister(digitalPinToPort(mPin));
              ^
D:\Arduino\libraries\FastLED\src/fastpin.h:43:14: error: expected ')' before 'volatile'
D:\Arduino\libraries\FastLED\src/fastpin.h: In member function 'void Pin::toggle()':
D:\Arduino\libraries\FastLED\src/fastpin.h:59:58: error: 'mInPort' was not declared in this scope
  inline void toggle() __attribute__ ((always_inline)) { *mInPort = mPinMask; }
                                                          ^
In file included from D:\Arduino\libraries\FastLED\src/FastLED.h:65:0,
                 from D:\Arduino\White_LED_Strip_Plus_IR_ESP32\White_LED_Strip_Plus_IR_ESP32.ino:121:
D:\Arduino\libraries\FastLED\src/fastspi.h: At global scope:
D:\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
Multiple libraries were found for "IRremote.h"
 Used: D:\Arduino\libraries\IRremote
 Not used: D:\Arduino\libraries\Arduino-IRremote-master
Multiple libraries were found for "FastLED.h"
 Used: D:\Arduino\libraries\FastLED
 Not used: D:\Arduino\libraries\FastLED-3.1.6
exit status 1
Error compiling for board LOLIN D32.

Here is how it behaves if I specify the path to 3.3.3:

In file included from D:\Arduino\White_LED_Strip_Plus_IR_ESP32\White_LED_Strip_Plus_IR_ESP32.ino:122:0:
sketch\src\FastLED-3.3.3\FastLED.h:14:13: error: '#pragma' is not allowed here
 #    pragma message "FastLED version 3.003.003"
             ^
In file included from sketch\src\FastLED-3.3.3\FastLED.h:48:0,
                 from D:\Arduino\White_LED_Strip_Plus_IR_ESP32\White_LED_Strip_Plus_IR_ESP32.ino:122:
sketch\src\FastLED-3.3.3\fastpin.h:36:11: error: 'RoReg' does not name a type
  volatile RoReg *mInPort;
           ^
sketch\src\FastLED-3.3.3\fastpin.h: In member function 'void Pin::_init()':
sketch\src\FastLED-3.3.3\fastpin.h:43:3: error: 'mInPort' was not declared in this scope
   mInPort = (volatile RoReg*)portInputRegister(digitalPinToPort(mPin));
   ^
sketch\src\FastLED-3.3.3\fastpin.h:43:23: error: ISO C++ forbids declaration of 'type name' with no type [-fpermissive]
   mInPort = (volatile RoReg*)portInputRegister(digitalPinToPort(mPin));
                       ^
sketch\src\FastLED-3.3.3\fastpin.h:43:14: error: expected primary-expression before 'volatile'
   mInPort = (volatile RoReg*)portInputRegister(digitalPinToPort(mPin));
              ^
sketch\src\FastLED-3.3.3\fastpin.h:43:14: error: expected ')' before 'volatile'
sketch\src\FastLED-3.3.3\fastpin.h: In member function 'void Pin::toggle()':
sketch\src\FastLED-3.3.3\fastpin.h:58:58: error: 'mInPort' was not declared in this scope
  inline void toggle() __attribute__ ((always_inline)) { *mInPort = mPinMask; }
                                                          ^
In file included from sketch\src\FastLED-3.3.3\FastLED.h:65:0,
                 from D:\Arduino\White_LED_Strip_Plus_IR_ESP32\White_LED_Strip_Plus_IR_ESP32.ino:122:
sketch\src\FastLED-3.3.3\fastspi.h: At global scope:
sketch\src\FastLED-3.3.3\fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
Multiple libraries were found for "IRremote.h"
 Used: D:\Arduino\libraries\IRremote
 Not used: D:\Arduino\libraries\Arduino-IRremote-master
exit status 1
Error compiling for board LOLIN D32.
[White_LED_Strip_Plus_IR_ESP32.zip](https://github.com/FastLED/FastLED/files/6201132/White_LED_Strip_Plus_IR_ESP32.zip)

samguyer, 2021-03-26:
@jon-bondy This is going to make you sad: you are just missing a semicolon on line 13, after the declaration:

char rcvdCh = '!'

The problem is that the subsequent code the compiler sees is actually the include of FastLED.h, and that one missing semicolon throws the whole thing off. It's a terrible error message.

Anyway, that should get you unstuck!

jon-bondy, 2021-03-26:
Sam:

Wow.  Just wow.  Thanks so much.  So frustrating that single missing 
character wreaked so much havoc.

Thanks, again!

Jon

GMDionicio, 2021-03-30:
hello I have the same problem with the esp32 Wrover could you help me.

jon-bondy, 2021-03-30:
As you can see, above, the problem was entirely in my source code, not in the FastLED library.  You might try to create a very simple test case and see if the compiles, and/or inspect your code one more time.

samguyer, 2021-03-30:
@GMDionicio That is not an error message. You can ignore it.