FastLED/FastLED #1211: Fastest way to change color of one LED ring among many rings

williamesp2015, 2021-03-22:
I cam across changing color of one LED ring (8-12 Leds) among many led rings (10-20) but I have no clue how to do that in simple and fastest way using Arduino. Any help would be appreciated.

marmilicious, 2021-03-22:
@williamesp2015  You could do something like this:  

fill_solid(leds+5, 10, CRGB::Blue);  // Fill 10 pixels with blue, starting at leds[5]

or  

fill_solid( &(leds[39]), 4, CRGB::Red )  //fill from 39 to 42 (ie. four pixels)  

Be careful not to write past the end of the CRGB array.  

We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

Please provide more specific info about your hardware and what you are wanting to do if you ask there.  If you want to share code please put it on pastebin.com or gist.github.com and Share a link.  Thank you.

samguyer, 2021-03-23:
@williamesp2015 and @marmilicious I'd like to close this issue, since it is more of a question about how to do something. Let me know if that's ok. We can also continue the discussion here, but I think you'll get a lot more interesting responses on the FastLED subreddit, as Marc noted.