FastLED/FastLED #1212: WS2811 0 bit high duration is too long, causes bit errors and random LED flashing

egjohnson17, 2021-03-24:
I am using FastLED with WS2811 LED strips and I sometimes get random LED colors when I write all 0s or clear. This it seems is because the high duration of the 0 bit is about 440ns as shown in the attached image while the period is 1500s or about 660kHz when the frequency should be 800 kHz and high duration for a 0 bit only 250ns according to the datasheet link below. At 440ns that is between the thresholds for a 0 and a 1 and very close to the 1 high time of 600ns +/- 150ns. 

I am trying to find a way to manual change the pulse widths to compensate for this. 
In the source code I tried to change the values in this and related lines but without any change noticed

case WS2811_PORTA: return addLeds(new InlineBlockClocklessController<NUM_LANES, PORTA_FIRST_PIN, NS(320), NS(320), NS(640), RGB_ORDER>(), data, nLedsOrOffset, nLedsIfOffset);

Any help would be greatly appreciated. Thanks!

https://cdn-shop.adafruit.com/datasheets/WS2811.pdf
[image]

egjohnson17, 2021-03-24:
I was able to compensate for this error by changing the following line in the file to include a '-9' to the T1 value
clockless_arm_d21.h because I am using the Arduino MKR wifi 1010 

return showLedData<8,4,(T1-9),T2,T3,RGB_ORDER, WAIT_TIME>(portBase, FastPin<DATA_PIN>::mask(), pixels.mData, pixels.mLen, &data);

I would like to understand if others have had issues with FASTLed in this regard?

You can see the corrected low bit below and the frequency is now 800 kHz. Note that the 1 bit high duration is also corrected by this to about
[image]
[image]
[image]

 600ns.

Ap1965, 2021-03-24:
Hello, i have the same Issue. Use NodeMCU and only 8 LEDS on the stripe. But if i change the value like your comment, it happens many more times. Sofry for my english, not very good.

egjohnson17, 2021-03-24:
Interesting, why do you think that changing the values for you made it worse? Did you probe the waveform with an oscilloscope?

Sojourneer, 2025-01-03:
I also have a problem with very long 0-bits, which I see as long as 15uS.
Context: I am monitoring the WS2812B stream to switch it between every n pixels using a hardware bus switch.  My code is interpreting these very long 0-bits as RESET events, which is messing up the parse. [doing the parse using a 16MHz ATMega MCU is rather tricky]