FastLED/FastLED #1213: How to make 12v  or 24v ledstrip wifi/ble and capacitive dim/switch

JacobsElements, 2021-03-25:
Hello everybody, I'm surching a way to make floorlamps able to connect BLE and WIFI. Also it has to be possible to turn the light on/off and dim with a capacitive touch switch.  I just need the code for arduino nano 33iot.
 If think this is the one to do this. I'm a rookie at this, but I should have this as quick as possible. I use 12v or 24v dimmable single coloured or ww ledstrips. Thank you

marmilicious, 2021-03-25:
@JacobsElements  We use github "issues" here mostly just for bugs or improvements in the library itself. For general discussions, help, and how-to it might be better to visit an Arduino or electronics group.  If you have FastLED related questions please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED  If you want to share code please put it on pastebin.com or gist.github.com and Share a link. Thank you.

JacobsElements, 2021-03-25:
@marmilicious  Thank you for responding my question. I'm sorry being at the wrong place, but it's still a bit walking in the dark for me :). Hopefully I see the light soon :)

closed 2021-03-25 by JacobsElements