FastLED/FastLED #1219: How to control special ws2811 led? The sequence time is different from that of standard ws2811.

feipinxiang, 2021-04-06:
Speed time 
|T0H|0 code,high voltage time|0.3 µs|±150ns|
|T0L|0 code,low voltage time|0.9 µs|±150ns|
|T1H|1 code,high voltage time|0.6 µs|±150ns|
|T1L|1 code,low voltage time|0.6 µs|±150ns|
|RES|low voltage time|80µs||

24bit data
G7……G0 R7……R0 B7……B0

[image]
[image]

feipinxiang, 2021-04-07:
Do I need to connect 12v- and arduino gnd?

feipinxiang, 2021-04-07:
It's ridiculous, the seller gave me the wrong datasheet.
The correct ic is SM16703PB, not ws2811.I can’t see the chip because it’s covered with soft plastic.

T0H	0.3us  ±100ns
T1H	0.9us  ±100ns
T0L	0.9us  -100ns
T1L	0.3us  -100ns
reset	200us

marmilicious, 2021-04-07:
> Do I need to connect 12v- and arduino gnd?

Yes, a common ground connection between the LED strip and your controller is **always** required.

feipinxiang, 2021-04-07:
> > Do I need to connect 12v- and arduino gnd?
> 
> Yes, a common ground connection between the LED strip and your controller is **always** required.

Yes，After some testing I used ws2811 to drive correctly. ic is SM16703PB，The seller says it is compatible with ws2811.

closed 2021-04-07 by feipinxiang