FastLED/FastLED #1220: ESP32 Arduino using v3.4.0 crashing due to rmt, v3.3.3 does not but has glitchy led outputs

joe-fang-iD, 2021-04-06:
Hi,

I'm using an ESP32-WROOM-32D using ESP32-Arduino (running version 1.0.5-rc6), and am trying to control 3 separate strands of approximately 100 leds altogether. The LEDs are powered off 3.3V.

I've been using v3.3.3 FastLED at the start, and realized that with enough leds and perhaps complex enough pattern (and perhaps fast enough frame rate), the strands seem to glitch once every few seconds. I was thinking one of the reasons perhaps is the voltage is too low. Without trying any hardware changes, I decided to try to use FastLED v3.4.0, and I was happily surprised to find that the leds were displaying the patterns smoothly and glitch free at 30-60fps. Unfortunately, it is buggy and crashes.

here are the typical exceptions:

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
												Core 1 register dump:
												PC      : 0x4010a0ac  PS      : 0x00060034  A0      : 0x80081328  A1      : 0x3ffbeb00  
												A2      : 0x3ffc49c0  A3      : 0x00000000  A4      : 0x00000020  A5      : 0x80000020  
												A6      : 0x00000003  A7      : 0x00060723  A8      : 0x800812d2  A9      : 0x00000020  
												A10     : 0x00000000  A11     : 0x00000000  A12     : 0x0000000c  A13     : 0x00188018  
												A14     : 0x0024800c  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
												EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
												Core 1 was running in ISR context:
												EPC1    : 0x40090ef5  EPC2    : 0x00000000  EPC3    : 0x40082da2  EPC4    : 0x4010a0ac

												ELF file SHA256: 0000000000000000

												Backtrace: 0x4010a0ac:0x3ffbeb00 0x40081325:0x3ffbeb20 0x400813c2:0x3ffbeb40 0x4008142e:0x3ffbeb70 0x40083b1d:0x3ffbeb90 0x40082d9f:0x3ffba300 0x40086397:0x3ffba320 0x400911aa:0x3ffba340

some different exceptions but all related to rmt accessing cache during ISR (could be invoking something at the wrong time):

0x401094e3: rmt_set_intr_enable_mask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c line 334
						0x40109569: rmt_set_tx_intr_en at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c line 370
						0x400812fe: ESP32RMTController::startOnChannel(int) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 244
						0x40081325: ESP32RMTController::startNext(int) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 204
						0x400813c2: ESP32RMTController::doneOnChannel(rmt_channel_t, void*) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 302
						0x4008142e: ESP32RMTController::interruptHandler(void*) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 333
						0x40082da8: spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c line 82
						0x40086397: ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c line 62
						0x400911aa: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

0x4010ef09: gpio_set_direction at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/gpio.c line 245
								0x4010a19e: rmt_set_pin at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c line 402
								0x400812cf: ESP32RMTController::startOnChannel(int) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 222
								0x40081325: ESP32RMTController::startNext(int) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 204
								0x400813c2: ESP32RMTController::doneOnChannel(rmt_channel_t, void*) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 302
								0x4008142e: ESP32RMTController::interruptHandler(void*) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 333
								0x40082d99: spi_flash_op_block_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_utils.c line 81
								0x40086397: ipc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c line 62
								0x400911aa: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

									0x4010a09c: rmt_set_pin at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt.c line 394
									0x40081325: ESP32RMTController::startNext(int) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 204
									0x400813c2: ESP32RMTController::doneOnChannel(rmt_channel_t, void*) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 302
									0x4008142e: ESP32RMTController::interruptHandler(void*) at C:\Users\joe.fang\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp line 333
									0x40096fba: esp_rom_spiflash_read_data at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 223
									0x400977d1: esp_rom_spiflash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c line 588
									0x4008248c: spi_flash_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c line 599
									0x400f0f0a: esp_partition_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c line 242
									0x40112536: spiffs_api_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs_api.c line 36
									0x40112759: spiffs_phys_rd at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_cache.c line 161
									0x40113676: spiffs_obj_lu_find_id_and_span_v at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c line 511
									0x40113c62: spiffs_obj_lu_find_entry_visitor at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c line 177
									0x4011406d: spiffs_obj_lu_find_id_and_span at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c line 537
									0x40115c72: spiffs_object_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c line 1996
									0x40111480: spiffs_hydro_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c line 414
									0x40111af9: SPIFFS_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c line 426
									0x401100c8: vfs_spiffs_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spiffs/esp_spiffs.c line 484
									0x401a011f: esp_vfs_read at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c line 446
									0x400e18a5: _fread_r at ../../../.././newlib/libc/stdio/fread.c line 262
									0x400e1915: fread at ../../../.././newlib/libc/stdio/fread.c line 301
									0x400ddb97: VFSFileImpl::read(unsigned char*, unsigned int) at C:\Users\joe.fang\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS\src\vfs_api.cpp line 339
[... cut ...]
I'm aware that the exception says I'm running 1.0.4 ESP32 Arduino, but I am actually running 1.0.5-rc6.
Same code running on FastLED v3.3.3 does not crash.
The crashes are infrequent but random; it happens anytime between 2 minutes and 1 hour.
I've tried a few things, such as:

- #define FASTLED_ALLOW_INTERRUPTS 1 and 0, 
- turning off dithering, 
- increasing stack sizes for my tasks, 
- trying other chipsets (I'm using SK6805's, which is essentially SK6812. But WS2811 also seem to work similarly)
- using delay instead of FastLED.delay

Any advice is appreciated!

TrigpointingUK, 2021-04-07:
> I was thinking one of the reasons perhaps is the voltage is too low

Yeah, that'd be my first guess too!  ;-)

The LEDs you're using are rated 3.5V -> 5.5V, so you're out of spec, though only just, so generally good behaviour but glitches when you push them hard sounds about right.  

The ESP you're using is rated 3.0V -> 3.6V.  You don't mention your PSU's rating, or the wiring you're using, but have you tried measuring the actual voltage you're getting at the ESP when all the lights are on?  Especially if you're running both ESP and LEDs from the same supply, and especially if you have long and/or thin wires between PSU and ESP, then you only need to drop 0.15V per wire and you're into glitch territory.  My expectation would have been crashes correlated to random WiFi activity, rather than random RMT / flash activity, but hey ho!  You mentioned you tried swapping out the SK6805's for WS2811's... the latter draw more current; did that make things worse?  

As for why you see the glitches only with v3.3.3, but crashes only with v3.4.0 I have no idea, but I'd suggest the first thing to try is to stick 5V across your LED strip and see if all your issues disappear.

joe-fang-iD, 2021-04-08:
Thanks for the voltage suggestion. It appears that I was able to solve the glitch issues on v3.3.3 by moving `FastLED.show()` into the same task as the one that sets the most complex CRGB arrays. It appears that guarding access to the arrays and `FastLED.show()` via the same semaphore isn't good enough for the FastLED + ESP32 combo. I have another strand that is very short that is configured in another task still, and it still works glitch free.

I have also tested using chipset strand settings SK6812, WS2811, and WS2812B; it seems like the behaviour is the same.

What is so strange is that before I moved the `.show()` function, v3.4.0 already had it glitch free, but somehow it crashes as per above. 

I'll be testing to see if I still get crashes in v3.4.0 now with the `.show()` function in the same thread, and also test the effects of running 5V on my pre-fix code.

sblantipodi, 2021-04-19:
Same error here but it still crashing 
Esp32 is a mess with Fastled while running constant WiFi transmission.

Errors are so random that is pretty impossible to debug.

What's the problem with Fastled, esp32 and WiFi?

sblantipodi, 2021-04-19:
@samguyer can you tell me what lib we should use for ESP32?
should we use this
https://github.com/FastLED/FastLED
or this
https://github.com/samguyer/FastLED???

Thank you!!!

samguyer, 2021-04-19:
@sblantipodi Use the main FastLED repo -- my repo is just for experimental changes. At this point, all the good stuff is merged in with the main repo.

The issue is that the WiFi services require frequent, high-priority interrupts that disrupt the timing of the LED signal. We've been looking at a bunch of different options to try to mitigate the problem. One thing you can try is using the I2S driver instead of the RMT driver (the default).

We've also got some changes coming that should help, including an implementation hand-coded in assembly to make it very fast.

sblantipodi, 2021-04-19:
@samguyer thanks for the answer, may I ask you how to switch from RMT to I2S?
I can't find any docs.

samguyer, 2021-04-19:
@sblantipodi Oh right, the docs are in the source files (not the most user-friendly, I know). Just add the following line *before* you the \#include of FastLED.h:

`#define FASTLED_ESP32_I2S true`

joe-fang-iD, 2021-04-19:
@sblantipodi your issue may also be related to most internal wifi/ble stuff being on core 0. I encounted a lot of glitchiness with fastled while ble is running, and fixed it but pinning all non-connectivity issues to core 1. I think there are earlier issue posts that detail this potential solution.

sblantipodi, 2021-04-20:
> @sblantipodi Oh right, the docs are in the source files (not the most user-friendly, I know). Just add the following line _before_ the #include of FastLED.h:
> 
> `#define FASTLED_ESP32_I2S true`

ok tried it, nothing changed, CORE0 and CORE1 is still crashing randomly...

CORE0/1 panic'ed is one of the recurring random error...
I see that this is a problem that happen since many years, I hope you'll find a solution to this one day :)

thank you for your good work here and for your continued support Samguyer.

sblantipodi, 2021-04-20:
> @sblantipodi your issue may also be related to most internal wifi/ble stuff being on core 0. I encounted a lot of glitchiness with fastled while ble is running, and fixed it by pinning all non-connectivity activities to core 1. I think there are earlier issue posts that detail this potential solution.

I tried pinning all the WiFi stuff to core 0 and the Fastled.show() to core 1, this helps a bit, there are less errors, but it still crash.
BLE is disabled in my code.

sblantipodi, 2021-04-20:
this is the excepton with exception decoder:
`
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40091f01  PS      : 0x00060033  A0      : 0x400939d7  A1      : 0x3ffde840  
A2      : 0x00050023  A3      : 0x00000001  A4      : 0x3ffc4c0c  A5      : 0x3ffc4c0c  
A6      : 0x3ff000dc  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000002  
A10     : 0x00000000  A11     : 0x00000081  A12     : 0x3ffc4c28  A13     : 0x203ffbbf  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000c9  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x40091f01:0x3ffde840 0x400939d4:0x3ffde860 0x4009398a:0x7dffffd6
C:\Users\sblantipodi\.platformio\packages\toolchain-xtensa32\bin\xtensa-esp32-elf-addr2line.exe: 'C:\dev\dpsoftware\ArduinoRaspbery\PUBLIC\glow_worm_luciferin\.pio\build\glowwormluciferinfull_esp32\firmware.elf': No such file
Esp32ExceptionDecoder: failed to call C:\Users\sblantipodi\.platformio\packages\toolchain-xtensa32\bin\xtensa-esp32-elf-addr2line.exe: Command '['C:\\Users\\sblantipodi\\.platformio\\packages\\toolchain-xtensa32\\bin\\xtensa-esp32-elf-addr2line.exe', '-fipC', '-e', 'C:\\dev\\dpsoftware\\ArduinoRaspbery\\PUBLIC\\glow_worm_luciferin\\.pio\\build\\glowwormluciferinfull_esp32\\firmware.elf', '0x40091f01:0x3ffde840']' returned non-zero exit status 1.
`

samguyer, 2021-04-20:
Hmmm. It's hard to tell what's going on -- the exception decoder doesn't
seem to be working on your executable.

We have not had reports of crashes with ESP32. The timing issues usually
cause occasional flashiness when the WiFi kicks in. Can you run it in a
configuration where the WiFi is not on?

sblantipodi, 2021-04-20:
> Hmmm. It's hard to tell what's going on -- the exception decoder doesn't seem to be working on your executable. We have not had reports of crashes with ESP32. The timing issues usually cause occasional flashiness when the WiFi kicks in. Can you run it in a configuration where the WiFi is not on?

not many people opens issues but if you search the web it's full of people asking for this problem :)
https://www.google.com/search?q=esp32+fastled+core+panic%27ed&oq=esp32+fastled+core+panic%27ed&aqs=chrome..69i57j33i160l2.6663j0j7&sourceid=chrome&ie=UTF-8

if I disable wifi all problems vanish

samguyer, 2021-04-20:
I can take a look later this week, but a lot of these cases turn out to be
bugs in the user's code. Three most common is writing past the end of an
array.

sblantipodi, 2021-04-21:
> I can take a look later this week, but a lot of these cases turn out to be bugs in the user's code. The most common is writing past the end of an array.
> […](#)

Thank you for your time Samguyer, If this can help, I'm sending colors informations via WiFi, this is what causes ESP to crash randomly...

sblantipodi, 2021-04-21:
just to add some bits, same identical code works now problem for hours if I just comment the 
>  FastLED.show();
call

samguyer, 2021-04-21:
Ok, interesting. Can you share the code with me? You don't need to make it
public. If it's complex to set up, maybe a stripped down version.

sblantipodi, 2021-04-21:
thank you very much for answering the issue.

project is open source, here you can find the complete source code:
https://github.com/sblantipodi/glow_worm_luciferin

here you can find the stripped version:
https://gist.github.com/sblantipodi/2a70f68a9c953626a752468cf2144fd6

various delays and the feedTheDog() function helps is avoiding core panic for WDT.

as you can see I have two tasks, one that manage the wifi stuff (tcpTask), and one that manage the (serialTask)
Fastled.show();

The serialTask have a boolean that says if there is new data from the MQTT callback, if there is, it call

samguyer, 2021-04-21:
Does the stripped-down version also crash? I'll keep looking at it. I won't
have a chance to run it myself for a couple of days, though.

One thing: you can avoid the issues with the watchdog timer if you use task
notification instead of busy waiting for the synchronization.

sblantipodi, 2021-04-21:
Stripped version does not compile unfortunantly, it's just to show you the significant part of the code, 
it's difficult to strip all the code because there is the wifi code and the mqtt callback you should call from a client 30 times per second.

thinking on how to give you something "runnable"...

samguyer, 2021-04-21:
No worries.  I'll try the full version first.

samguyer, 2021-04-29:
@sblantipodi Hello! I'm planning to work on this problem today. The most useful thing you could give me is a stack trace of the crash (using the exception decoder). If that doesn't work, I'll try to install and set up your code myself.

samguyer, 2021-04-29:
@joe-fang-iD I think I just figured out what's going on with your code -- are you using SPIFFs or one of the other libraries for accessing flash memory on the ESP32?

joe-fang-iD, 2021-04-29:
That's right, the pattern to be shown is stored in spiffs and read once every frame rate.

samguyer, 2021-04-29:
@joe-fang-iD So, there's a peculiarity of the ESP32 that access to the flash memory locks the regular memory bus. As a result, ESP-IDF enforces a fairly draconian policy: when flash access is going on, all code on both cores is disabled *except* interrupt handlers residing in IRAM. Here's where it gets interesting. All of our code is in IRAM (which was a huge pain, btw), BUT the RMT library code is not. So, what you're seeing is a bus violation: our interrupt handler is calling an RMT function that is not in IRAM while a flash access operation is going on.

OK, that's probably way more detail than you wanted! ;-) There are two ways to address this problem. One is to add your own synchronization to make sure that SPIFFs access is not going on while FastLED.show() is running. That's a pain. The other option is to add the following \#define directive *before* you include FastLED.h:

`#define FASTLED_ESP32_FLASH_LOCK 1`

What this will do is lock out the SPIFFs access until the FastLED.show() is done. It's not a guarantee, but it should help.

In the meantime, I'm trying to figure out a longer-term solution.

joe-fang-iD, 2021-04-29:
@samguyer sounds promising, will definitely try it out and let you know. Do you know why the behaviour between 3.3.3 and 3.4.0 is different? Is there a difference between the two versions with respect to using rmt?

sblantipodi, 2021-04-29:
> @sblantipodi Hello! I'm planning to work on this problem today. The most useful thing you could give me is a stack trace of the crash (using the exception decoder). If that doesn't work, I'll try to install and set up your code myself.

I'll give you it on the next crash. 

Your help here is incredible, I really appreciate how you manage this project, thanks.

samguyer, 2021-04-29:
@sblantipodi That's a good question -- it doesn't seem like there should be a big difference between 3.3.3 and 3.4.0

joe-fang-iD, 2021-04-29:
@samguyer Upon quick inspection, and please let me know if this is not the case, it appears that 3.3.3 does not have this ``FASTLED_ESP32_FLASH_LOCK`` definition or ``spi_flash_op_lock();``. I'm guessing the likely scenario is that this will get 3.4.0 to be crash-free but run at a similar performance as 3.3.3, unless there are other ways performance is increased. 

I've managed to get 3.3.3 to work glitch-free by putting the ``.show()`` function in the same task as my ``spiffs read`` functions, which was kind of a shot in the dark that worked, but now it KIND of makes sense, and it also makes sense now that other tasks that also control leds do not glitch or crash despite them not running in the same task, and it's because they don't use spiffs.

I'll do some tests, but I guess I'm still worried about this happening because I don't really understand the difference between how esp32 is reading program flash during runtime and spiffs read. Is there a chance that ESP32 will crash even if it's not using SPIFFS by trying to read any part of the program flash?

samguyer, 2021-04-29:
@joe-fang-iD I'll look more carefully at the 3.3.3 vs 3.4.0 situation. That's interesting. Add the spi_flash_op_lock will guarantee that no flash operations can happen while you're doing show (and vice versa). But it will throttle the performance a little bit. Putting the SPIFFs operations and the show() call on the same core basically gets rid of the race condition.

sblantipodi, 2021-04-30:
I think that ESP32 is simply not mature for pushing it hard, better hardware than ESP8266 with way worse ecosystem around.

those are the recurring and completely random errors I have.

```
***ERROR*** A stack overflow in task DLE0 has been detected.
abort() was called at PC 0x4008fe70 on core 0
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (DLE0) 

ELF file SHA256: 0000000000000000

Backtrace: 0x4000c2e4:0x3ffca6f0 0x4008fc1d:0x3ffca700 0x4008fd9d:0x3ffca7d0 0x40090085:0x3ffca7f0 0x4008d2b2:0x3ffca810 0x4008fbd9:0x3ffca8d0 0x4008fe59:0x3ffca8f0 0x4008fe70:0x3ffca910 0x40091a63:0x3ffca930 0x40093654:0x3ffca950 0x4009360a:0x00000000
  #0  0x4000c2e4:0x3ffca6f0 in ?? ??:0
  #1  0x4008fc1d:0x3ffca700 in commonErrorHandler_dump at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x4008fd9d:0x3ffca7d0 in commonErrorHandler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #3  0x40090085:0x3ffca7f0 in xt_unhandled_exception at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #4  0x4008d2b2:0x3ffca810 in _xt_user_exc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:757
  #5  0x4008fbd9:0x3ffca8d0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #6  0x4008fe59:0x3ffca8f0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #7  0x4008fe70:0x3ffca910 in vApplicationStackOverflowHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #8  0x40091a63:0x3ffca930 in vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #9  0x40093654:0x3ffca950 in _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:406
  #10 0x4009360a:0x00000000 in _frxt_int_exit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:206

```

```
abort() was called at PC 0x40086509 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4008fbdc:0x3ffd0f60 0x4008fe59:0x3ffd0f80 0x40086509:0x3ffd0fa0 0x40086635:0x3ffd0fd0 0x400f3953:0x3ffd0ff0 0x400efd51:0x3ffd12b0 0x400efcec:0x3ffd1300 0x40091ea1:0x3ffd1330 0x40084df7:0x3ffd1360 0x4008f2c1:0x3ffd1380 0x4013e765:0x3ffd13b0 0x401728fd:0x3ffd13e0 0x401724d9:0x3ffd1400 0x4016cbf5:0x3ffd1420 0x4016cef5:0x3ffd1440 0x4016de66:0x3ffd1470 0x4016dea9:0x3ffd14b0 0x4016ded6:0x3ffd14e0 0x4016a421:0x3ffd1510 0x4016a4cf:0x3ffd1540 0x401676d1:0x3ffd1570 0x4016db47:0x3ffd1590 0x4017242a:0x3ffd15c0 0x4015efd3:0x3ffd15e0 0x40090e6a:0x3ffd1610
  #0  0x4008fbdc:0x3ffd0f60 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x4008fe59:0x3ffd0f80 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x40086509:0x3ffd0fa0 in lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:143
  #3  0x40086635:0x3ffd0fd0 in _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:171
  #4  0x400f3953:0x3ffd0ff0 in _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)
  #5  0x400efd51:0x3ffd12b0 in fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:50
  #6  0x400efcec:0x3ffd1300 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)
  #7  0x40091ea1:0x3ffd1330 in vPortCPUAcquireMutexIntsDisabledInternal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
      (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
      (inlined by) vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201
  #8  0x40084df7:0x3ffd1360 in wifi_int_disable_wrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_adapter.c:194
  #9  0x4008f2c1:0x3ffd1380 in pp_post at ??:?
  #10 0x4013e765:0x3ffd13b0 in esp_wifi_internal_tx at ??:?
  #11 0x401728fd:0x3ffd13e0 in low_level_output at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/port/esp32/netif/wlanif.c:122
  #12 0x401724d9:0x3ffd1400 in ethernet_output at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:309
[... cut ...]
Backtrace: 0x40091b81:0x3ffdf0c0 0x40093654:0x3ffdf0e0 0x4009360a:0x4000bff0
  #0  0x40091b81:0x3ffdf0c0 in vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #1  0x40093654:0x3ffdf0e0 in _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:406
  #2  0x4009360a:0x4000bff0 in _frxt_int_exit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:206
```

sblantipodi, 2021-05-05:
Just another random exceptions that happen on esp32 when WiFi is used intensively.

```
assertion "recv_tcp: recv for wrong pcb!" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40091eb9  PS      : 0x00060033  A0      : 0x4009398f  A1      : 0x3ffee7b0  
A2      : 0x00050023  A3      : 0x00000001  A4      : 0x3ffc5ebc  A5      : 0x3ffbd8ec  
A6      : 0x00000001  A7      : 0x3ffba57e  A8      : 0x00000000  A9      : 0x00000002  
A10     : 0x00000000  A11     : 0xbc3ffbd8  A12     : 0x3ffc5ec4  A13     : 0x3ffcc1ac  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbc3ffc20  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x40091eb9:0x3ffee7b0 0x4009398c:0x3ffee7d0 0x40093942:0xf61eadd8
  #0  0x40091eb9:0x3ffee7b0 in vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #1  0x4009398c:0x3ffee7d0 in _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:406
  #2  0x40093942:0xf61eadd8 in _frxt_int_exit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:206

```

chordophony, 2021-05-09:
I love it when I discover a bug that has some active chatter!  I too am having issues with ESP23/Wifi/Fastled, using this project https://github.com/alf45tar/PedalinoMini

I have managed a workaround for my use case:  replaced all instances of FastLED.show() with "if (!WiFi.isConnected()) FastLED.show();"....when I want to have normal operations with LED on, then I simply disable Wifi.  The pedalino project has a great start-up monitor that allows you to select operating mode--for normal use I only need Bluetooth active, and I only need to activate Wifi when I'm configuring the device.

Thanks for this great library and support, FASTLED Team!  And I'm looking forward to a real fix in the lib so I can remove the hack I put in place.  Any help I can provide (logs, etc.), please let me know!

tomioe, 2022-02-17:
Excellent thread! This is all gold... Unfortunately adding `#define FASTLED_ESP32_FLASH_LOCK 1` didn't solve the issue at hand. Still getting either random crashes or just straight up freezes. 

FastLED on Core0 and WiFi on Core1. Keep getting these crashes:

```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40081275  PS      : 0x00050034  A0      : 0x4008786c  A1      : 0x3ffbfc30
A2      : 0x00000001  A3      : 0x01000000  A4      : 0x00000018  A5      : 0x4008c89c
A6      : 0x3ff000dc  A7      : 0x00000001  A8      : 0x00000008  A9      : 0x00000003
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00060023  A13     : 0x3ffce02c
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x40085cc8  EPC2    : 0x00000000  EPC3    : 0x40085cc8  EPC4    : 0x40081275

ELF file SHA256: 0000000000000000

Backtrace: 0x40081275:0x3ffbfc30 0x40087869:0x3ffbfc50 0x40085cc5:0x3ffcde30 0x40089ffd:0x3ffcde50 0x40081361:0x3ffcde90 0x400d3940:0x3ffcdeb0 0x400d350f:0x3ffcded0 0x400d13a7:0x3ffcdf20 0x4008166d:0x3ffcdf60 0x4008a1b6:0x3ffcdf90
  #0  0x40081275:0x3ffbfc30 in ESP32RMTController::interruptHandler(void*) at .pio\libdeps\atomlight-d1\FastLED\src\platforms\esp\32/clockless_rmt_esp32.cpp:548 (discriminator 4)
  #1  0x40087869:0x3ffbfc50 in _xt_medint3 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1323
[snip]
```

Running on the Wemos D1 Mini (ESP32-WROOM) as well. Disabling WiFi (and in turn SPIFFS) resolves the issue, as others have noted as well.