FastLED/FastLED #1221: Support for the RP2040 (Pi Pico) with ArduinoCore

joachimhs, 2021-04-11:
Hello, 

Will this library support the Pi Pico (ArduinoCore for the RP2040 have landed a few days ago in the MBED repo - https://github.com/arduino/ArduinoCore-mbed).

dtgent, 2021-04-18:
+1

Got myself excited with the Arduino IDE support for RP2040 so dove in and did a test for FastLED 

 #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."
 
 Too early I know but with the very low cost, number of pins and speed of the Pico, I'm really hoping to use the Arduino IDE with Picos and FastLED is underpinning a lot of what I'm tinkering with.

evenstephen85, 2021-04-21:
My assumption is that the team will be working hard to optimize fastLED for the use of the PIO, but I was curious if we'll be able to get just a quick port to get us up and running?

ageppert, 2021-07-19:
A work around in the short-term is to use the Adafruit NeoPixel library which has been updated to support the rp2040 and pico with a PIO implementation: https://github.com/Bodmer/Adafruit_NeoPixel

nmaas87, 2022-05-10:
https://github.com/FastLED/FastLED/pull/1261

zackees, 2026-09-06:
Closing — RP2040 support shipped and is covered by CI and the bench.

The original ask was ArduinoCore-mbed. **Both** Pico cores are detected today, in `src/platforms/arm/rp/is_rp2040.h:15`:

```c
#if defined(ARDUINO_ARCH_RP2040) || defined(TARGET_RP2040) || \
    defined(PICO_32BIT) || defined(ARDUINO_RASPBERRY_PI_PICO)
#define FL_IS_RP2040
#endif
```

`TARGET_RP2040` / `ARDUINO_RASPBERRY_PI_PICO` are the ArduinoCore-mbed macros; `ARDUINO_ARCH_RP2040` is Earle Philhower's arduino-pico, which is the core FastLED builds against in CI.

Current state:

- **Support landed** in #1261 (as @nmaas87 noted), and has since grown well past the original request.
- **Drivers**: `src/platforms/arm/rp/` carries PIO clockless output (including parallel multi-lane), hardware SPI, and UART paths, shared across RP2040 and RP2350 via `rpcommon/`. PIO resources are acquired through the SDK claims system so FastLED coexists with other libraries that also use PIO.
- **CI environments**: `rp2040`, `rp2350`, `rp2350w`.
- **Build proof** (this week, local bench):
  - `bash compile rp2040 --examples Blink` → flash 309.11 KB / 2.00 MB, RAM 57.04 KB / 256.00 KB
  - `bash compile rp2350 --examples Blink` → flash 563.20 KB / 2.00 MB, RAM 56.70 KB / 512.00 KB
- **Deployment** goes through fbuild's managed ROM BOOTSEL/UF2 flow.

@robertlipe — yes, your 2025 question was right; this just never got closed.

Remaining RP work is tracked in its own scoped issues (RP2350(W) HIL in #3899, peer/network bring-up in #3832), so nothing is lost by closing this one. Please reopen if the mbed core specifically misbehaves for you.

closed 2026-09-06 by zackees