FastLED/FastLED #1223: please help atmega128

fadi0albaghdadi, 2021-04-21:
I am using a mega core library atmega128 but it doesn't work with fastled

fadi0albaghdadi, 2021-04-21:
In file included from C:\Users\fadi-\AppData\Local\Temp\arduino_modified_sketch_328439\sketch_apr20a.ino:1:0:

C:\Users\fadi-\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000

 #    pragma message "FastLED version 3.004.000"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\fadi-\Documents\Arduino\libraries\FastLED\src/FastLED.h:65:0,

                 from C:\Users\fadi-\AppData\Local\Temp\arduino_modified_sketch_328439\sketch_apr20a.ino:1:

C:\Users\fadi-\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\fadi-\Documents\Arduino\libraries\FastLED\src/FastLED.h:48:0,

                 from C:\Users\fadi-\AppData\Local\Temp\arduino_modified_sketch_328439\sketch_apr20a.ino:1:

C:\Users\fadi-\Documents\Arduino\libraries\FastLED\src/fastpin.h: In instantiation of 'class FastPin<17>':

C:\Users\fadi-\Documents\Arduino\libraries\FastLED\src/platforms/avr/clockless_trinket.h:100:49:   required from 'class ClocklessController<17, 4, 10, 6, (EOrder)66, 0, false, 10>'

C:\Users\fadi-\Documents\Arduino\libraries\FastLED\src/chipsets.h:474:7:   required from 'class WS2812Controller800Khz<17, (EOrder)66>'

C:\Users\fadi-\Documents\Arduino\libraries\FastLED\src/FastLED.h:103:52:   required from 'class WS2812<17, (EOrder)66>'

C:\Users\fadi-\Documents\Arduino\libraries\FastLED\src/FastLED.h:302:39:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = WS2812; unsigned char DATA_PIN = 17; EOrder RGB_ORDER = (EOrder)66]'

C:\Users\fadi-\AppData\Local\Temp\arduino_modified_sketch_328439\sketch_apr20a.ino:6:55:   required from here

C:\Users\fadi-\Documents\Arduino\libraries\FastLED\src/fastpin.h:210:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^~~~~~~~~~~~~

Multiple libraries were found for "FastLED.h"
 Used: C:\Users\fadi-\Documents\Arduino\libraries\FastLED
Using library FastLED at version 3.4.0 in folder: C:\Users\fadi-\Documents\Arduino\libraries\FastLED 
exit status 1
Error compiling for board ATmega128.

samguyer, 2021-04-29:
@fadi0albaghdadi There are two different issues here. First, the two "pragma" messages that you see are just informational -- they are not errors, and they don't stop the program from compiling.

The second issue is that you chose a pin number that is not supported. We use template instantiation to automatically stop people from using pins that are not valid for the given microncontroller.

fadi0albaghdadi, 2021-04-29:
> @fadi0albaghdadi There are two different issues here. First, the two "pragma" messages that you see are just informational -- they are not errors, and they don't stop the program from compiling. [...]

fadi0albaghdadi, 2021-04-29:
thank you for replying
How do I solve the problem I have

closed 2021-04-29 by fadi0albaghdadi

eag77, 2021-05-06:
Добавте в файл: ...\Documents\Arduino\libraries\FastLED\src\platforms\avr\fastpin_avr.h

#elif  defined(__AVR_ATmega128__)
_FL_DEFPIN( 0, 0, E); _FL_DEFPIN( 1, 1, E); _FL_DEFPIN( 2, 2, E); _FL_DEFPIN( 3, 3, E);
_FL_DEFPIN( 4, 4, E); _FL_DEFPIN( 5, 5, E); _FL_DEFPIN( 6, 6, E); _FL_DEFPIN( 7, 7, E);
_FL_DEFPIN( 8, 0, B); _FL_DEFPIN( 9, 1, B); _FL_DEFPIN(10, 2, B); _FL_DEFPIN(11, 3, B);
_FL_DEFPIN(12, 4, B); _FL_DEFPIN(13, 5, B); _FL_DEFPIN(14, 6, B); _FL_DEFPIN(15, 7, B);
_FL_DEFPIN(16, 3, G); _FL_DEFPIN(17, 4, G); _FL_DEFPIN(18, 0, D); _FL_DEFPIN(19, 1, D);
_FL_DEFPIN(20, 2, D); _FL_DEFPIN(21, 3, D); _FL_DEFPIN(22, 4, D); _FL_DEFPIN(23, 5, D);
_FL_DEFPIN(24, 6, D); _FL_DEFPIN(25, 7, D); _FL_DEFPIN(26, 0, G); _FL_DEFPIN(27, 1, G);
_FL_DEFPIN(28, 0, C); _FL_DEFPIN(29, 1, C); _FL_DEFPIN(30, 2, C); _FL_DEFPIN(31, 3, C);
_FL_DEFPIN(32, 4, C); _FL_DEFPIN(33, 5, C); _FL_DEFPIN(34, 6, C); _FL_DEFPIN(35, 7, C);
_FL_DEFPIN(36, 2, G); _FL_DEFPIN(37, 7, A); _FL_DEFPIN(38, 6, A); _FL_DEFPIN(39, 5, A);
_FL_DEFPIN(40, 4, A); _FL_DEFPIN(41, 3, A); _FL_DEFPIN(42, 2, A); _FL_DEFPIN(43, 1, A);
_FL_DEFPIN(44, 0, A); _FL_DEFPIN(45, 0, F); _FL_DEFPIN(46, 1, F); _FL_DEFPIN(47, 2, F);
_FL_DEFPIN(48, 3, F); _FL_DEFPIN(49, 4, F); _FL_DEFPIN(50, 5, F); _FL_DEFPIN(51, 6, F);
_FL_DEFPIN(52, 7, F);

#define SPI_DATA 10
#define SPI_CLOCK 9
#define SPI_SELECT 8

#define AVR_HARDWARE_SPI 1
#define HAS_HARDWARE_PIN_SUPPORT 1