FastLED/FastLED #1227: How to control brightness individually per LED?

thewh1teagle, 2021-04-25:
#include <FastLED.h>
#define LED_PIN     7
#define NUM_LEDS    2
CRGB leds[NUM_LEDS];

void setup() {
  FastLED.addLeds<WS2812, LED_PIN, GRB>(leds, NUM_LEDS);
  leds[0].setRGB(255, 0, 0);
  // leds[0].setBrightness(??? half_brightness);
  leds[1].setRGB(255, 0, 0);
  // leds[1].setBrightness(??? max_brightness);
  FastLED.show();
}

marmilicious, 2021-04-25:
Hello @thewh1teagle.  We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

Brightness can be adjusted by either using lower RGB numbers, or it's really easy if you use HSV you can use a lower V value.
leds[0] = CRGB(128,0,0);  // half brightness red
leds[1] = CHSV(0,255,128);  // half brightness red

See here for more info on using HSV:  
https://github.com/FastLED/FastLED/wiki/FastLED-HSV-Colors

setBrightness is a global brightness adjustment for the whole strip.

thewh1teagle, 2021-04-25:
@marmilicious 
There is no such a brightness function per led like there is for the global brightness?
I don't understand what CRGB or CHSV is and it has 3 parameters but it confusing me because as I understand I need 4 parameters to control the brightness too - (rgb0, rgb1, rgb2, brightness)
Thank you.

yesyesuk, 2021-04-25:
You just need to work with the RGB values for each LED to "set the brightness".
```
leds[0].setRGB(255, 0, 0); // LED 0 full brightness red
leds[1].setRGB(127, 0, 0); // LED 1 half brightness red
leds[2].setRGB(63, 0, 0); // LED 2 quarter brightness red
```

thewh1teagle, 2021-04-25:
```
leds[0].setRGB(255, 255, 255); // LED 0 full brightness white
leds[0].subtractFromRGB(155);
```
So this one will set the white color to be half brightness?

closed 2021-04-25 by thewh1teagle

marmilicious, 2021-04-26:
Glad that worked. 

Instead of: 
```leds[0].setRGB(255, 255, 255); // LED 0 full brightness white

leds[0].subtractFromRGB(128);```

You can assign the exact RGB values you want to begin with.

```leds[0].setRGB(128,128,128); // LED 0 half brightness white```

But I find it much easier to use HSV if that works for your project.  I highly recommend checking it out, and the rest of [the wiki here](https://github.com/FastLED/FastLED/wiki), and our [wiki on Reddit](https://www.reddit.com/r/FastLED/wiki/index) for links to even more examples and tutorials.

tonton81, 2021-04-26:
For reference I do it a bit differently in my project, here for reference:
```
CHSV _targetHSV = rgb2hsv_approximate(CRGB(red, green, blue)); // choose RGB color
_targetHSV.v = value;  // set the brightness 0 -> 100
leds[0] = _targetHSV;
```

vincentj87, 2023-06-28:
> For reference I do it a bit differently in my project, here for reference:
> 
> ```
> CHSV _targetHSV = rgb2hsv_approximate(CRGB(red, green, blue)); // choose RGB color
> //*** OR ***
> CHSV _targetHSV = rgb2hsv_approximate(leds[0]); // to use the current color of leds[0]
> _targetHSV.v = value;  // set the brightness 0 -> 255 , the limit is 255
> leds[0] = _targetHSV; // update leds[0] with the HSV result
> FastLED.show(); // show result
> ```