FastLED/FastLED #1228: Wiki: Add link to https://github.com/ben-xo/FastLED/tree/avr-interrupts?

SteffenPL, 2021-04-26:
Thanks for the nice package.

I got stuck at the infamous Interrupt on AVR problem and found this fork:
https://github.com/ben-xo/FastLED/tree/avr-interrupts
which worked very well for my application.

Would it be possible to add that into the wiki? Maybe on the bottom?
I understand that a fork is not as well-maintained and stable, so, it should not be the recommended solution...

ben-xo, 2021-07-25:
Hello! Thanks for mentioning me!

I have actually been working on that branch recently (as well as some other things which are related).

FastLED's AVR driver has gone out of its way to conform to the timings in the WS2811/WS2812 specification - but based on https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/ , you may find that the timings can be much more relaxed than most people think. This is the basis on which my branch allows interrupts: going out of spec on the timings. *this is not guaranteed to work on all WS2811/WS2812's* - but it works just fine on mine. It may also fail on long strips - again, I have only experimented on what I have to hand, which is strips up to 100 long.

You will still need to make sure your interrupts are very fast - comfortably less than 6000ns, which is about 96 cycles on a 16MHz Arduino. Not a lot! - but I have examples of using this for LEDPWM and audio sampling. The clock interrupt is also fast enough in the arduino core too.

If you're using timers for interrupts you may need to set them to start with a staggered offset from each other so they don't all interrupt at the same time, with two ordinarily-fast-enough ISRs adding up to more than 6000ns.

I will rebase that branch against master soon

ben-xo, 2021-07-25:
@samguyer is there a better forum for requests regarding the wiki/doc?

samguyer, 2021-07-26:
@ben-xo Thanks for your work on this interrupts issue! I really need to look at the Wiki situation -- I'm not sure who can edit it and what the process is.

ben-xo, 2021-07-26:
Apologies I think I meant to tag @kriegsman.

ben-xo, 2021-08-01:
Hi all, 

I have raised an official PR for the interrupts work which is cleaner and would be better referenced.

In the mean time feel free to use the branch https://github.com/ben-xo/FastLED/tree/release/ben-xo-integration-branch for to test all my PRs at once, including the cleaner AVR interrupts branch. The original branch `avr-interrupts` is no longer the best one to go to.