FastLED/FastLED #1231: FastLED.ESP32 crashes in case one controller has more than 276 LEDS

Jueff, 2021-05-06:
FastLED on ESP32 works fine with 8 parallel controllers and ~2900 LEDs.

This configuration works fine
```
 CLEDController& controller0 = FastLED.addLeds<NEOPIXEL, 27>(leds+  0, 43); 
 CLEDController& controller1 = FastLED.addLeds<NEOPIXEL, 32>(leds+ 43,  3); 
 CLEDController& controller2 = FastLED.addLeds<NEOPIXEL, 16>(leds+ 46, 276); 
 CLEDController& controller3 = FastLED.addLeds<NEOPIXEL, 14>(leds+557, 276);
 CLEDController& controller4 = FastLED.addLeds<NEOPIXEL, 18>(leds+1068, 276)); 
 CLEDController& controller5 = FastLED.addLeds<NEOPIXEL, 19>(leds+1579, 276)); 
 CLEDController& controller6 = FastLED.addLeds<NEOPIXEL, 23>(leds+2090, 276)); 
 CLEDController& controller7 = FastLED.addLeds<NEOPIXEL,  0>(leds+2601, 276)); 

```

But if e.g. controller0 has more than 276 LEDs assigned it crashes.

```
CLEDController& controller0 = FastLED.addLeds<NEOPIXEL, 27>(leds+  0, 277); 
```

or  controller2 has more than 280 Leds  (funny, some more than on controller 0)

```
CLEDController& controller0 = FastLED.addLeds<NEOPIXEL, 27>(leds+  0, 43);
CLEDController& controller1 = FastLED.addLeds<NEOPIXEL, 32>(leds+ 43,  3);
CLEDController& controller2 = FastLED.addLeds<NEOPIXEL, 16>(leds+ 46,281);
```

Problem appears immediatly after program startup - FastLED.show()

samguyer, 2021-06-22:
@Jueff Thanks for the bug report. I will take a look at it as soon as I can. I can't imagine why the number 276 would cause a problem!

Jueff, 2021-08-12:
Here is a complete sample code which reproduced the crash, including three 'WORKAROUND' defines.
If you set one of these WORKAROUNDS to true the crash doesn't occur anymore

WORKAROUND_1: reduce number of leds of controller7 from 312 to 256
WORKAROUND_2: don't run clearLeds on controllers
WORKAROUND_3: the most funny one - don't call Serial.available in main loop

Hope that helps to find the problem.

```
#define FASTLED_INTERNAL
#include <FastLED.h>

#define WORKAROUND_1 false
#define WORKAROUND_2 false
#define WORKAROUND_3 false

const int c0Len = 43;
const int c1Len = 3;
const int c2Len = 308;
const int c3Len = 256;
const int c4Len = 256;
const int c5Len = 256;
const int c6Len = 256;
const int c7Len = 312;

const int totalLen = c0Len+c1Len+c2Len+c3Len+c4Len+c5Len+c6Len+c7Len;
CRGB leds[totalLen];           // Define the array of leds

void setup() {
  // put your setup code here, to run once:
  CLEDController& controller0 = FastLED.addLeds<NEOPIXEL, 27>(leds, c0Len); 
  CLEDController& controller1 = FastLED.addLeds<NEOPIXEL, 32>(leds+c0Len,  c1Len); 
  CLEDController& controller2 = FastLED.addLeds<NEOPIXEL, 16>(leds+c0Len+c1Len, c2Len); 
  CLEDController& controller3 = FastLED.addLeds<NEOPIXEL, 14>(leds+c0Len+c1Len+c2Len, c3Len); 
  CLEDController& controller4 = FastLED.addLeds<NEOPIXEL, 18>(leds+c0Len+c1Len+c2Len+c3Len, c4Len); 
  CLEDController& controller5 = FastLED.addLeds<NEOPIXEL, 19>(leds+c0Len+c1Len+c2Len+c3Len+c4Len, c5Len); 
  CLEDController& controller6 = FastLED.addLeds<NEOPIXEL, 23>(leds+c0Len+c1Len+c2Len+c3Len+c4Len+c5Len, c6Len); 
  CLEDController& controller7 = FastLED.addLeds<NEOPIXEL,  0>(leds+c0Len+c1Len+c2Len+c3Len+c4Len+c5Len+c6Len, WORKAROUND_1 ? 256 : c7Len); 

  if (WORKAROUND_2==false) {
    // Delete all leds that are connected to a controller in terms of hardware but are not addressed in terms of software.
    #define LEDS_TO_CLEAR 256
    controller0.clearLeds(LEDS_TO_CLEAR);  
    controller1.clearLeds(LEDS_TO_CLEAR);  
    controller2.clearLeds(LEDS_TO_CLEAR);  
    controller3.clearLeds(LEDS_TO_CLEAR);  
    controller4.clearLeds(LEDS_TO_CLEAR);  
    controller5.clearLeds(LEDS_TO_CLEAR);  
    controller6.clearLeds(LEDS_TO_CLEAR);  
    controller7.clearLeds(LEDS_TO_CLEAR);  
  }
  Serial.begin(115200);
}
int cnt=0;
void loop() {
  FastLED.show();

  if (WORKAROUND_3==false) Serial.available();

  if (cnt++>100) {
    Serial.println();
    cnt=0;
  }
  Serial.print(".");
}
```

Jueff, 2021-08-13:
The message output on serial port

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
assertion "res == coreID || res == portMUX_FREE_VAL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.inc.h", line 105, function: vPortCPUAcquireMutexIntsDisabledInternal
abort() was called at PC 0x400d9faf on core 1

Backtrace: 0x4008b880:0x3ffb1e90 0x4008baad:0x3ffb1eb0 0x400d9faf:0x3ffb1ed0 0x4008926a:0x3ffb1f00 0x40088479:0x3ffb1f30 0x400d2d6d:0x3ffb1f50 0x400d2061:0x3ffb1f70 0x400d1c16:0x3ffb1f90 0x400d2721:0x3ffb1fb0 0x40088535:0x3ffb1fd0

Rebooting...

```

Jueff, 2021-09-20:
another investigation result: calling Serial.begin before **controller.clearLeds(..)** call also solves the problem.

samguyer, 2021-09-20:
@Jueff Ah, that is very useful information. OK, I think I know what's going on. Something is happening during FastLED.show() that causes it to miss the timing requirements, and in my current version **it prints a message to the Serial output**. So, if you haven't set up Serial yet, it probably crashes. Let me take a look and see what I can do.