FastLED/FastLED #1233: I think that we should admit that ESP32 is broken, another flickering thread.

sblantipodi, 2021-05-07:
As you can see from the number of issues ESP32 is a broken platform, long live ESP8266.

Using WiFi while using Fastled causes LED flickering, I tried everything, 
LEDs flicker if leds infos is is sent via serial cable and some data is transferred using wifi in the meanwhile.

This does not happen if WiFi is off.

samguyer, 2021-06-22:
@sblantipodi I have some ideas that might help, but it is a very hard problem. The WiFi is very demanding on the ESP32 (not sure why it doesn't happen with the ESP8266, which is much less powerful)

sblantipodi, 2021-06-22:
@samguyer I love your work here Sam, say something and I'll do it 😁

samguyer, 2021-07-26:
@sblantipodi Have you tried the latest version (Git head)? I also have a pull request from someone who rewrote the core of the ESP32 driver in hand-coded assembly. Do you want to give that version a try?

sblantipodi, 2021-07-27:
Hi @samguyer, thanks for your work here, flickering improved by a little margin but it still there. :(
it looks like ESP32 have a broken WiFi implementation.

https://github.com/espressif/arduino-esp32/issues/5297

samguyer, 2023-08-30:
I would love to make another attempt to improve this situation. I don't think the problem is totally fixable -- with a limited amount of compute power, it might not be possible to service heavy WiFi use AND get the timing right for LED output.

closed 2024-02-13 by sblantipodi