FastLED/FastLED #1237: FastLed not working with ESP8266 Boards 3.0.0

Guenni75, 2021-05-19:
Since I update the ESP-8266 Board to 3.0.0 i can't compile FastLed with my AZ-Delivery D1 Mini NodeMcu with ESP8266-12F.
Every FastLed-Line gives the error:
..........libraries\FastLED\chipsets.h:###:##: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]

ESP-8266 Board 2.7.0 works fine.

Why??

sblantipodi, 2021-05-20:
I can compile my code with lots of warning but core 3.0 destroyed FastLED, it flickers like a mess, pretty unusable.
#1236

stoer, 2021-05-21:
I can confirm this issue. No Warnings for me, only Code not working anylonger.

5chmidti, 2021-05-21:
Compiling the blink example with this platformio config: 
```
[env:d1]
platform = https://github.com/platformio/platform-espressif8266.git
framework = arduino
board = d1
```
works fine. Well, except for the warnings about the register keyword, wich is something that should be fixed (#1195) 'sometime'. But no actual error while compiling. What errors are you guys getting? (I don't have an 8266, I only compiled for it with pio).

sblantipodi, 2021-05-21:
I have no compilaton errors, only dozens of warnings but the result is a nearly no working led strip.

5chmidti, 2021-05-21:
Hm, then I can't help (don't have an 8266), let's hope someone else has one and can help.

SonnyJim, 2021-05-28:
I had been getting mightily confused as I've not be able to get working even a simple blink on my ESP8266 after setting up a new Arduino dev environment (ESP8266 v3 + FastLED v3.4.0).

Rolling back ESP8266 to v2.7.4 seems to have fixed it.  There's definitely something broken in v3.0 :(

stoer, 2021-05-28:
> Hm, then I can't help (don't have an 8266), let's hope someone else has one and can help.

with the blink example i get no warnings but also no working led strip. Any way I can help to debug the Problem?

sblantipodi, 2021-06-11:
may I ask if there are someone working on this please? 
will fastled support the new ESP8266 core 3.0.0?

kriegsman, 2021-06-11:
We’ll definitely take a look at this; we definitely want to support the new ESP environment!

topjor, 2021-06-16:
When trying core platform `3.0.0`, `FastLED.show()` flickers the first led on my strip (16 leds total). first it was green (full brightness even when set to a 64) after a few sketch flashes it changed to pink (no clue if the color is some indication for something) It also does not look to mater what I write to the strip, random colors, a flame pattern, or just no new colors. At the start it did not flicker at a constant rate, but after a few seconds it changed to a constant flashing (I am also calling the show function at a quick constant rate)
If I am correct the rate of the flickering is not as fast as calling the show function if called in really fast succession, but definitely related as disabling it stops the flickering. manually calling it with a button also makes it flicker every time.

I hope this can be helpful. If you want some help testing some things, just let me know.

hardware:
IDE: platformIO (vscode)
FastLED version: 3.4.0 and 3.3.3
Board: d1_mini
Led strip: SK6812 (16 leds) configured as NEOPIXEL

samguyer, 2021-06-16:
@topjor Thank you for that report -- this kind of information is often very useful. I'm hoping to get some time to work on FastLED soon, so I'll let you know how it goes.

sascha432, 2021-06-20:
Replacing showRGBInternal in ClocklessController (src\platforms\esp\8266\clockless_esp8266.h) fixes the issue for me, but I was not able to find the actual issue in the showRGBInternal method.

Boiling the method down to the code below still shows the issue.
All LEDs (116) are green at a low level, but the first one flickers. Mostly in a brighter green.

```c++
static uint32_t IRAM_ATTR showRGBInternal(PixelController<RGB_ORDER> pixels) {
	os_intr_lock();
	uint32_t start = __clock_cycles();
	uint32_t last_mark = __clock_cycles();
	uint16_t num = pixels.size();
	while(num--) {
		writeBits<8+XTRA0>(last_mark, 5);
		writeBits<8+XTRA0>(last_mark, 0);
		writeBits<8+XTRA0>(last_mark, 0);
	};
	os_intr_unlock();
	return __clock_cycles() - start;
}
```

sascha432, 2021-07-01:
Hi,

I found a solution for my issue. When checking the LEDs with my scope, I can see that the glitching occurs when show() is called too frequently. There is a gap of 49µs between each frame, but that does not seem to be enough. Playing around with an additional delay, it gets better with 17µs and disappears with 20µs. So they need at least a 70µs gap.

I am using following test code to display 3 pixels in red, green and blue on a strip with 90 pixels. The position does not matter too much unless it is near the start or end.

```
void setup()
{
    FastLED.addLeds<WS2812, NEOPIXEL_OUTPUT_PIN, GRB>(pixelData, NEOPIXEL_NUM_PIXELS);
    FastLED.setMaxPowerInVoltsAndMilliamps(5, powerLimit / 5);
    FastLED.setBrightness(10);
    FastLED.setDither(BINARY_DITHER);
    FastLED.clear();
}

void loop() {
    pixelData[5] = 0xff0000;
    pixelData[10] = 0x0000ff;
    pixelData[15] = 0x00ff00;
    FastLED.show();
    FastLED.delay(5);
    // delayMicroseconds(100);  <- this fixes the issue
}

```
Looking at the code, I can see a delayMicros(WAIT_TIME) there, but that is only called for retries inside show().
To be sure I would add a delay of at least 100µs between each call for WS2812...

Inside FastLLED.delay(), show() is called without delay()

The glitchting starts when show() is interrupted otherwise the LED pattern just repeats itself at the end of the string. my test strip is actually 1m with 144 LEDs. It also depends on other factors like number of LEDs, different batches of WS2812, how busy the system is, the milliseconds of the FastLED.delay() call etc...

sascha432, 2021-07-01:
After fixing the glitching issue with show() uncommenting mWait.wait(); / mWait.mark();, I ran into the first pixel flickering issue again.

The WAIT_TIME is still 50, but it turns out that running it with some code around adds there extra few microseconds needed for the LEDs to detect the gap.

The pixel is flickering red while being turned off. Here are some scope shots of it

[Single shot of how it looks without flickering](https://ibb.co/0Z2zMgd)
[A frame with a red pixel, this one is great since it has 50% brightness](https://ibb.co/9Vb4SsF)

I added another signal toggling a pin high/low before and after writeBits and you can see the difference in the faint signal marked.

I measured the time of the single pixels (incl. debug code, 24bit + delay till next one) and it is 36.12µs, the flickering one is 37.28µs. So something seems to interrupt the second writeBits (GRB, so basically red) call. Interrupts are locked with os_intr_lock() / xt_rsil(3). I think I saw it flickering in different colors, but today I only can reproduce red.

If it does not flicker, I can still see a gap but the LED output signal is low in this case.
[First pixel with a gap while low](https://ibb.co/Ttn49Cn)

[This is normal mode with extra signals](https://ibb.co/5T6SZPW)

Fixing this seems to be hard since I ran out of clock cycles, it is pretty busy there.

sblantipodi, 2021-07-04:
is still now work for me, upgraded to arduino core 3.0.1 and the first led goes crazy.
lot of flickering now...

vonSchlau, 2021-07-05:
I noticed that the first LED always flickers in one of the LED Base Colors (Red, Green, Blue) - when I change something in the code and compile again - it can be another base color - but never  a mixed color like grey or white or yellow - tis could be a hint...

RVV058, 2021-07-06:
I have same issues and it's getting quite frustrating.. It did work about a year ago but I lost my code and started over. 
After several days and hours of trying I'm starting to get to the point of giving up. But I noticed some. Because of the example FastLED "100-lines-of-code" demo reel works flawless I cut some code and decided to disable the Wifi on the ESP8266. Things started to get a little bit better. Now I only have blinking first led on startup since I need Wifi once to fetch time from NTP.  

This might be another hint..

sascha432, 2021-07-07:
I have patches the current version and it works much better for 100-250 LEDs. It rarly glitches for a single frame with no delay, and a longer delay seems to fix it.

Basically it detects the NMI interrupt and aborts the current frame to avoid glitching.
Always use delay(1) or at least delayMicros(100) after show.

https://github.com/sascha432/FastLED

sblantipodi, 2021-07-07:
@sascha432 if you think that you improved it why don't you open a pull request to let devs review it?
maybe it could be included in the main branch.

samguyer, 2021-07-07:
@sascha432 Would you be willing to contribute your code back to the project?

sascha432, 2021-07-07:
Sure, do I need to create a new fork, merge the changes and submit the request?
It looks like I cannot select particular changes from my fork but only all of them.

sascha432, 2021-07-08:
I figured it out, you need to add another remote, create a branch from it and merge your changes.

https://github.com/FastLED/FastLED/pull/1256

vonSchlau, 2021-07-08:
@sascha432 
I merged your corrections into my code (2 files) - but it does not work for me. My configuration is with 64 LEDs. When I try your code, not only the first LED flickers but nearly all other LED are flickering (mostly white colored). I tried with different animations and delays - some animaions not running att all.
Maybe my timing and configuration is too different from yours.

sascha432, 2021-07-11:
@vonSchlau 
Looks like you are running into the issue that the LEDs RES timeout is too short.
I tried a couple different strings, most are fine with WAIT_TIME 50, but some start flickering under heavy WiFi load.

I increased WAIT_TIME to 85 and that seems to make it stable. I have the "bad strip" running for about an hour without any glitches or flickering. Would be great if you can check if the last commit fixes your issues as well.

For the devs:
Looking at the data in and out, you can see that the LEDs do not detect the gap for the next frame and just continue with the last frame. That seems to happen only with recent WS2812 I bought, older ones (2-3 years old) are working fine.

vonSchlau, 2021-07-11:
@sascha432 
OK - tested your last commit - and it seems to work!
I tested about 20 animations which all work fine now (only one of my animations crashes now - it is the longest of them -  I think I can adjust it to work also )

I think this solution solved the issue for me - Thanks !

vonSchlau, 2021-07-11:
@sascha432
update: ALL animations working now (I just had to feed the dog for the last animation)
again a great thanks for your work .

kriegsman, 2021-07-18:
This issue appears to be addressed now, thanks to @sascha432, and everyone who helped test and give feedback.  PR https://github.com/FastLED/FastLED/pull/1256

Please do open a new issue (or continue here) if the new code isn't working correctly.

closed 2021-07-18 by kriegsman

sblantipodi, 2021-07-18:
@kriegsman will we see a release soon with this fix? thanks!

kriegsman, 2021-07-18:
Yes. We just merged a bunch of changes today.  I think there are just a couple more to merge in and then we’ll do an official release. In the meantime, any and all testing and feedback on the new code would be greatly appreciated. We’ll try to get the release out shortly.

sblantipodi, 2021-07-22:
hi @kriegsman, the PR has been merged, is there a new release coming ?

sblantipodi, 2021-07-23:
oops sorry, I made another question but I haven't seen your last answer kriegsman

sblantipodi, 2021-07-27:
@sascha432 trying an ESP8266 with WS2812b, core 3.0.0 is still unsable with latest main branch of fastled.

please read here: #1264

macegr, 2021-08-01:
Latest `master` branch is now way worse on both 2.7.4 and still broken on 3.0.2 haha. I had to roll back to the Dec 2020 3.4 release and ESP 2.7.4 to stop getting massive stuttering and flickering on 176 WS2812. Leaving this here in case anyone is trying to get stuff to work, using the latest versions of things, and wondering why it just won't work.

sascha432, 2021-08-04:
@macegr 
I found another issue that might cause this, depending on your code.

https://github.com/FastLED/FastLED/pull/1274

Another note which has nothing to do with the pull request:

I also had an issue recently with an LED matrix that glitches from time to time. Turns out that one LED was broken and had a messed up timing. You can identify such issues if your strip has a particular point where the glitching starts.
Or try with another strip...
The thing with the timing is, it might work perfectly fine until you get close to timing thresholds. So if that LED sees a certain color, it starts to misbehave.

hulkinBrain, 2022-09-11:
> Hi,
>
> I found a solution for my issue. When checking the LEDs with my scope, I can see that the glitching occurs when show() is called too frequently. There is a gap of 49µs between each frame, but [...]

`FastLED.delay(5);` worked for me. Now i can run WiFi with FastLED