FastLED/FastLED #1238: Issue with WS2812B

Gilpiro, 2021-05-21:
Hey,
I am quite new to this whole thing but I have a problem.
First of all, I am using:
- WS2812B
- An Arduino Mega for programming
- I have 1200 leds
The problem is that when I run a code, the first 600 leds work fine but the other 600 do not work good. It's like only red is shown there.
Do you have any idea?

5chmidti, 2021-05-21:
The cause of this is probably [voltage drop](https://learn.adafruit.com/adafruit-neopixel-uberguide/powering-neopixels#distributing-power-2894492-17). You need to inject power at some point of your strip to keep the voltage high enough.
The guide i linked is also a good read in general.

If you have any further questions, [reddit](https://www.reddit.com/r/FastLED/) would be the best place to ask, as there are more people there and the github issues are used for problems/suggestions with the library.

Gilpiro, 2021-05-21:
> The cause of this is probably [voltage drop](https://learn.adafruit.com/adafruit-neopixel-uberguide/powering-neopixels#distributing-power-2894492-17). You need to inject power at some point of your [...]

Ok, thanks!

closed 2021-05-21 by Gilpiro