FastLED/FastLED #1239: http://fastled.io/ is down

gnbl, 2021-05-23:
I am looking for the documentation but cannot find it anywhere else.

gnbl, 2021-05-23:
https://web.archive.org/web/20210417194724/http://fastled.io/ points to https://github.com/FastLED/FastLED/wiki/Overview

Please change the links (repo description, readme).

gslandtreter, 2021-05-24:
Looks like a DNS issue.
I was able to access fastled.io by hardcoding its IP address into my hostfiles
https://spyse.com/target/domain/fastled.io
You can use this as a workaround and at your own risk.

Tyzzics, 2021-05-24:
Dam, i kinda need it too, hope it comes back soon

uberjay, 2021-05-25:
I'm attempting to wrangle the necessary DNS changes. With some luck, we may be back in a day or two. (Without luck, it may take longer.)

kosigaga, 2021-05-30:
Should be closed? @uberjay

closed 2021-05-31 by gnbl