FastLED/FastLED #1241: Could FastLED be More Readable?

BrendaEM, 2021-05-31:
Firstly, I appreciate all the effort that likely even goes beyond my comprehension to preform the small miracle necessary to sequence so many NeoPixels from an Arduino. Thank you.

Respectively, the library and it's example usage, being full out C++ resembles very little like most anything in Arduino space. I've done some object-oriented programming in processing, but on the Arduino, I try to avoid it.

Perhaps I am the only one who has a problem with this, but could it be rewritten to be more user...apparent, so I would not be so tempted to work reinvent lower performing wheel?

Thank you for your consideration.
--Brenda

marmilicious, 2021-05-31:
Can you provide a bit more info on what you're thinking?  
What would you change or like to see?

BrendaEM, 2021-05-31:
Firstly, thank you for not immediately putting a hit out on me, yet.

Well, starting with the perhaps easier stuff, in the sample code, can any of the DEFINE   be const ?
could uint8_t   be   byte?
Could startIndex = startIndex + 1    be   startIndex++

Moving on...
I might guess what these overloaded operators <> do, but I won't really know.
I've not met this yet: extern const 

Syntactically, I have not met these double colon OO operators:

void SetupPurpleAndGreenPalette()
{
    CRGB purple = CHSV( HUE_PURPLE, 255, 255);
    CRGB green  = CHSV( HUE_GREEN, 255, 255);
    CRGB black  = CRGB::Black;
    
    currentPalette = CRGBPalette16(
                                   green,  green,  black,  black,
                                   purple, purple, black,  black,
                                   green,  green,  black,  black,
                                   purple, purple, black,  black );
}
const TProgmemPalette16 myRedWhiteBluePalette_p PROGMEM =
{
    CRGB::Red,
    CRGB::Gray, // 'white' is too bright compared to red and blue
    CRGB::Blue,
    CRGB::Black,
    
    CRGB::Red,
    CRGB::Gray,
    CRGB::Blue,
    CRGB::Black,
    
    CRGB::Red,
    CRGB::Red,
    CRGB::Gray,
    CRGB::Gray,
    CRGB::Blue,
    CRGB::Blue,
    CRGB::Black,
    CRGB::Black
};

5chmidti, 2021-06-03:
### can any of the DEFINE be const ?
Taking this as an example:
```c++
#define DATA_PIN    3
#define LED_TYPE    WS2811
#define COLOR_ORDER GRB
#define NUM_LEDS    64
```
It's possible to not use `#defines` for `DATA_PIN`, `COLOR_ORDER` and `NUM_LEDS`.
Now these are used as (non-type) template arguments (templates are the <> things) for the `FastLED.addLeds...` call. The requirements for template arguments are that they have to be known at compile time. Making the mentioned variables const might deduce that they are known at compile time, but it doesn't convey the intent. It would be better to use `constexpr` (must be known at compile time):
`constexpr int DATA_PIN = 3;` or `constexpr EOrder COLOR_ORDER = GRB;`.

Because the value of `LED_TYPE`, `WS2911` is a templated class, i.e. a (templated) typename, you have to used either typedef (discouraged for cpp, is more of a c thing, but is compatible) or use `using`:
```c++
template<uint8_t DATA_PIN, EOrder RGB_ORDER>
using LED_TYPE = WS2811;
```
Changing your LED_TYPE would still be possible with this, but you'd need to check if the new chipset you want to set is also a template with these two types of template parameters (many are, but not all).

### `uint8_t`
A `uint8_t` is an 8 bit unsigned integer type whose size is one `byte`. There is also `int8_t` which is signed and hase the same size. I don't think the definition of a byte specifies if it is signed or unsigned, only that it can hold 256 different values, but I might be wrong.

### `Could startIndex = startIndex + 1 be startIndex++?`
Yes.

### `::`
At a high level you can think of these as accessing variables in named scopes, which might not be the current one.
One use are `namespaces`, which are not used in FastLED, the other one is to access members (variables or types) of classes, structs, enums for various usecases.
`CRGB::Red`: this accesses the value of Red defined withing the enumeration type `HTMLColorCode` withing the class `CRGB`.

Another example is this:
You said you use globals in your current code. Maybe you write a class that does 'something'.
You can save global state withing that class (i.e. there exists a global variable that belongs to the class).
```c++
class Foo {
    static int bar =0;
};

int baz() {
    Foo f1{};
    f1.bar = 10;

    Foo::bar++ // is now 11

    Foo f2{};
    return f2.bar + Foo::bar; // returns 11+11=22
}
```

### templates
This is called generic programming.
Say you write a function that multiplies it's input by 2.
```c++
int foo(int x) { return x*2; }
```
but then you want one that does the same for floats:
```c++
float foo(float x) { return x*2; }
```
You've now overloaded your function to accept `int`'s and `float`'s (or anything that is convertible to those two).
With templates you only have to write it once:
```c++
template <typename T>
T foo(T x) {return x*2; }
```
For every call to foo it stamps out a function from the template foo for the specific type you provided. This may error out depending on the provided type (it would generate an error for a string, because there is no multiplication operator for string with an int).

Both can be used like this:
```c++
float bar() {
    int a = 10;
    float b = 4.2;
    return return foo(a) + foo(b); 
    // return foo<int>(a) + foo<float>(b); // without using template argument deduction you could write
    // the function call to the templated foo like this
}
```
This example may not fully show how powerful templates are, but they should give you an idea.
Also, as stated at the beginning, template arguments (everything within `<>`) needs to be known at compile time (for the example, that is the type of the argument).

BrendaEM, 2021-06-14:
Thank you for responding,. Is the answer generally "no" it cannot be made to be read and used similarly to most Arduino libraries?

For instance, in Arduino coding, is not uint8_t, same as: byte ?

marmilicious, 2021-06-15:
@BrendaEM  In Arduino environment, yes I believe they are equivalent. (But not everyone uses the Arduino environment.) We recently specifically changed several FastLED examples to use uint8_t instead of byte.  I believe some of this was for uniformity in the code where we are using uint8_t, uint16_t, and uint32_t.  It also makes it very clear what is being stored.  @kriegsman or @samguyer might have more reasons I haven't thought of.

As you continue to use the library hopefully it will feel more readable to you.

BrendaEM, 2021-06-20:
Unfortunately, FastLED as written is the only Arduino library I've encountered that I will not be using only because of the syntax. I guess I will have to learn how to frame and shift out the data for myself. It seems that no steps were made to naturalize it for Arduino users, and I find that disappointing.

If you drive pointy nails through a hammer handle, it makes it unwieldable.

samguyer, 2021-06-21:
@marmilicious Yes, we've seen more people using PlatformIO instead of Arduino to do microcontroller programming.

@BrendaEM I hope we can convince you to keep trying! C++ does have a steep learning curve (I think the analogy of a hammer with nails in the handle is a pretty good one ;-). But you don't need to learn all of it. There are a few crucial parts, and once you get those, you'll be flying. One thing I would encourage you to do is go on the FastLED subreddit and ask for help. It is an incredible community of people who will be more than happy to provide you with bits a pieces of code to get you started.

BrendaEM, 2021-06-22:
I did ask for help here. Is it supposed to be an Arduino library, or not?

marmilicious, 2021-06-22:
@BrendaEM If you would like some ideas on how to do something, or if you would like some info on how some piece of code you've run across works, please post to the FastLED Reddit group.

samguyer, 2021-06-22:
@BrendaEM I don't think there is any particular standard for what makes something an Arduino library. Arduino is primarily a C/C++ platform, so anything in that language family will work. Admittedly, FastLED uses some very "fancy" C++ constructs, mainly in the name of achieving high performance on low-power devices like the Uno. Even as a seasoned developer, there are aspects of the library implementation I don't fully understand.

That said, I think you can get a lot of mileage out of starting with the examples, making small changes and seeing what happens. The Reddit community is better for getting help because it taps into a much bigger group of people. Here, you are basically just talking to the people who work on the library itself.

But I also think you raise a legitimate issue: could we have a version of the library that is much more accessible to beginners (possibly at the expense of some of the power)?

BrendaEM, 2021-08-30:
No, I am sorry, I did not anything about beginners; it appears that you have either misunderstood me, or you have deliberately slandered me. I would like to think it was the former.

I am surprised that you reality-adjusted not wanting C++ showing in my Arduino libraries--all the way to I am a beginner at Arduino programing.

It's true that there are no structure or style guidelines for Arduino libraries, but after seeing one that was written with no attempt at integration with the target audience, perhaps their should be guidelines.

zinefer, 2021-09-10:
@5chmidti thanks so much for the insights above they are very helpful. While I don't really agree with the author of this issue's attitude I do wonder if there's room for a more elaborate example program. 

For example, [I have used](https://github.com/zinefer/edc-2018-totem/blob/master/code/code.ino) FastLED quite successfully for a prototype in the past. Today I am creating an updated version and trying to make the code much more extensible, with a base class that abstracts creating an led strip 'effect' down to just manipulating the `CRGB` array. I have core pinning and semaphores down and working perfectly but just failing at figuring out how to share these constant values across my codebase.

Do you have any recommendations as to how I can place my configuration into a header file that I can include across my code? I have tried a header file with the `#define`s in it but that throws a ton of errors.