FastLED/FastLED #1243: Not suited for beginners

StefanL38, 2021-06-03:
The documentation got lost. Or did you never write one?

5chmidti, 2021-06-03:
The [GitHub wiki](https://github.com/FastLED/FastLED/wiki) is a good place to look at.
Also [here](http://fastled.io/docs/3.1/) are the api docs (also linked to in the wiki) and if you've got questions, the [subreddit](https://www.reddit.com/r/FastLED/) is the best place to ask (as stated in the readme). Community guides, videos and more can be found [here](https://www.reddit.com/r/FastLED/wiki/index/user_examples), in the reddit wiki.

StefanL38, 2021-06-03:
OK I was a bit unprecise:
The example-codes or **UN**commented and do **not** show basic principles they are just a collection of several effects.

5chmidti, 2021-06-03:
The comments currently provided in the examples should be enough and I don't think they will get expanded (until someone does). The examples are part examples of using the library (like Blink, Cylon, DemoReel100, ColorPalette) and part examples of more complex efffects (Pacifica, Fire2012, Pride2015).
Explanaitions for using the library can be found in the [wiki](https://github.com/FastLED/FastLED/wiki).
If you are looking generally for examples with better explanations the youtube series from Scott Marley and Dave's Garage i [linked](https://www.reddit.com/r/FastLED/wiki/index/user_examples) are great. But that might not help if you meant explicitly the provided examples of the library.