FastLED/FastLED #1244: ESP32 + FastLED Unstable LED String

Retired2021, 2021-06-03:
Hi All,

Obtained a great Arduino sketch for a string of WS2812B's written by Mike Senese, Executive Editor at Make:.  Sketch can be found at github.com.Mazn1191/Arduino-Borealis.

I can successfully run the sketch on a Teensy 3.2 MCU with both a string of 66 LED's I have on my workbench and a much longer string of 840 LED's I have on my backyard deck.

I would like to migrate this sketch to an ESP32 to take advantage of OTA Updates and NTP Time.

I can't run the sketch on my ESP32 DEV MCU.  The sketch complies within the Arduino IDE without error however when it runs; the lights are unstable and flickering.  Tried slowing down the CPU speed from 240Mhz to 80Mhz with no effect.

What else can I try?

Arduino IDE 1.8.15
FastLED 3.4
Time 1.6

[Borealis Sketch.txt](https://github.com/FastLED/FastLED/files/6593973/Borealis.Sketch.txt)

marmilicious, 2021-06-04:
Try another pin, 16 or 17? (but that's probably not it)
Are you using a level shifter?

We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

Also, sounds cool.  If you get it running, please share a video on the reddit group.

Retired2021, 2021-06-04:
Thanks Marc.  I’m going to try a level shifter.  I forgot the ESP32 is a 3.3v device whereas the WS2812 us expecting a 5v signal.  I’ll report back

marmilicious, 2021-06-04:
There was a post just recently where I linked to a great article on level shifters.  You need to use one that is fast enough for use with addressable LED strips.  

https://www.reddit.com/r/FastLED/comments/nrizh5/is_there_a_dual_level_shifter_in_8_pin_dip_package/

Retired2021, 2021-06-04:
Thanks Marc.  I have a 74AHCT125 coming from Amazon tomorrow.  Adafruit was out of stock!

sblantipodi, 2021-06-14:
@Retired2021 did 74AHCT125 solved your flickering problem?

Retired2021, 2021-06-14:
Yes it did!  LED’s are performing normally.  Thanks for checking back

samguyer, 2021-06-22:
@Retired2021 That's great! Can we close this issue?

Retired2021, 2021-06-22:
Yes Sam.  Issue can be closed.  Level shifter fixed my problem.  Next time I will raise the issue in Reddit first instead of Github.  Thank you

samguyer, 2021-07-26:
@Retired2021 Glad to hear you're unstuck. Yeah, questions tend to get answered more quickly on Reddit. But don't hesitate to submit issues here, too, if you don't get good answers there.

closed 2021-07-26 by samguyer