FastLED/FastLED #1247: Just an FYI: fastest Dotstar the Newstar NS107S

jake653, 2021-06-13:
The Newstar NS107S has a 26+Khz refresh rate and up to 30 Mhz Data rate.  Important for POV imaging projects. (APA102 driver compatible)

I have seen announcements that the original APA102 was discontinued, and the replacements reportedly have slower pwm refresh rates, more like 8Khs.

https://www.datasheet4u.com/datasheet-pdf/NEWSTAR/NS107S/pdf.php?id=1311758

www.newstar-ledstrip.com/products-195.html

Joshfindit, 2022-06-17:
Here's a nice demo video of the dimming on the HD108S:

https://www.youtube.com/watch?v=ZMrfkSrmDDk

zackees, 2026-05-24:
Closing — informational FYI; HD107 support has since landed.

The FYI in this thread was about HD107S and NS107S being the fastest dotstars at the time. HD107 / HD107HD are now in tree at `src/fl/chipsets/apa102.h:343-368` (APA102 with a 40 MHz default clock), and the HD108 16-bit per-channel-gain variant has its own controller at `src/fl/chipsets/hd108.h`.

If anyone wants NS107S as a distinct controller (or any other dotstar variant not covered by APA102 / HD107 / HD108 / SK9822), a separate feature-request issue is welcome.

🤖 Reviewed with [Claude Code](https://claude.com/claude-code)