FastLED/FastLED #1248: Want to use the HTML enumerated color types but can't GRB to become RGB

dwc309, 2021-06-14:
I am using WS1812b led's for a display and would like to use the color types enumerated here http://fastled.io/docs/3.1/struct_c_r_g_b.html but cannot because the Green and Red order of the calls are swapped and I cannot seem to get them to swap.  I want to use them because I got tired of experimenting with individual RGB  values to get the colors that I want.

I'm using .... 

In file included from 

 #    pragma message "FastLED version 3.004.000"

here are my include files

#include <FastLED.h>                        // For Pixel Strip
#include <NTPClient.h>                      // Don't remember what this is
#include <pixeltypes.h>

This is my LED's setup code in the Setup function.

   FastLED.addLeds<WS2812B,PIXEL_PIN,RGB>(leds, NUM_LEDS);

   FastLED.clear();
   FastLED.show();
   FastLED.setBrightness(0xfe);
   FastLED.show();
   delay(2000);

   leds[0] = CRGB(255,0,0);
   leds[1] = CRGB(0,255,0);
   leds[2] = CRGB(0,255,0);
   leds[3] = CRGB(0,0,255);
   leds[4] = CRGB(0,0,255);
   leds[5] = CRGB(0,0,255);
   leds[6] = CRGB(0,0,0);
   FastLED.show();

   delay(20000);

I've looked at  Issue #282 which seems directly related to this but doesn't work as shown above.  There were a couple other issues related but not applicable to what I'm dong.

notice that I put in the Calibrate LED's which confirms the problem.  Blues are ok.

Am I using the wrong enums?

Do I need to get a different version of FastLED?  If so what version and maybe where is it?

Is there a way to fix this?

marmilicious, 2021-06-15:
I'm not sure you need the #include <pixeltypes.h> Try removing that.  

So changing the RGB order, to GRB for example, in the addLeds line doesn't swap the red and green in your basic example here?

marmilicious, 2021-06-15:
I don't see how issue 282 is related.  That's about dynamic pins, not HTML colors, right?

kostyamat, 2021-06-15:
@dwc309 You can try that
FastLED.addLeds<WS2812B,PIXEL_PIN,GBR>(leds, NUM_LEDS);

dwc309, 2021-06-15:
Thanks all for replying.  In order to use the enumerations in that link, the order needs to be RGB.  The order of the parts is GRB regardless of what is in the FastLED.addLeds statement.  I have tried both ways.  I removed the #include <pixeltypes.h> line and it made no difference.  

I previously had #include <colorutils.h> in there and it made no difference.  I replaced it when I took another look at the link containing the enumerations.  

Issue 282 was about a similar issue to mine in that the order for that person is GRB and he was unable to change the order to RGB either.    I could make my own enumerations but would prefer not to do so.  

The fact that I cannot change the order indicates that I don't understand something or am doing something wrong and that is why I posted.

marmilicious, 2021-06-15:
Not sure why it's not working for you. As a work around, instead of assigning the HTML color directly, try assigning the HTML color to a custom CRGB color, swap the r and g values and then use that.

dwc309, 2021-06-15:
That's essentially what I have been doing except that I was hoping to find out what the issue is and fix it.  Thanks again for responding.