FastLED/FastLED #1253: ATmega1284 alternative pin mappings

Delph, 2021-06-23:
Another library I use uses an older version of fastpin from here. I recently had some issues migrating from a 382 to a 1284 as I chose the Bobuino pin mapping so I didn't have to remap pins to find that the copy of fastpin didn't support these multiple pin mappings. On checking, I found the same here, and I just know it'll bite me one day.

I've added support for the Bobuino pinout to this other library locally, but since the `#defines` for it will be core/board vendor specific, I wanted to check what the "right" way, or whether it'd even be accepted, before making PRs.

kriegsman, 2021-07-29:
Thanks for opening the issue here, and for raising the questions about how to handle this.
What do the #defines for the pin mappings for the Bobuino look like?

Delph, 2021-07-29:
The version of FastPin the other library uses seems a little older, I had simply added as this for my own use;
```cpp
#ifdef BOBUINO_PINOUT
_DEFPIN_AVR( 4, (1<<0), B); _DEFPIN_AVR( 5, (1<<1), B); _DEFPIN_AVR( 6, (1<<2), B); _DEFPIN_AVR( 7, (1<<3), B);
_DEFPIN_AVR(10, (1<<4), B); _DEFPIN_AVR(11, (1<<5), B); _DEFPIN_AVR(12, (1<<6), B); _DEFPIN_AVR(13, (1<<7), B);

_DEFPIN_AVR( 0, (1<<0), D); _DEFPIN_AVR( 1, (1<<1), D); _DEFPIN_AVR( 2, (1<<2), D); _DEFPIN_AVR( 3, (1<<3), D);
_DEFPIN_AVR(30, (1<<4), D); _DEFPIN_AVR( 8, (1<<5), D); _DEFPIN_AVR( 9, (1<<6), D); _DEFPIN_AVR(31, (1<<7), D);

_DEFPIN_AVR(22, (1<<0), C); _DEFPIN_AVR(23, (1<<1), C); _DEFPIN_AVR(24, (1<<2), C); _DEFPIN_AVR(25, (1<<3), C);
_DEFPIN_AVR(26, (1<<4), C); _DEFPIN_AVR(27, (1<<5), C); _DEFPIN_AVR(28, (1<<6), C); _DEFPIN_AVR(29, (1<<7), C);

_DEFPIN_AVR(21, (1<<0), A); _DEFPIN_AVR(20, (1<<1), A); _DEFPIN_AVR(19, (1<<2), A); _DEFPIN_AVR(18, (1<<3), A);
_DEFPIN_AVR(17, (1<<4), A); _DEFPIN_AVR(16, (1<<5), A); _DEFPIN_AVR(15, (1<<6), A); _DEFPIN_AVR(14, (1<<7), A);
#else
_DEFPIN_AVR( 0, (1<<0), B); _DEFPIN_AVR( 1, (1<<1), B); _DEFPIN_AVR( 2, (1<<2), B); _DEFPIN_AVR( 3, (1<<3), B);
_DEFPIN_AVR( 4, (1<<4), B); _DEFPIN_AVR( 5, (1<<5), B); _DEFPIN_AVR( 6, (1<<6), B); _DEFPIN_AVR( 7, (1<<7), B);

_DEFPIN_AVR( 8, (1<<0), D); _DEFPIN_AVR( 9, (1<<1), D); _DEFPIN_AVR(10, (1<<2), D); _DEFPIN_AVR(11, (1<<3), D);
_DEFPIN_AVR(12, (1<<4), D); _DEFPIN_AVR(13, (1<<5), D); _DEFPIN_AVR(14, (1<<6), D); _DEFPIN_AVR(15, (1<<7), D);

_DEFPIN_AVR(16, (1<<0), C); _DEFPIN_AVR(17, (1<<1), C); _DEFPIN_AVR(18, (1<<2), C); _DEFPIN_AVR(19, (1<<3), C);
_DEFPIN_AVR(20, (1<<4), C); _DEFPIN_AVR(21, (1<<5), C); _DEFPIN_AVR(22, (1<<6), C); _DEFPIN_AVR(23, (1<<7), C);

_DEFPIN_AVR(24, (1<<0), A); _DEFPIN_AVR(25, (1<<1), A); _DEFPIN_AVR(26, (1<<2), A); _DEFPIN_AVR(27, (1<<3), A);
_DEFPIN_AVR(28, (1<<4), A); _DEFPIN_AVR(29, (1<<5), A); _DEFPIN_AVR(30, (1<<6), A); _DEFPIN_AVR(31, (1<<7), A);
#endif
```

Pinout diagrams are available on [MCUdude's MightyCore repo](https://github.com/MCUdude/MightyCore#pinout), the other two #defines used are `STANDARD_PINOUT` and `SANGUINO_PINOUT`. https://github.com/MCUdude/MightyCore/tree/master/avr/variants