FastLED/FastLED #1254: "Invalid Pin Specified" Error Arduino Nano Every

Cai-Williams, 2021-06-28:
I have a Arduino Nano Every and I have a WS2812B led strip hooked up to it and I'm trying to run FastLED on it but whenever I try to upload the sketch I get this output:

C:\Users\Documents\Arduino\libraries\FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"

 And

C\Users\Documents\Arduino\libraries\FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specifiedstatic_assert(validpin(), "Invalid pin specified") 

And no matter where I look I can't find any suggestions or fixes. This is my code:

#include <FastLED.h>

#define LED_PIN 4
#define NUM_LEDS 20

CRGB leds[NUM_LEDS];

void setup() {
    FastLED.addLeds<WS2812B, LED_PIN, GRB>(leds, NUM_LEDS);
}

void loop() {
  for (int i = 0; i <= 19; i++) {leds[i] = CRGB ( 0, 0, 255);
    FastLED.show();
    delay(40);
  }
  for (int i = 19; i >= 0; i--) {leds[i] = CRGB ( 255, 0, `0);
    FastLED.show();
    delay(40);
  }
}

If anyone knows how to fix this or anything that might help please contact me or reply to this!

Thanks,
Cai

marmilicious, 2021-06-28:
That warning is always present on platforms that haven't had SPI support added yet or don't support it.  You can ignore that.

Cai-Williams, 2021-06-28:
@marmilicious Thanks for your reply.  When I upload the sketch It says "unable to upload to Arduino Nano Every" and I don't get any led output either. 🤷‍♂️ That might just be a fault with my led strip 🤣

marmilicious, 2021-06-28:
I believe with release FastLED 3.4.0 the nano every was added.
"Merged in contributed support for Arduino Nano Every / Arduino Uno Wifi Rev. 2"
Are you using the latest version of FastLED?
Try some different pins for your data line?
Try posting to the FastLED Reddit group where hopefully someone that has that board can give a suggestion.

Cai-Williams, 2021-06-28:
Okay. Thank you!

zackees, 2026-06-03:
**fbuild build-system status update** (tracker: FastLED/fbuild#389)

Build-tool side is healthy (same as FastLED/FastLED#716):

- Board metadata `nano_every.json` present.
- `arduino_megaavr` framework registered.
- Smoke fixture at `tests/platform/nano_every/` builds cleanly.
- FastLED has m4809 pin support at `src/platforms/avr/atmega/m4809/fastpin_avr_nano_every.h`.

`LED_PIN 4` on Nano Every maps to PORTC bit 6 — make sure your sketch is using the m4809 fastpin table (recent FastLED versions) and not the legacy ATmega328 path. If the `Invalid Pin Specified` static_assert still fires on a current FastLED master, please reopen with the full preprocessor dump.