FastLED/FastLED #1255: Request support for  ESP32-C3-DevKitM-1 too.

BlynkGO, 2021-07-04:
Arduino ESP32 core 2.0.0 alpha1  now support for ESP32-C3-DevKitM-1.
and ESP32-C3-DevKitM-1 has WS2812 LED at GPIO8 on board.

Do you mind to support FastLed library for ESP32-C3-DevKitM-1  too?

Thank you very much.

lybgo, 2021-07-17:
Request support for ESP32-C3  too

samguyer, 2021-07-26:
@BlynkGO  and @lybgo Do you know what the differences are between the standard ESP32 and the ESP32-C3? Would you be willing to help us make the necessary changes to FastLED?

lybgo, 2021-07-28:
@samguyer Esp32c3 is cheaper than esp32. It supports WiFi + ble5.0 (esp32 ble4.2) and risc-v MCU (esp32 xtensa)

samguyer, 2021-07-29:
@lybgo What happens if you just run the regular ESP32 code for FastLED on an ESP32-C3? Does it work? If not, why?

lybgo, 2021-07-30:
@samguyer The code failed to compile

samguyer, 2021-07-30:
@lybgo I don't have an ESP32-C3 module -- can you share the error output with me? From the datasheet, I don't see any reason why our code wouldn't compile. I would caution you, however, that even the dual-core ESP32 has trouble supporting Wifi and FastLED at the same time, especially when a lot of data is being sent on the network. Even if we get it compiled and working, you might not get the performance you want.

lybgo, 2021-08-02:
@samguyer 
error output:
```
In file included from e:\ArduinoDevelopNew\Esp32\Study\FastLedTest\FastLedTest.ino:9:
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
#    pragma message "FastLED version 3.004.000"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:3,
from C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms.h:36,
from C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/FastLED.h:52,
from e:\ArduinoDevelopNew\Esp32\Study\FastLedTest\FastLedTest.ino:9:
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::hi()':
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h:15:17: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'out1_w1ts'; did you mean 'out_w1ts'?
else GPIO.out1_w1ts.val = MASK;
                 ^~~~~~~~~
                 out_w1ts
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::lo()':
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h:20:17: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'out1_w1tc'; did you mean 'out_w1tc'?
       else GPIO.out1_w1tc.val = MASK;
^~~~~~~~~
                 out_w1tc
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::set(_ESPPIN<PIN, MASK>::port_t)':
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h:25:17: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'out1'; did you mean 'out'?
       else GPIO.out1.val = val;
                 ^~~~
                 out
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h: In static member function 'static void _ESPPIN<PIN, MASK>::toggle()':
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastpin_esp32.h:31:31: error: no match for 'operator^=' (operand types are 'volatile<unnamed struct>::<unnamed union>' and 'long unsigned int')
if(PIN < 32) { GPIO.out ^= MASK; }
                      ~~~~~~~~~^~~~~~~
In file included from C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/freertos/port/riscv/include/freertos/portmacro.h:44,
from C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/freertos/include/freertos/portable.h:52,
from C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/freertos/include/freertos/FreeRTOS.h:64,
from C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:33,
from C:\Users\lybgo\AppData\Local\Temp\arduino_build_162118\sketch\FastLedTest.ino.cpp:1:
C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:123:25: note: candidate: 'touch_pad_intr_mask_t& operator^=(touch_pad_intr_mask_t&, touch_pad_intr_mask_t)'
FORCE_INLINE_ATTR TYPE& operator^=(TYPE& a, TYPE b) { a = a ^ b; return a; } \
                         ^~~~~~~~
C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:127:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
#define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
                             ^~~~~~~~~~~~~~
C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:128:19: note: in expansion of macro 'FLAG_ATTR_U32'
#define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/hal/include/hal/touch_sensor_types.h:165:1: note: in expansion of macro 'FLAG_ATTR'
FLAG_ATTR(touch_pad_intr_mask_t)
 ^~~~~~~~~
C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:123:25: note:   no known conversion for argument 1 from 'volatile<unnamed struct>::<unnamed union>' to 'touch_pad_intr_mask_t&'
FORCE_INLINE_ATTR TYPE& operator^=(TYPE& a, TYPE b) { a = a ^ b; return a; } \
^~~~~~~~
C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:127:29: note: in expansion of macro 'FLAG_ATTR_IMPL'
 #define FLAG_ATTR_U32(TYPE) FLAG_ATTR_IMPL(TYPE, uint32_t)
^~~~~~~~~~~~~~
C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/esp_common/include/esp_attr.h:128:19: note: in expansion of macro 'FLAG_ATTR_U32'
#define FLAG_ATTR FLAG_ATTR_U32
                   ^~~~~~~~~~~~~
C:\Users\lybgo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/esp32c3/include/hal/include/hal/touch_sensor_types.h:165:1: note: in expansion of macro 'FLAG_ATTR'
FLAG_ATTR(touch_pad_intr_mask_t)
 ^~~~~~~~~
In file included from C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:3,
                 from C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/platforms.h:36,
from C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/FastLED.h:52,
                 from e:\ArduinoDevelopNew\Esp32\Study\FastLedTest\FastLedTest.ino:9:
[... cut ...]
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src/fastpin.h:210:24: error: static assertion failed: Invalid pin specified
  static_assert(validpin(), "Invalid pin specified");
~~~~~~~~^~
"FastLED.h" 对应多个库
已使用： C:\Users\lybgo\Documents\Arduino\libraries\FastLED
 未使用：C:\Users\lybgo\Documents\Arduino\libraries\FastLED-3.3.3
exit status 1
IntelliSense configuration updated. To manually rebuild your IntelliSense configuration run "Ctrl+Alt+I"
[Error] Uploading sketch 'FastLedTest.ino': Exit with code=1

```

lybgo, 2021-08-02:
@samguyer 
```
fastpin_esp32.h:15:17: error: 'gpio_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'out1_w1ts'; did you mean 'out_w1ts'?
else GPIO.out1_w1ts.val = MASK;
```
Similar to the above problems, I solved the problem by modifying the "fastpin_esp32.h" document. The contents of the modified file are as follows:
```
#pragma once

FASTLED_NAMESPACE_BEGIN

template<uint8_t PIN, uint32_t MASK> class _ESPPIN {
public:
  typedef volatile uint32_t * port_ptr_t;
  typedef uint32_t port_t;

  inline static void setOutput() { pinMode(PIN, OUTPUT); }
  inline static void setInput() { pinMode(PIN, INPUT); }

  inline static void hi() __attribute__ ((always_inline)) {
      GPIO.out_w1ts.val = MASK;
  }

  inline static void lo() __attribute__ ((always_inline)) {
      GPIO.out_w1tc.val = MASK;
  }

  inline static void set(register port_t val) __attribute__ ((always_inline)) {
      GPIO.out.val = val;
  }

  inline static void strobe() __attribute__ ((always_inline)) { toggle(); toggle(); }

  inline static void toggle() __attribute__ ((always_inline)) {
      GPIO.out.val ^= MASK;
  }

  inline static void hi(register port_ptr_t port) __attribute__ ((always_inline)) { hi(); }
  inline static void lo(register port_ptr_t port) __attribute__ ((always_inline)) { lo(); }
  inline static void fastset(register port_ptr_t port, register port_t val) __attribute__ ((always_inline)) { *port = val; }

  inline static port_t hival() __attribute__ ((always_inline)) {
      return GPIO.out.val | MASK;
  }

  inline static port_t loval() __attribute__ ((always_inline)) {
      return GPIO.out.val & ~MASK;
  }

  inline static port_ptr_t port() __attribute__ ((always_inline)) {
      return &GPIO.out.val;
  }

  inline static port_ptr_t sport() __attribute__ ((always_inline)) {
      return &GPIO.out_w1ts.val;
  }

  inline static port_ptr_t cport() __attribute__ ((always_inline)) {
      return &GPIO.out_w1tc.val;
  }

  inline static port_t mask() __attribute__ ((always_inline)) { return MASK; }

  inline static bool isset() __attribute__ ((always_inline)) {
      return GPIO.out.val & MASK;
  }
};

#define _FL_DEFPIN(PIN)  template<> class FastPin<PIN> : public _ESPPIN<PIN, ((PIN<32)?((uint32_t)1 << PIN):((uint32_t)1 << (PIN-32)))> {};

[... 52 lines cut ...]
esp_err_t rmt_set_pin(rmt_channel_t channel, rmt_mode_t mode, gpio_num_t gpio_num)
^~~~~~~~~~~
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In member function 'void ESP32RMTController::tx_start()':
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:278:9: error: 'rmt_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'conf_ch'
RMT.conf_ch[mRMT_channel].conf1.mem_rd_rst = 1;
         ^~~~~~~
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:279:9: error: 'rmt_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'conf_ch'
RMT.conf_ch[mRMT_channel].conf1.mem_rd_rst = 0;
^~~~~~~
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:282:9: error: 'rmt_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'conf_ch'
RMT.conf_ch[mRMT_channel].conf1.tx_start = 1;
^~~~~~~
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::doneOnChannel(rmt_channel_t, void*)':
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:304:9: error: 'rmt_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'conf_ch'
RMT.conf_ch[channel].conf1.mem_rd_rst = 1;
         ^~~~~~~
C:\Users\lybgo\Documents\Arduino\libraries\FastLED\src\platforms\esp\32\clockless_rmt_esp32.cpp:305:9: error: 'rmt_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'conf_ch'
RMT.conf_ch[channel].conf1.mem_rd_rst = 0;
^~~~~~~

exit status 1
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Ctrl+Alt+I"
[Error] Uploading sketch 'FastLedTest.ino': Exit with code=1

```

lybgo, 2021-08-02:
@samguyer 
```
clockless_rmt_esp32.cpp:278:9: error: 'rmt_dev_t' {aka 'volatile struct<unnamed>'} has no member named 'conf_ch'
RMT.conf_ch[mRMT_channel].conf1.mem_rd_rst = 1;
```
Similar to the above problems, I solved the problem by modifying the "clockless_rmt_esp32.cpp" document. The contents of the modified file are as follows:
```
void IRAM_ATTR ESP32RMTController::tx_start()
{
    // rmt_tx_start(mRMT_channel, true);
    // Inline the code for rmt_tx_start, so it can be placed in IRAM
    RMT.tx_conf[mRMT_channel].conf1.mem_rd_rst = 1;
    RMT.tx_conf[mRMT_channel].conf1.mem_rd_rst = 0;
    RMT.int_ena.val &= ~(1 << (mRMT_channel * 3));
    RMT.int_ena.val |= (1 << (mRMT_channel * 3));
    RMT.tx_conf[mRMT_channel].conf1.tx_start = 1;
    mLastFill = __clock_cycles();
}
```

esp32c3 api "Arduino\hardware\espressif\esp32\tools\sdk\esp32c3\include\soc\esp32c3\include\soc\rmt_struct.h" 
Related content:
```
union {
        struct {
            uint32_t tx_start:           1;
            uint32_t mem_rd_rst:         1;
            uint32_t mem_rst:            1;
            uint32_t tx_conti_mode:      1;
            uint32_t mem_tx_wrap_en:     1;
            uint32_t idle_out_lv:        1;
            uint32_t idle_out_en:        1;
            uint32_t tx_stop:            1;
            uint32_t div_cnt:            8;
            uint32_t mem_size:           3;
            uint32_t reserved19:         1;
            uint32_t carrier_eff_en:     1;
            uint32_t carrier_en:         1;
            uint32_t carrier_out_lv:     1;
            uint32_t afifo_rst:          1;
            uint32_t conf_update:        1;
            uint32_t reserved25:         7;
        } conf1;
        uint32_t val;
    } tx_conf[2];
```

lybgo, 2021-08-02:
@samguyer 
```
FastLED\src/platforms/esp/32C3/clockless_rmt_esp32.h:139:52: error: impossible constraint in 'asm'
   __asm__ __volatile__ ("rsr %0,ccount":"=a" (cyc));
```

Above problems, I solved the problem by modifying the "clockless_rmt_esp32.cpp" document. The contents of the modified file are as follows:
```
__attribute__ ((always_inline)) inline static uint32_t __clock_cycles() {
  uint32_t cyc;
  //__asm__ __volatile__ ("rsr %0,ccount":"=a" (cyc));
  asm volatile ("rdcycle x1" : "=r" (cyc) );
  return cyc;
}
```

maddogjt, 2021-11-17:
I've put up a pull request with support for C3 #1320

closed 2021-11-30 by samguyer

societyofrobots, 2021-12-30:
> I've put up a pull request with support for C3 #1320

Your code fixed everything for me except this which I get when compiling the demos with FastLED. Safe to ignore?
```
In file included from C:\Users\palmi\Documents\Arduino\libraries\FastLED\examples\FirstLight\FirstLight.ino:6:
C:\Users\palmi\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\palmi\Documents\Arduino\libraries\FastLED\src/FastLED.h:65,
                 from C:\Users\palmi\Documents\Arduino\libraries\FastLED\examples\FirstLight\FirstLight.ino:6:
C:\Users\palmi\Documents\Arduino\libraries\FastLED\src/fastspi.h:145:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

```