FastLED/FastLED #1257: ESP8285 + wifi = led glitching

slowgergo, 2021-07-13:
Hello! Today, I found some weird glitching with some ws2812b leds.

### Hardware/Software:
```
Platformio under VSCode
FastLED 3.4.0
ESP01-F (ESP8285) 
A custom pcb (the 6 pcbs are soldered together to form a cube, 1 side has the esp, all sides have 9 ws2812b leds) 
adxl345 3 axis accelerometer
```

### Goal:
I designed some animations for the cube but it would be great to switch between them on a phone  through wifi (+ lot more settings are possible vs the one single button on the cube itself) so it would be awesome to make the html page work without the leds glitching.

### Problem
After some hours of debugging I found the cause of a weird glitch.  I have an animation where the glitching is really obvious. It uses the adxl345 to make an electronic dice so if its sitting still only one side is lit with the appropriate number of leds.

While this is working great I saw that often the 0th led is flashing random colors while the middle on cycles through the colors as it should because the top side is "1".
I figured out that if I comment out all the code that makes wifi and the html page work the flashing disappears.
With the wifi and asyncWebserver code in tact sometimes it doesn't flash for 3-4 seconds or it flashes constantly, but it flashes the most when I'm connecting to it with my phone.

As far as I understand my code is not blocking the wifi (the esp never crashes) so why does it interfere with the leds?

<details> 
  <summary>The code</summary>

```C++
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <FS.h>
#include <FastLED.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_ADXL345_U.h>

#define NUM_LEDS 54
#define PIN 13
#define BTN 12
#define latch 14
#define offVoltage 637
AsyncWebServer server(80);
CRGB leds[54];
Adafruit_ADXL345_Unified accel = Adafruit_ADXL345_Unified(12345);

IPAddress local_IP(192,168,4,2);
IPAddress gateway(192,168,4,1);
IPAddress subnet(255,255,255,0);

unsigned long lastDiceColor = 0;
byte univDelay = 15; //delay var
byte delays[5] = {15, 60, 40, 150, 20}; //delays for normal anim
byte doubleDelays[5] = {20}; //delays for adxl anim
bool staticMode = false; //static mode for web 
byte h,s,v; //color variables
byte buf = 0; //num of leds on load 
bool singlePress = false;
bool doublePress = false;
bool longPress = false;
unsigned long lastSensorCheck = 0; //adxl last time
byte sensorDelay = 20;  //adxl loop delay #anim is delayed aswell FIX
unsigned long currMillis = 0; //curent time
byte animNormal = 0; //num of normal anim
byte animDouble = 0; //num of adxl anim
float x,y,z; //adxl
byte side = 0; //side id 1 is top
byte MODE = 1; 
/*
0 normal mode
1 adxl mode
[... 230 lines cut ...]
monitor_filters = esp8266_exception_decoder
monitor_dtr = 0
monitor_rts = 0
upload_speed = 921600
upload_resetmethod = nodemcu
lib_deps = 
    me-no-dev/ESP Async WebServer @ ~1.2.3
    fastled/FastLED @ ~3.4.0
    adafruit/Adafruit ADXL345 @ ~1.2.2
    adafruit/Adafruit Unified Sensor @ ~1.1.4

build_flags = 
    -Wno-register
    -Wno-deprecated-declarations
    -Wno-comment

```
</details>

This is 1/3 of the code, its stripped down for debugging, some variables are not used/missing and some server functions are useless.
The flashing is noticeable in every animation but this is the most obvious.

Any help is appreciated!

PS: I once commented out `dice(side);` in the loop after the sensor code,(line 252) as expected no led was lit except for the flashing 0th one.

sascha432, 2021-07-14:
@slowgergo 
this might help

https://github.com/FastLED/FastLED/issues/1237

slowgergo, 2021-07-14:
With that edited lib now the whole thing flickers and the first one still flickers in that different way. (Im now trying with the exact code that I provided here updating shortly)

slowgergo, 2021-07-14:
I'm happy to test any ideas that might fix this issue. But in the meantime while I complete the code/project I'm using the 2.6.3 esp8266 core where everything seems to work fine apart from one dumb function somewhere...

sascha432, 2021-07-14:
You could try to increase WAIT_TIME to 100, if that does not help it is probably another issue.

https://github.com/FastLED/FastLED/pull/1256/commits/5967bc3f1dbf203fa60158da48497a58fff3ff2a

slowgergo, 2021-07-14:
I tried with WAIT_TIME 100 and 200: results same as before flickering like crazy
I changed it back to the original 50 and it was still flickering so some of the other changes made it flicker more.

I might add that the other thread was about esp core 3.0.0. My version is the latest 3.0.1

kriegsman, 2021-07-18:
Please try the lastest code, just merged into FastLED from @sascha432, and let us know if this changes the problem, or fixes it, or makes it worse.

slowgergo, 2021-07-19:
Sorry for taking a bit too long had problems with platformio and its library updating...

So long story short it made it worse than before the merge. But with this version of the lib and esp core 2.6.3 it works fine. 

I should add that I use gpio 13 for the leds which is `MTCK  I/O 13; HSPI_MOSI; UART0_CTS1` according to the datasheet so not an spi pin. 
My arduino ide installation is acting up so I cant test the behavior with that unfortunately. Will try to though. 

Should I try it with a NodeMCU board to see if the 8285 is the culprit?

DmytroKorniienko, 2021-07-19:
> Please try the lastest code, just merged into FastLED from @sascha432, and let us know if this changes the problem, or fixes it, or makes it worse.

Not working for me at all (ESP8266 NodeMCU dev board), please look at explenation here: https://github.com/FastLED/FastLED/pull/1256#issuecomment-882602084