FastLED/FastLED #1258: arduino rp2040 support

SiliCarbConverter, 2021-07-15:
could you please let us know if there is going to be support for the rp2040 nano from arduino
in the near future. i have a project that relies on it and being a complete idiot forgot to check on compatibility issues. i am using APA102 leds the only library option i have at the moment is APA102.h . thank you.

ageppert, 2021-07-19:
A work around in the short-term is to use the Adafruit NeoPixel library which has been updated to support the rp2040 and pico with a PIO implementation: https://github.com/Bodmer/Adafruit_NeoPixel

SiliCarbConverter, 2021-07-19:
Thanks i will take a look.

ATLTVHEAD, 2021-08-11:
Would also love support for the rp2040 chip!

chucks13, 2021-12-24:
The neopixel library works great for driving led strips, but it would be nice to use the utility functions in FastLED for creating animations.  Is there a way to use the library for boards that are not supported, even if it means there are no output capabilities?

nmaas87, 2022-05-10:
https://github.com/FastLED/FastLED/pull/1261

zackees, 2024-10-12:
It's supported and there's a builder for it that's green.

closed 2024-10-12 by zackees