FastLED/FastLED #1259: Teensy4 WS2812B not working

imwhocodes, 2021-07-19:
Hi

I'm trying to use Fastled with a Teensy4 but the resulting colors on the connected ledstrip are wrong
It seem like the color output is shifted or Xored is some way

Here  a sample code for reproducing the issue, all the led strip stay at half power white
(On a Teensy 3.2 the code work without any problem)

```
#include <Arduino.h>

#include "FastLED.h"

#define LED_PIN_1 9
#define STRIP_LENGTH_1 35
CRGB data_1 [STRIP_LENGTH_1];

#define LED_PIN_2 10
#define STRIP_LENGTH_2 35
CRGB data_2 [STRIP_LENGTH_2];

#define LED_PIN_3 11
#define STRIP_LENGTH_3 25
CRGB data_3 [STRIP_LENGTH_3];

#define LED_PIN_4 12
#define STRIP_LENGTH_4 19
CRGB data_4 [STRIP_LENGTH_4];

void setup() {

  FastLED.addLeds<NEOPIXEL, LED_PIN_1>(data_1, STRIP_LENGTH_1);

  FastLED.addLeds<NEOPIXEL, LED_PIN_2>(data_2, STRIP_LENGTH_2);

  FastLED.addLeds<NEOPIXEL, LED_PIN_3>(data_3, STRIP_LENGTH_3);

  FastLED.addLeds<NEOPIXEL, LED_PIN_4>(data_4, STRIP_LENGTH_4);

  FastLED.clear();

}

void loop() {

  FastLED.show();

}
```

Also attached a photo of the signal of one of the pins if it can be of any help
[image]

Tested on:
- Platformio 5.1.1
- FastLed 3.3.0 / 3.4.0 / git-master
- TennsyCore 1.53 / 1.54

Thanks,
Luca

marmilicious, 2021-07-22:
Hello @imwhocodes   You need to specify NUM_STRIPS when using Teensy 4  

https://github.com/FastLED/FastLED/wiki/Parallel-Output#parallel-output-on-the-teensy-4

FastLED.addLeds<NUM_STRIPS, WS2812, PIN, GRB>(leds, NUM_LEDS);

imwhocodes, 2021-07-23:
Thanks @marmilicious 

I already tried "Parallel Output", but as you can see from the code I posted above I have no access to the "hardcoded" pins
Also the strips have different sizes(number of led) from one to another, from yours example seem like "Parallel Output" is only usable with same dimension led strips

I was searching how to do plain vanilla "Bit-Bangin" on the Teensy4, as now I felt back to `<Adafruit_NeoPixel.h>` for driving the leds

For sure "Parallel Output" is an amazing feature, but it will be cool to also have the generic interface for the more trivial cases

Thanks,
Luca

lpaolini, 2021-07-23:
Hello @imwhocodes,

Teensy 4.x works amazingly well with FastLED via the [WS2812Serial](https://github.com/PaulStoffregen/WS2812Serial) lib by Paul Stoffregen, and it's non-blocking too.

The only limitation is that you have to use one of the Serial ports, 5 on Teensy 4.0, 8 on Teensy 4.1, which become 3 and 6 in case you also need the [Audio Shield](https://www.pjrc.com/audio-shield-for-teensy-4-0/).

Actually, Teensy 4.0 has an extra Serial port on a solder pad underneath the board.

I'm using it in [my project](https://github.com/lpaolini/Striptease/), and it works like a charm.

Cheers,
Luca

imwhocodes, 2021-07-23:
Guys Thanks

I know and already used WS2812Serial, and I fully grasp the "Parallel Output" mode

I'm asking if there is the generic/default way (so bitbanging, _no hardware nor other platform specific_ ) to use Fastled on a Teensy4.0

**In my case choosing which pin number to use for the leds is not an option**

WS2812Serial and Parallel Output are amazing, but they force you to use _specific pins_ or use _up serial lines_

I find really strange that the most trivial case does not work:
- random PIN
- blocking is OK
- one single strip at time is OK

Thank Luca

P.s I started using FastLed on the esp8266 almost 5 years ago, for sure it was and it is one of the best library on the arduino world, thanks you to everybody

somewhatlurker, 2021-07-27:
Yeah, I don't have Teensy 4, but checking out `clockless_arm_mxrt1062.h` (which I believe is the correct file for its regular implementation), the writeBits implementation seems kinda screwy: https://github.com/FastLED/FastLED/blob/26e222e8eda4c91c829dcd22016b8d91b51e48ab/src/platforms/arm/mxrt1062/clockless_arm_mxrt1062.h#L53

　

Breakdown, both so we can establish what's going on and anyone can correct me if I'm wrong:
(let's just consider the for loop right now, which I believe is responsible for 7 out of 8 bits in normal operation with 8-bit LEDs)
1. (not something the function does, but worth noting) Observe the `off` array: https://github.com/FastLED/FastLED/blob/26e222e8eda4c91c829dcd22016b8d91b51e48ab/src/platforms/arm/mxrt1062/clockless_arm_mxrt1062.h#L90
    * [0]: total time to send a bit
    * [1]: total time minus time high for a low bit (low bit is high for T1, then low)
    * [2]: total time minus time high for a high bit (high bit is high for T1 + T2, then low)
    * Honestly not sure why this is being re-initialised on every show, as it's constant, but that's not really gonna hurt anything
2. Waits until `next_mark` has been reached before writing each bit, then sets `next_mark` for the next iteration
    * (`next_mark` is always current time at start of bit plus `off[0]` -- which is correct)
    * Okay, reasonable enough.
3. Drives `DATA_PIN` high
    * Perfect
4. If bit of byte being sent is high, wait until `next_mark` is less than `off[1]` time units away, else (if it's low) wait until it's `off[2]` time units away
    * **Well here's an issue: I think the bits are flipped!**
    * If you wait only until `off[1]` is remaining, you should send a low bit. If you wait until only `off[2]` is remaining, that should be a high bit.
5. Drives `DATA_PIN` low
    * Well that's fine
6. Shifts the input byte so that next time the second bit will be output

And then there's the special case for last bit? which seems to be using different `next_mark` timing for some reason? AFAIK that's not part of how normal clockless LEDs are driven -- this should probably just be handled by the same loop as other bits.
**And it uses the same timing for both high and low bits!**

So, yeah...  I think this is just a broken implementation. Did it ever work??

somewhatlurker, 2021-07-27:
I haven't even tested compilation, but if anyone with the hardware wants to try it, I believe this may work as a writeBits implementation.
```
	template<int BITS> __attribute__ ((always_inline)) inline void writeBits(register uint32_t & next_mark, register uint32_t & b)  {
		for(register uint32_t i = BITS; i > 0; --i) {
			while(ARM_DWT_CYCCNT < next_mark);
			next_mark = ARM_DWT_CYCCNT + off[0];
			FastPin<DATA_PIN>::hi();
			if(b&0x80) {
				while((next_mark - ARM_DWT_CYCCNT) > off[2]);
				FastPin<DATA_PIN>::lo();
			} else {
				while((next_mark - ARM_DWT_CYCCNT) > off[1]);
				FastPin<DATA_PIN>::lo();
			}
			b <<= 1;
		}
	}
```

kriegsman, 2021-07-29:
Thank you for the report of the problem, @imwhocodes, and thank you for looking into the implementation, @somewhatlurker.   I'll try to dig up my Teensy4 and see what works, but if anyone else has a Teensy4 and can try the new implementation from @somewhatlurker, that would be very helpful.

DavidMenting, 2021-08-10:
I can confirm that the change by @somewhatlurker solves the issue at least on the Teensy 4.1. I was getting inverted colors without the change but now the colors seem fine.

devoh747, 2022-05-22:
@marmilicious any thoughts on when this fix may make it into mainline?

marmilicious, 2022-05-23:
Hello @devoh747 
This should be a question for @kriegsman or @samguyer