FastLED/FastLED #1260: First led of WS2812B is glitching with NodeMCU

pla10, 2021-07-20:
Hello! I decided to work again in a old project with the WS2812B and noticed that after updating the libraries, the first led started blinking randomly. I could isolate the problem. It happens only when the esp8266 connects to the wifi and the FastLED.show() is called, otherwise no glitch happens. Do you have any suggestions?

pla10, 2021-07-20:
`#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#define FASTLED_ALLOW_INTERRUPTS 0
#include <FastLED.h>

#define LED_PIN     D5
#define NUM_LEDS    279
#define LED_TYPE    WS2812B
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

int BRIGHTNESS = 64;
int preBrightness = 64;
int r = 255, g = 255, b = 255;
bool change = false;
int i = 0, dir = 1;

#define UPDATES_PER_SECOND 100

bool cylonFlag;
CRGBPalette16 currentPalette;
TBlendType    currentBlending;

// VARIABILI DA MODIFICARE CON LE INFORMAZIONI DEL DISPOSITIVO CONTROLLATO -----------------------------------------------
#define IDX 11                                                       // Codice idx del dispositivo sul server domoticz
#define TOPIC_NAME "Scale/Led"                                       // Nome del topic relativo al dispositivo controllato
//------------------------------------------------------------------------------------------------------------------------

const char* ssid = "**************";                                     // Dati connessione WIFI
const char* password = "***************";
const char* mqtt_server = "***.***.*.*";

WiFiClient espClient;
PubSubClient client(espClient);

char hellomsg[40];                                                  // Formattazione del messaggio di handshake

void setup() {
  delay( 3000 ); // power-up safety delay
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
  FastLED.setBrightness(  BRIGHTNESS );
  
  //Serial.begin(9600);                                           // Configurazione Seriale
  pinMode(LED_BUILTIN, OUTPUT);                                     // Imposto LED per stato WiFi
  
  sprintf(hellomsg,"{\"content\":\"%s connesso\"}",TOPIC_NAME);
  
  setup_wifi();                                                     // Connessione alla rete WIFI
  client.setServer(mqtt_server, 1883);                              // Connessione al server MQTT
  client.setCallback(callback);                                     // Configurazione dell'ascolto del topic in MQTT
}

void loop() {
  if (!client.connected()) {                                        // Gestione connessione WIFI
    reconnect();
  }
  client.loop();

  if(change)
    ChangePalettePeriodically();

  if(cylonFlag) cylon();
  else{ 
    static uint8_t startIndex = 0;
    startIndex = startIndex + 1; // motion speed
    
    FillLEDsFromPaletteColors( startIndex);
    
    FastLED.show();
[... 192 lines cut ...]

void setup_wifi() {                                                 // Collegamento alla rete WIFI
  delay(10);
  //Serial.println();
  //Serial.print("Connecting to ");
  //Serial.println(ssid);
  digitalWrite(LED_BUILTIN, LOW);                                   // Accensione led per segnalare inizio connessione
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);                                       // We start by connecting to a WiFi network
  
  while (WiFi.status() != WL_CONNECTED) {
    for(int i=0;i<10;i++) {
      delay(50);
    }
    //Serial.print(".");
  }
  //Serial.println();
  //Serial.println("WiFi connected");
  //Serial.println("IP address: ");
  //Serial.println(WiFi.localIP());
  digitalWrite(LED_BUILTIN, HIGH);                                    // Spegnimento led per segnalare l'avvenuta connessione
}

`

pla10, 2021-07-20:
```
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#define FASTLED_ALLOW_INTERRUPTS 0
#include <FastLED.h>

#define LED_PIN     D5
#define NUM_LEDS    279
#define LED_TYPE    WS2812B
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

int BRIGHTNESS = 64;
int preBrightness = 64;
int r = 255, g = 255, b = 255;
bool change = false;
int i = 0, dir = 1;

#define UPDATES_PER_SECOND 100

bool cylonFlag;
CRGBPalette16 currentPalette;
TBlendType    currentBlending;

// VARIABILI DA MODIFICARE CON LE INFORMAZIONI DEL DISPOSITIVO CONTROLLATO -----------------------------------------------
#define IDX 11                                                       // Codice idx del dispositivo sul server domoticz
#define TOPIC_NAME "Scale/Led"                                       // Nome del topic relativo al dispositivo controllato
//------------------------------------------------------------------------------------------------------------------------

const char* ssid = "**************";                                     // Dati connessione WIFI
const char* password = "***************";
const char* mqtt_server = "***.***.*.*";

WiFiClient espClient;
PubSubClient client(espClient);

char hellomsg[40];                                                  // Formattazione del messaggio di handshake

void setup() {
  delay( 3000 ); // power-up safety delay
  FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
  FastLED.setBrightness(  BRIGHTNESS );
  
  //Serial.begin(9600);                                           // Configurazione Seriale
  pinMode(LED_BUILTIN, OUTPUT);                                     // Imposto LED per stato WiFi
  
  sprintf(hellomsg,"{\"content\":\"%s connesso\"}",TOPIC_NAME);
  
  setup_wifi();                                                     // Connessione alla rete WIFI
  client.setServer(mqtt_server, 1883);                              // Connessione al server MQTT
  client.setCallback(callback);                                     // Configurazione dell'ascolto del topic in MQTT
}

void loop() {
  if (!client.connected()) {                                        // Gestione connessione WIFI
    reconnect();
  }
  client.loop();

  if(change)
    ChangePalettePeriodically();

  if(cylonFlag) cylon();
  else{ 
    static uint8_t startIndex = 0;
    startIndex = startIndex + 1; // motion speed
    
    FillLEDsFromPaletteColors( startIndex);
    
[... 192 lines cut ...]
}

void setup_wifi() {                                                 // Collegamento alla rete WIFI
  delay(10);
  //Serial.println();
  //Serial.print("Connecting to ");
  //Serial.println(ssid);
  digitalWrite(LED_BUILTIN, LOW);                                   // Accensione led per segnalare inizio connessione
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);                                       // We start by connecting to a WiFi network
  
  while (WiFi.status() != WL_CONNECTED) {
    for(int i=0;i<10;i++) {
      delay(50);
    }
    //Serial.print(".");
  }
  //Serial.println();
  //Serial.println("WiFi connected");
  //Serial.println("IP address: ");
  //Serial.println(WiFi.localIP());
  digitalWrite(LED_BUILTIN, HIGH);                                    // Spegnimento led per segnalare l'avvenuta connessione
}
```

heumann-a, 2021-07-22:
Got the same Problem with nearly identical setup. 
I am calling FastLED.show() every minute to update matrix LEDS, i divided 11 Strips with 11 LEDs each on 2 Pins ( one 55, other 66 LEDS). On both Strips the first LED after Data-In lights up in random color (blue and pink) 
forcing it to black doesnt work.

kriegsman, 2021-07-29:
Thanks for reporting this; all of the reports from ESP8266 are helpful to try and track this down and fix itt.

samguyer, 2021-07-29:
@pla10  and @heumann-a: our working theory right now is that changes to the 8266 core (probably the WiFi part) are causing new/more interrupts that disrupt the sensitive timing of the sending data to the pixels. What you're seeing is partial data sent to the pixel -- when we detect the timing problem, we abort and try again. Eventually, it works, but in the meantime the first pixel shows a lot of weird data. One thing we might try to add is a counter that you can access to determine if frames are getting dropped. But the interrupt problem is hard to fix.

pla10, 2021-07-29:
Thank you for the reply! Is there any known quick fix to this in the meantime?

SergeSkor, 2021-07-30:
I have the same problem on multiple projects. Before esp8266 core v3.0.0 never see such problem.
Are there any working fix?

samguyer, 2021-07-30:
@SergeSkor I think the only quick fix is to revert to a pre-3.0.0 version of the core.

SergeSkor, 2021-07-31:
> @SergeSkor I think the only quick fix is to revert to a pre-3.0.0 version of the core.

Yep, seems like...

pla10, 2021-08-03:
Reverting back to version 2.7.4 of esp8266 core solved the problem for me

ArtursGailis1995, 2021-08-09:
Same issue with 3.0.1 or 3.0.2 core and latest FastLED. Tried latest dev FastLED and it went into ful lred color and bootloop with stack trace. 

Downgrading to 2.7.4 core and 3.4.0 FastLED fixed it for me too.

Anatoliy464, 2021-08-13:
Try to add
#define FASTLED_INTERRUPT_RETRY_COUNT 0 
Before
#include <FastLED.h>

It fixed problem for me

SergeSkor, 2021-08-13:
#define FASTLED_INTERRUPT_RETRY_COUNT 0
does not work for me, still blinking on 3.0.X

phil31, 2021-10-03:
Hello,

on my custom ESP32 board, same issue ! first led is glitching.
using D4 or D5 pin on ESP32-WROOM-32UE 4MB and 9 x WS2812B.

example "FirstLight" show it ! even without any wifi connection or webserver running.

Arduino 1.8.13, FastLED 3.4.0
same issue using platformIO

Any advice please ?
regards

SergeSkor, 2021-10-04:
I did not play with ESP32, I have this problem with ESP8266.
I have solved this issue by rolling back to the older esp8266 core version.
2.7.4 is working for me, starting from 3.0.0 - is not.
Still looking for some better fix...

phil31, 2021-10-04:
yes i rerad that you used ESP8266, just saying that i notice this issue with my ESP32 board !
i don't know how to use previous version with platformIO

SergeSkor, 2021-10-05:
I am not using platformio, so cannot tell anything, sorry.  From: phil31Sent: Monday, October 4, 2021 5:27 PMTo: FastLED/FastLEDCc: SergeSkor; MentionSubject: Re: [FastLED/FastLED] First led of WS2812B is glitching with NodeMCU (#1260) yes i read that you used ESP8266, just saying that i notice this issue with my ESP32 board !i don't know how to use previous version with platformIO

bradanlane, 2021-11-08:
Thanks for opening this issue. I recently revisited an old project and the "first LED" problem appeared. Forcing to `platform = espressif8266@2.6.3` eliminated the problem.

robs113, 2021-11-08:
Thanks, I will do that.

 

Rob Stevens

kriegsman, 2021-11-08:
As of the recent commit https://github.com/FastLED/FastLED/commit/ccd2c7b426179ad50e4e2c0d3f132954fc6aa72c this issue may be improved when using the ESP8266 3.0.x core.  Please check out this new revision of the code, and let us know if it helps.

g3gant, 2021-12-03:
Got the same issue, using ESP8266 core 3.0.2, FastLED 3.4.0
#define FASTLED_INTERRUPT_RETRY_COUNT 0
#define FASTLED_ALLOW_INTERRUPTS 0
and   WiFi.setSleepMode(WIFI_NONE_SLEEP) fixes the problem until trying to connect to WiFi with static IP, if I'm using WiFi.Config(IP,GWY,MASK). First LED start blinking randomly.
Tried to downgrade core to 2.7.4, but got an error with FastLED:
"src/fastpin.h:210:2: error: static assertion failed: Invalid pin specified  static_assert(validpin(), "Invalid pin specified");

SergeSkor, 2021-12-03:
g3gant, not clear what pin it is complaining about.
I have no problems at all switching between 2.7.4 and 3.*.* without
changing in the code.

g3gant, 2021-12-03:
> g3gant, not clear what pin it is complaining about. I have no problems at all switching between 2.7.4 and 3.*.* without changing in the code.
> […](#)

Because ESP8266 core below 3.x.x does not support pin mapping like D1, D2 and so on. I had to change them to the numbers. And I had FastLED pin defined as D8, after downgrade I changed it to 15 according to ESP pinout.

SergeSkor, 2021-12-03:
g3gant, I am also using FastLED pin D8, but I believe it is defined in the
FastLED library, not in the board core.
P.s. But you mast to have define FASTLED_ESP8266_RAW_PIN_ORDER before
FastLed include...
--- 8< ---
#define FASTLED_ESP8266_RAW_PIN_ORDER
#include <FastLED.h>
FASTLED_USING_NAMESPACE
....
FastLED.addLeds<CHIPSET, *D8*, COLOR_ORDER>(leds,
NUM_LEDS).setCorrection(TypicalSMD5050);
--- 8< ---

g3gant, 2021-12-03:
> g3gant, I am also using FastLED pin D8, but I believe it is defined in the FastLED library, not in the board core. P.s. But you mast to have define FASTLED_ESP8266_RAW_PIN_ORDER before FastLed [...]

My bad, I had to mention that I had to change pins for other libraries I'm using. And I was missing this line "define FASTLED_ESP8266_RAW_PIN_ORDER", that solved it. Thanks a lot!
I hope the fix for 3.x.x core will be found soon :-)

perkinsinnovation, 2021-12-17:
I'm simply trying to flash my first led independently of the others, and it changes random colors. The other LED's seem to be fine, however, flashing LED's 2 and 3, sometimes change the color of 1.

I'm having the same problem with 3.2.0. I also tried 3.0.0 with no luck.

I wasn't able to go back to 2.6.3 since my build fails, and I really don't want to troubleshoot backward compatible issues.

I tried the suggestions you guys posted, but with no luck. Are there any other options? I'm just looking to flash three LED's reliably without color change.

Any help or suggestions would be greatly appreciated.

Thanks,
Shawn

Johboh, 2021-12-18:
Same here. Going back to 2.7.4 eliminates the flicker on led 0.

dKatsuro, 2021-12-26:
Was setting up this myself and I was using a ESP8266 12-E NodeMCU board.

It did work when using esp8266 2.7.4 (but not 3.0.2) with FastLed 3.4.0 and defined FASTLED_ESP8266_NODEMCU_PIN_ORDER with data pin set to 4 since I'm using D4. Of course, It also worked using RAW_PIN_ORDER with data pin set as 2.