FastLED/FastLED #1263: NO_CORRECTION define is used for two different purposes

ben-xo, 2021-07-25:
I noticed that NO_CORRECTION is used both [here](https://github.com/FastLED/FastLED/blob/5cc17b2be88982eb34b1998de22b83fee56d4f07/src/platforms/avr/clockless_trinket.h#L90) and [here](https://github.com/FastLED/FastLED/blob/5cc17b2be88982eb34b1998de22b83fee56d4f07/src/controller.h#L151) for different purposes.

The first one is to turn off clock-correction in AVR (which you might want to do if you have got interrupts working) and the second is for color-correction.

Given that this is a compatibility break I'd suggest renaming both variable (NO_COLOR_CORRECTION / NO_CLOCK_CORRECTION) and checking both old and new defines.

kriegsman, 2021-07-29:
Finding this is a good catch; thank you. 
I like the idea of checking both the new and old name.  I don't think a warning would be net-helpful; the people who need to separately control the two settings will figure it out, and the people who don't need to do that probably shouldn't be bothered with a new warning on their existing code.  But: I want to think about that a little bit, because in general I do like giving informative warnings to people who might will benefit from seeing them.

ben-xo, 2021-07-29:
@kriegsman is there a project chat / IRC / Slack / whatever? I know the reddit, but that doesn't seem like a chat venue for project discussion

ben-xo, 2021-08-01:
Raised a PR for this. #1271

ben-xo, 2022-01-13:
Closed as my PR was merged

closed 2022-01-13 by ben-xo