FastLED/FastLED #1264: ESP8266, core 3.x.x and hundreds of warnings

sblantipodi, 2021-07-27:
Hi all,
I have seen that this issue #1237 has been closed but there are hundred of warning at compile time.

I would not consider the support "done" untile all those warnings are fixed, aren't you?

I attach a file with the warnings because there are too many warnings for this form :)

sblantipodi, 2021-07-27:
[warning.txt](https://github.com/FastLED/FastLED/files/6883688/warning.txt)

sblantipodi, 2021-07-27:
this issue can't be considered closed, there are still hundred of warnings at compile time and strip still not work correctly (WS2812B here).

I noticed that a simple 

```
for (int i = 0; i < NUM_LEDS; i++) {
    leds[i].red = 255;
    leds[i].green = 255;
    leds[i].blue = 255;
  }
  FastLED.show();
```

does not work 90% of the time...

the strange things is that if I add two   FastLED.show(); call it works 100% of the time but it's not a solution clearly.

this works:
```
for (int i = 0; i < NUM_LEDS; i++) {
    leds[i].red = 255;
    leds[i].green = 255;
    leds[i].blue = 255;
  }
  FastLED.show();
  FastLED.show();
```

kriegsman, 2021-07-29:
I'll take a look at dealing with the 'register' warnings; thanks.

samguyer, 2021-07-29:
@sblantipodi Can you tell me more about what's going wrong? Are the frames getting aborted by interrupts? Or is the problem more about latching the LEDs? Does adding a delay after (or before) the call to show() change the behavior.
Thank you for helping us fix these issues!

sascha432, 2021-07-29:
@samguyer
The NMI that causes the glitching occurs every 100ms, at least if WiFi is not too busy. Not sure if I mentioned that before.

 @sblantipodi 
What happens if you use only half of the LEDs in your strip? Does it turn on some of the unused ones?
That would indicate that the LEDs do not detect the reset signal between frames.

sblantipodi, 2021-07-29:
> @sblantipodi Can you tell me more about what's going wrong? Are the frames getting aborted by interrupts? Or is the problem more about latching the LEDs? Does adding a delay after (or before) the call to show() change the behavior.
> Thank you for helping us fix these issues!
> 

If I try to run an animation it seems to work ok, don't know if it loose some frames.

If I try to simply turn on and off the strip with a solid colour like all white, all red, all green and so on, first led turn on and off ok, all the other LEDs remains of the previous colors.

Delay before and after does not help at all.

I don't know how to see if the LEDs are aborted by interrupts, on the console I see no errors at runtime.

Hey, thank you for your work here! :)

sblantipodi, 2021-07-29:
> @samguyer
> The NMI that causes the glitching occurs every 100ms, at least if WiFi is not too busy. Not sure if I mentioned that before.
>
>  @sblantipodi
> What happens if you use only half of the LEDs [...]

It seems that only the first led get the input, all the others not.

scaytrase, 2021-09-22:
For me new core makes first led messy (and a lot of warnings, yes), but others work fine. Rollng back to 2.6 core fixes it.

kriegsman, 2021-11-01:
Just posting here to let folks know that @samguyer and I are looking at the ESP 8266 issue some more right now.

I just pulled down the ESP8266 3.0.2 core, and "DemoReel100" is working okay on a 30-pixel strip.  We're investigating.

kriegsman, 2021-11-02:
As of https://github.com/FastLED/FastLED/commit/ccd2c7b426179ad50e4e2c0d3f132954fc6aa72c, the problem with the ESP8266 core 3.0.x may be fixed; please give it a try and let us know if you can.

mspoorendonk, 2021-11-09:
@sblantipodi  Regarding the register warnings: Adding "-Wno-register" to your build_flags suppresses them for the moment until the issue is resolved.

DrJaymz, 2021-11-18:
The only solution that seems to work is   
platform = espressif8266@2.6.3

This may not be the place to ask but I am uncertain in platform IO how to specify the specific commit https://github.com/FastLED/FastLED/commit/ccd2c7b426179ad50e4e2c0d3f132954fc6aa72c

I feel a bit stupid as I have been using FastLed for a good 5 years on many projects but it was only recently when the core updated it all broke and I have been around the houses trying to fix it, even replacing the LEDs.

DrJaymz, 2021-11-18:
> As of [ccd2c7b](https://github.com/FastLED/FastLED/commit/ccd2c7b426179ad50e4e2c0d3f132954fc6aa72c), the problem with the ESP8266 core 3.0.x may be fixed; please give it a try and let us know if you can.

I am having issues and I think I have the latest build, when I specify core  2.6.3 problems go away.

poldim, 2022-03-04:
@kriegsman @focalintent 

any progress on this?  per the mentions above it seems to work on framework 2.7.4 but not newer

Guenni75, 2023-03-10:
FastLED 3.5.0 and still the same problems with ESP8266.
Can't do anything with them.

m4rc77, 2023-04-07:
If you are using [PlatformIO](https://platformio.org/) you can add in your `platformio.ini` file the following lines, in order to "silence" the warnings from FastLED until a patch is available:

```
; fix in order to get rid of 'warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]' from FastLED Library <= 3.5.0
; fix in order to get rid of 'warning: this 'while' clause does not guard... [-Wmisleading-indentation]' from FastLED Library <= 3.5.0
build_flags =
    -Wno-register
    -Wno-misleading-indentation
```
**But be warned, this will also hide similar issues in your own code!**

Barcley1990, 2023-04-16:
Same problem here: ESP8266 WROOM-02 module.

sascha432, 2023-04-17:
For the warnings you can write your own header file "FastLED_no_warnings.h" and disable warnings inside the file. This is just from the GCC documentation, I did not test it. You won't include FastLED.h anymore, just your file.

```
#pragma once
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wmisleading-indentation"
#pragma GCC diagnostic ignored "-Wregister"
#ifndef FASTLED_INTERNAL
#define FASTLED_INTERNAL
#endif
#include <FastLED.h>
#pragma GCC diagnostic pop
```

When it comes to flickering or repeating patterns at the end, try this:

- Use less LEDs and/or more output pins
- Increase the speed to 160MHz
- Make sure your MCU has enough idle time to handle WiFi. The NMI (Non Maskable Interrupt) is what is interrupting FastLED and causing flicking or wrong colors towards the end of the string
- The new WS2812B require more delay between frames, otherwise the stop signal is ignored. New datasheets (i.e. 2813B) have a note about this, but they never updated the WS2812B sheet. The stop time was increased from 50 to 150ns, but I have some strings that require 300ns to work probably especially when they get hot
- Check if your string voltage is stable at the end. If issues occur with high brightness, this could indicate a voltage drop
- Check for bad contacts if a single LED keeps making trouble
- Make sure they do not get too hot. Once overheated they tend to cause flickering or wrong colors
- Check if your power supply can handle the load

If your first pixel flickers, try following:

- Use an inline resistor (easiest way is a 10K poti to find out what resistance you need)
- Make sure you have a **fast** level shifter (requires more inline resistance due to ringing with long wires)
- Use short cables to the first LED and make sure the data line is in between GND and 5V
- Inject 5V into the string away from the data line
- Try a different first LED, some just flicker and you cannot do anything about it
- For long wires, put one LED every 30-50cm to improve the signal (+inline resistor if required, the datasheet suggests 300R for 50cm)
- Make sure you have a good GND connection from the string to the MCU if you use multiple power supplies

I debugged those issues for quite some time and mostly it is a distorted signal caused by cables/level shifters and interference from power supplies, single LEDs out of spec(or damaged) and the different behavior of the 3.x framework.

MarcBresson, 2026-10-11:
should nt this be closed by [#2351](https://github.com/FastLED/FastLED/pull/2351) ?