FastLED/FastLED #1273: Apollo3 support no longer working with board software 2.x

rbrome, 2021-08-02:
When I updated to the new 2.1.1 core software for SparkFun's Artemis Apollo3 boards, it broke FastLED support. Now I get the error "Unrecognised APOLLO3 board" when I try to compile. 

I'm guessing the new core software changed the board names that FastLED looks for. So "ARDUINO_AM_AP3_SFE_BB_ARTEMIS_NANO" is now something else... I'm just not sure what. 

(I would simply downgrade, but I need the new core software for its Bluetooth support.)

EthanCSmith102, 2021-08-25:
I ran into this problem today and you are correct, Sparkfun changed the name of the boards in the boards.txt located (for me) in C:\Users\[yourcomputername]\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.1.1\boards.txt
I fixed my board by changing the name FastLED looks for in fastpin_apollo.h which is located (for me) in C:\Users\[yourcomputername]\Documents\Arduino\libraries\FastLED\src\platforms\apollo3\fastpin_apollo.h
For my board, the Artemis Nano, I went down until I found the name ~line 89 and changed the ARDUINO_AM_AP3_SFE_BB_ARTEMIS_NANO to the new ARDUINO_APOLLO3_SFE_ARTEMIS_NANO. Notice the prefix ARDUINO_ stayed the same and I just replaced "AM_AP3_SFE_BB_ARTEMIS_NANO" with the new "APOLLO3_SFE_ARTEMIS_NANO" which can be found in the 2.1.1 boards.txt. Hope this helps at least getting your board working again until they can release a proper fix.

rbrome, 2021-08-25:
Excellent detective work. Those are the strings I couldn't find. Thank you!

rbrome, 2024-12-18:
This keeps coming up every time I update FastLED and it's annoying. Can someone working on FastLED please update fastpin_apollo3.h to accept both old and new board names? (This is a really easy bug fix!) 

For example, the line that's currently: 

`#elif defined(ARDUINO_AM_AP3_SFE_BB_ARTEMIS_NANO)`

...should (I believe) be changed to: 

`#elif defined(ARDUINO_AM_AP3_SFE_BB_ARTEMIS_NANO) || defined(ARDUINO_APOLLO3_SFE_ARTEMIS_NANO)`

I imagine other Artemis board names changed as well.

zackees, 2024-12-19:
Can you help test this? I don't have a platformio config for this board. I you can provide this then we can make sure it works from here on out for you, by putting it under test.

closed 2024-12-19 by zackees

rbrome, 2024-12-19:
I am happy to help. Physically, I only have a Nano, not the other boards. I've attached the boards.txt file that contains all of the newer Artemis/Apollo3 board names. 
[boards.txt](https://github.com/user-attachments/files/18191572/boards.txt)

zackees, 2024-12-19:
Can you convert this to a platformio.ini?

That's all we need.

rbrome, 2024-12-19:
I don't know what that is.

zackees, 2024-12-19:
You don't know what know a platformio.ini is.

Ok. No worroes.

Mick-Hoshi, 2024-12-20:
I'm using Arduino IDE v2.3.4 and M5Atom Matrix.
It seems to be   FastLED  3.9.6 has a bug.
I tested the button.ino sketch for the M5 Atom Matrix.
The button.ino sketch is included in the M5Atom library.
As a result, LED0 remains permanently ON with a green color.
The issue does not occur when using FastLED version 3.9.5, where it works as expected.

zackees, 2024-12-20:
Try both the following define before including FastLED.h

#define FASTLED_RMT5_RECYCLE=0

Then alternatively:

#define FASTLED_RMT5_RECYCLE=1

And see if one or the other addresses the issue.

zackees, 2024-12-20:
Try out the new 3.9.7 that was released. It should appear shortly in the arduion ide.