FastLED/FastLED #1275: Precision update timing?

CubicEarth, 2021-08-08:
I am using FastLED.show() to update WS2812B leds. I am looking to have my control over the moment the leds update to the new state be as precise as possible.

My understanding is that FastLED.show() spends as long as needs to push the bits defining the future state to each led, and then when that process has completed, it sends a signal causing all leds to update at the same moment.

Is currently a way to separate these two processes? I would like to be able to control when the 'update' signal is sent, and not have it occur just whenever all the leds have finished receiving their instructions.

somewhatlurker, 2021-08-12:
WS2812B and other similar "clockless" LED chipsets actually don't have a specific update signal. Instead, what they do is latch data and display it a certain time after the last LED's data is sent. (often quoted as 50 microseconds, though it can vary)
As such, to toggle LEDs at a more accurate time you'd have to calculate or measure how long updating the LEDs takes and simply start the process earlier.
(It would also technically be feasible to indefinitely send extra bits to non-existent additional LEDs and simply stop sending them when the precise update is needed, but I think such a feature would be quite awkward to add into FastLED.)

And one other thing to consider is that even after precisely controlling the time when you finish sending data, WS2812B LEDs won't have perfectly consistent latch timing. (due to how the PWM circuitry operates there can be ~2ms of jitter -- link for reference: https://wp.josh.com/2014/05/15/getting-physical-to-uncover-neopixel-pwm-secrets/)
So the actual practical benefit of such a feature may be quite limited.

LEDs with SPI-like protocol (such as APA102 and SK9822, which are both supported by FastLED) are commonly used as a faster alternative and avoid many problems like this. They allow data to be sent at 12 megahertz (or faster!), which is fast enough to write to hundreds of LEDs in less than a millisecond, and often internally operate much faster than WS2812B to help avoid the jitter issue too.

I think it makes a lot of sense to switch to the more precise LEDs rather than trying to work around hardware limitations in code, but I'm curious to hear back from you and also see if the maintainers of the library have any opinions.

CubicEarth, 2021-08-13:
Thanks for the great explanations, and the mention of the APA102 and SK9822.

I agree one way of controlling when they update is to control when they start, but that requires anticipation of the ideal moment, which isn't always known advance. One situation would trying to update the LEDs in response to a trigger signal, say a button push, where it could be nice to have the LEDs update as quickly as possible. Understood that 2ms is a lower limit.

I have no idea how complicated the workaround you mention would be to implement, or if enough people would find it useful for it to be worth it.

somewhatlurker, 2021-08-15:
Implementing such a workaround as part of FastLED looks to me like it would probably require editing the output controllers for all supported microcontrollers to enter an endless loop until a certain pin is toggled (plus associated API changes) -- not impossible to do, but perhaps tedious and not something I would deem worth undertaking. (Especially so when considered that by using parallel output on platforms that support it, you could already update ~500 LEDs in less time than the jitter would skew results by.)

It's probably actually possible to do such a thing without FastLED supporting it anyway, if you're using a fast microcontroller (needs to be relatively fast so that the data line isn't left idle for long enough to trigger a reset before you have a chance to start sending your own data). After calling show, you'd need to use your own code to output an appropriate waveform for extra bits until the update is desired (~25% duty cycle at 800kHz should work). How difficult this is would depend on your own level of experience and the platform you're using (it looks surprisingly simple to do with Teensy's PWM features, but not all platforms are anywhere near as straightforward).

ben-xo, 2021-08-21:
Unfortunately not really. Ideally you'd call FastLED.show() exactly when you want. Doing what you want us kind of possible in the theory but would involve modifying FastLED to give you control of the latch delay.

henrygab, 2021-09-02:
It would also be more difficult on some controllers, such as nRF52840, because of the inability to prevent wireless interrupt handling. On nRF52840, the entire transfer needs to be setup beforehand and offloaded to hardware peripheral for DMA transfer, to achieve stable output.  Not easily modified to end with infinite extra data waiting on a pin toggle.

Clockess LEDs are probably not your best option here....