FastLED/FastLED #1279: SK6812 Support

MythicalForce, 2021-08-17:
So i just bought some SK6812 and was REALY hoping i could use FastLED as i found this in the example
 
[ https://github.com/FastLED/FastLED/blob/5cc17b2be88982eb34b1998de22b83fee56d4f07/examples/Blink/Blink.ino#L34](URL)

but it seems that support for the SK6812 is not in FastLED yet! i also found this "hack" -> [https://www.partsnotincluded.com/fastled-rgbw-neopixels-sk6812/](url)

but as my project uses alot of the FastLED stuff it is just to much to rewrite :/ this "hack" only covers the basic to get it to work like showing solid colors!

how far along is the support for SK6812 in FastLED? is there a test branch i can use or?

ATLTVHEAD, 2021-08-20:
I believe the Sk6812's are just clones of the WS2812Bs/neopixels. I could have sworn sk6812 was also an option for the ledtype. If not try the ws2812 type or neopixel. I've drive sk6812s with Fastled, just cannot remember the exact led I specified in code.

MythicalForce, 2021-08-21:
> I believe the Sk6812's are just clones of the WS2812Bs/neopixels. I could have sworn sk6812 was also an option for the ledtype. If not try the ws2812 type or neopixel. I've drive sk6812s with Fastled, just cannot remember the exact led I specified in code.

i tried both WS2812B and SK6812 as types but did not work :S

MythicalForce, 2021-08-21:
i should clearify that i use the SK2812 RGBW !

somewhatlurker, 2021-08-22:
Indeed, SK6812 chipset is supported for RGB, but there's no support for RGBW LEDs.
It's actually quite an old issue that unfortunately never really got a proper solution implemented (#106).

MythicalForce, 2021-08-23:
> Indeed, SK6812 chipset is supported for RGB, but there's no support for RGBW LEDs.
> It's actually quite an old issue that unfortunately never really got a proper solution implemented (#106).

this -> https://github.com/EmteZogaf/FastLED/tree/dma claims to be a FastLED fork with RGBW support but i can't get it to work :S

in the readme all it says is 
```
To use this, add this define somewhere above where you #include <FastLED.h>:
#define FASTLED_RGBW
```

isocorDev, 2021-11-03:
> but it seems that support for the SK6812 is not in FastLED yet! i also found this "hack" -> [https://www.partsnotincluded.com/fastled-rgbw-neopixels-sk6812/](url)

This hack works pretty well in that you can drive the white channel of SK6812 RGBW strips. I used this hack along with the theory presented in the DMA fork you linked to in a follow up comment. The idea is to have two led buffers for your strip: RGB(where you complete all of your animations and use FastLED to the fullest like you are accustomed) and RGBW(where you simply convert the RGB into RGBW by removing the maximum shared value from RGB and use it as Ws value). It costs more memory but the colors you can achieve are stunning. Plus, your power consumption will go down 👍

MythicalForce, 2021-11-03:
> > but it seems that support for the SK6812 is not in FastLED yet! i also found this "hack" -> [https://www.partsnotincluded.com/fastled-rgbw-neopixels-sk6812/](url)
> 
> This hack works pretty well in that you can drive the white channel of SK6812 RGBW strips. I used this hack along with the theory presented in the DMA fork you linked to in a follow up comment. The idea is to have two led buffers for your strip: RGB(where you complete all of your animations and use FastLED to the fullest like you are accustomed) and RGBW(where you simply convert the RGB into RGBW by removing the maximum shared value from RGB and use it as Ws value and then drive the LEDs). It costs more memory but the colors you can achieve are stunning. Plus, your power consumption will go down 👍

i tried to get it to work but failed so i stopped trying :/

sofakng, 2021-12-04:
Has there been any updates to this?  I'm looking to use SK6812 RGBW LEDs...

MythicalForce, 2021-12-04:
> Has there been any updates to this? I'm looking to use SK6812 RGBW LEDs...

still nothing im afraid :(

semka95, 2021-12-13:
> this -> https://github.com/EmteZogaf/FastLED/tree/dma claims to be a FastLED fork with RGBW support but i can't get it to work :S
>
> in the readme all it says is
>
> ```
> To use this, add this define [...]

I tried this fork, and it's working, but not like author claims. For some reason library seem to not distinguish white and rgb leds. For example, I am setting all 60 leds to `CRGB(255, 255, 255)`, it suppose to turn on only white leds, but in reality half of leds emit light, rgb and white at the same time. Also, first led periodically blinking.

toine512, 2021-12-22:
Hello

I added white channel support retaining compatibility with existing code (thank you C++): [https://github.com/toine512/FastLED-W](https://github.com/toine512/FastLED-W) However **only ESP32 RMT is supported**.
Just use SK6812W controller. Colours are defined by 0xWWRRGGBB and there's a CRGB::WhiteLight constant.
FASTLED_RMT_BUILTIN_DRIVER doesn't work since 3.4 and I'm not tooled to debug it.

I am just starting to test this with a real strip so keep in mind that things can be broken. Currently there is no white channel interpolation like WLED does. You're juste given the white channel to control.
I hope someone can have fun with that.

sofakng, 2021-12-22:
I really appreciate the effort being made to make this library compatible with RGBW.  I'll try to perform some tests but the primary reason I was interested in compatibility was for other projects (ie. SignalRGB) that requires FastLED.

toine512, 2021-12-23:
Well that's the point: it is supposed to be a drop-in replacement. I use it with FastFX.
(SignalRGB looks like an amazing promise... 🤩)

zark0-UwU, 2022-01-24:
> So i just bought some SK6812 and was REALY hoping i could use FastLED as i found this in the example
>
> [ [...]

Hi!, I have seen many opened issues as 'lack of support for SK6812' as it turns out it is a clone of a supported chipset, I think it should be clarified on the docs, and maybe even make a binding, so selecting SK6812 would do the same as WS2812, which is the original version, from where the SK6812 was cloned.

yesyesuk, 2022-01-24:
When we say "no SK6812 support" we refer to the RGBW version, not RGB.

MrBeepoint, 2022-06-26:
May I ask for a status update here? Is there a plan to create a pull request to improve the FastLED main library? Is there support
 by the owners of FastLED to do so? I have a sk6812 stripe here and would like to support if possible.

sammcj, 2023-01-22:
Just wondering if there's any update on supporting SK6812 / RGBW strips? They've fast become the most common LED strips in Australia and while the basics of them work with FastLED the white LED control seems to be lacking.

[image]

polykoma, 2023-05-01:
Leaving a +1 here. I bought some SK6812 RBGNW since I found this "hack" for the fastled library. But I want to drive my RGBNW with Adalight protocol to use with Ambilight and Prismatik with a teensy 4.1.
But seems like this is preparred to fail.

Guess I need to order some APA102 then. Sadpepe.

I really thought there MUST be RGBW support for ambilight projects in 2023. But guess Im wrong.

polykoma, 2023-05-01:
> > Has there been any updates to this? I'm looking to use SK6812 RGBW LEDs...
> 
> still nothing im afraid :(

I guess no update from your end aswell? <3

sammcj, 2023-05-01:
Happy to help with testing if needed. I've got a couple of these sitting around not being very useful if anyone has a PR they want me to try?

polykoma, 2023-05-01:
> Happy to help with testing if needed. I've got a couple of these sitting around not being very useful if anyone has a PR they want me to try?

Hey there sammcj, wow thats very kind, tysvm. So for my usecase I try to use a teensy 4.1 + 120 SK6812RGBNW + Adalight Sketch (needs FastLed in Libraries installed) + SK6812RGBW "hack"

Steps would be like:
1) Configure and Flash the Ledstream_FastLed Sketch from
https://github.com/dmadison/Adalight-FastLED/

but include the "hack" into it

Possible hack :
https://repos.ctdo.de/interfisch/Adalight-FastLED_rgbwMod

Possible hack 2:
https://www.partsnotincluded.com/fastled-rgbw-neopixels-sk6812/

Possible hack 3:
https://gist.github.com/tuxuser/e817d4968655a14a3c8671ebb1bb4944

(not) Possible Hack 4:
There is a third hack but its ESP8266 only.

Im not sure if those work together with Adalights  LEDSTREAM_FASTLED Sketch (The base for many Ambilight projects)

The Software to drive the ambilight would be classic PRISMATIK or AMBIBOX.

Possible controllers could be Arduino Uno, Pro Mini, Clones, Teensys .
Im currently running 200 LPD8806 per screen (3 screen setup) on an Arduino Uno. But wanting to switch to either RGBW LED or APA102 if RGBW is not possible. As well as use a teensy 4.1 cuz UNO a lil slow.

Thank you so much and BR

marmilicious, 2023-05-02:
> If you use a teensy above 3.2 you might need to limit CPU to 120MHZ or you could end up with flickering issues.
> Its reported that too much cpu power can glitch the leds

Interesting. I wonder if adding something like a few millisecond delay after FastLED.show() or putting FastLED.show() inside an EVERY_N timer so it limits how quickly it's called might help? Seems a shame to limit CPU if one wanted to be using it for something else.

zackees, 2023-05-02:
The big issue here is that FastLED uses the RGB format everywhere. Adding RGBW to the core library would be a major change.

However, as mentioned in other issues, there exists a formula for converting RGB -> RGBW using maths. Meaning this conversion could happen at the driver level as the LED is being written out, meaning the entire library could stay RGB, minimizing changes.

It would be really nice if someone smarter than myself could figure out what the math is for RGB -> RGBW conversion, preferably without having to drop into floating point math. For example, an interpolation table would be sufficient if we knew what the control points would be. Then integrating this change would be straightforward.

zackees, 2023-05-02:
Chat GPT 4.0 says this, but I'm not sure it's correct:

```
To convert an RGB color (with 8 bits per channel) to an RGBW color (also with 8 bits per channel), you can follow these steps:

Calculate the white component (W) by extracting the minimum value of the three RGB components.
Subtract the minimum value from each of the three RGB components to obtain the remaining color components.
Normalize the resulting RGB components to fit within an 8-bit range (0 to 255).
Here's the formula for the conversion:

W = min(R, G, B)
R' = R - W
G' = G - W
B' = B - W
The resulting RGBW color is represented as (R', G', B', W), with each component being an 8-bit value.

Please note that this formula assumes that the input RGB values are also in the 8-bit range (0 to 255). If your input values are in a different range, you'll need to scale them accordingly before applying the formula.
```

toine512, 2023-05-02:
Hi.
Having a fourth channel in the core is trivial.
Adding a white channel interpolation (as WLED does) to get better colours wouldn't be that bad.
The ridiculous amount of work, if uncoordinated, is updating each and every controller to send 3 or 4 bytes par LED.

isocorDev, 2023-05-02:
This is correct, but I have never seen min() with three arguments.

zackees, 2023-05-02:
> Hi. Having a fourth channel in the core is trivial. Adding a white channel interpolation (as WLED does) to get better colours wouldn't be that bad. The ridiculous amount of work, if uncoordinated, is updating each and every controller to send 3 or 4 bytes per LED.

Would we have to support assembly for the bit-banging? Most chips now are sufficiently fast enough to do it all in software.

toine512, 2023-05-02:
I don't believe dropping support for old chips (that's what you seem to imply) is a good idea as even the venerable ATmega 328P just got the NRND status.
That said, I have only read a few drivers (ESP) so I don't know. Moreover bit-banging is software so I don't get the remark.

zackees, 2023-05-02:
> Moreover bit-banging is software so I don't get the remark.

I looked up the term. You might be right. I assumed non-bit banging meant dedicated hardware but ASM might apply too.

zackees, 2023-05-02:
> I don't believe dropping support for old chips

If the algorithm is fast enough, such as an integer conversion that's "good enough", dropping support for old micros might not be necessary.

toine512, 2023-05-02:
The whole point of FastLED is that it's fast on AVR.

zackees, 2023-05-02:
I just took a shower and realized something very important: by thinking about the problem slightly differently the math becomes much easier.

Instead of thinking of white as "white", let's instead think about it as three leds mixing together to *form* white.

In this case, W component can be expressed as = (1/3r, 1/3g, 1/3b), but let's call this (r2, g2, b2). Let's call the RGB in the prefix of RGBW (r1, g1, b1).

Then the total perceived value of each of the components (r, g, b) can be expressed as:

(r, g, b) = (r1 + r2, g1 + g2, b1 + b2) = (r1, g1, b1, r2+g2+b2)

For simplicity, I'll assume there is only one type of W component, so this will discount warm vs cool white, but this can be added later.

Let's look at an example. RGB = (85, 170, 255) aka (1/3 * 255, 2/3 * 255, 3/3 * 255), we want to convert this to RGBW, so:

```
r1 = 0, r2 = 255 / 3
  => so r1 + r2 = 85 = r (checks!)
g1 = 85, g2 = 255 / 3
  => so g1+g2 = g = 170 (checks!)
 
 b1 = 170, b2 = 255 / 3
  => so b1+b2 = b 255 (checks!)
```

So in this case RGB(85, 170, 255) = RGBW(r1, g1, b1, r2+g2+b2) = (0, 85, 170, 255)

I don't have the exact formula for this, but it would be something like this:

Select the min color component. Multiply it by three. Let's assume it's always <= 85 for simplicity, and let's assume that it's r.

In this case, W = (r*3) and then RGB = (r-r, g-r, b-r)

What happens if the minimum component is > 85? I haven't figured that out yet...

zackees, 2023-05-02:
I asked ChatGPT4 to go ahead and implement this as an algorithm and it fed me this:

```C++
#include <cstdint>
#include <tuple>

using uint8_t = std::uint8_t;
using RGB = std::tuple<uint8_t, uint8_t, uint8_t>;
using RGBW = std::tuple<uint8_t, uint8_t, uint8_t, uint8_t>;

RGBW rgb_to_rgbw(const RGB& rgb) {
    uint8_t r, g, b;
    std::tie(r, g, b) = rgb;

    uint8_t min_component = std::min({r, g, b});
    uint8_t w;

    if (min_component <= 84) {
        w = 3 * min_component;
    } else {
        w = 255 - min_component / 3;
    }

    uint8_t r_prime = r - min_component;
    uint8_t g_prime = g - min_component;
    uint8_t b_prime = b - min_component;

    return std::make_tuple(r_prime, g_prime, b_prime, w);
}
```

And a unit test:

```C++
#define CATCH_CONFIG_MAIN
#include "catch.hpp"
#include <cstdint>
#include <tuple>

using uint8_t = std::uint8_t;
using RGB = std::tuple<uint8_t, uint8_t, uint8_t>;
using RGBW = std::tuple<uint8_t, uint8_t, uint8_t, uint8_t>;

RGBW rgb_to_rgbw(const RGB& rgb) {
    // ... (rgb_to_rgbw implementation as provided earlier)
}

TEST_CASE("RGB to RGBW conversion", "[rgb_to_rgbw]") {
    SECTION("Example 1: (85, 170, 255) -> (0, 85, 170, 255)") {
        RGB input(85, 170, 255);
        RGBW expected_output(0, 85, 170, 255);
        REQUIRE(rgb_to_rgbw(input) == expected_output);
    }

    SECTION("Example 2: (255, 255, 255) -> (170, 170, 170, 255)") {
        RGB input(255, 255, 255);
        RGBW expected_output(170, 170, 170, 255);
        REQUIRE(rgb_to_rgbw(input) == expected_output);
    }

    SECTION("Example 3: (254, 255, 255) -> (170, 171, 171, 252)") {
        RGB input(254, 255, 255);
        RGBW expected_output(170, 171, 171, 252);
        REQUIRE(rgb_to_rgbw(input) == expected_output);
    }
}
```

toine512, 2023-05-02:
I'm done talking with the mechanical turk, please someone unplug it.

zackees, 2023-05-02:
Hey at least someone is trying to solve this very hard problem that's been outstanding for over a year.

polykoma, 2023-05-02:
Wow Im stunend by so many answer! <3
And honestly I'm in hope for solution since over a decade when I build my very first ambilight. Sadly Im no programmer myself. Im very happy to at least like make the discussion flame up again with my post. And as said Im baffled there hasnt been anyone challenging this in all those past years.

From my POV the solution zackee suggested also was on my mind at some point. Basically just define value ranges where the RGB would usually output white due to nature how RGB works and then tell the RGB to run on low and the white to turn on.
To overly simplify it. I cant tell if that would be too much calculation for small chipsets tho.

Anyway - I thank you all so much for thinking about this really ancient and never solved issue. <3
Maybe one day there will be a solution. Maybe chatGPT 4 can play a role in it too.

However - whoever might solve this and get RGBW support for FastLed and therefor also any Ambilight project: 
Give me a ping and I will make sure to get a lunch and a coffee over to your desk <3

yesyesuk, 2023-05-02:
> However, as mentioned in other issues, there exists a formula for converting RGB -> RGBW using maths. Meaning this conversion could happen at the driver level as the LED is being written out, meaning the entire library could stay RGB, minimizing changes.

For me personally that is going in the wrong direction. I would much prefer to be able to control RGB and W independently, rather than calculating W from RGB.

zackees, 2023-05-02:
> For me personally that is going in the wrong direction. I would much prefer to be able to control RGB and W independently, rather than calculating W from RGB.

That requires a rewrite of fastled. Not going to happen.

polykoma, 2023-05-02:
> For me personally that is going in the wrong direction. I would much prefer to be able to control RGB and W independently, rather than calculating W from RGB.

I mean I agree with you but essentially yeah, that requires a complete rewrite according to this issue thread and the other 2 over the past years. So that narrows it down to a "dirty" solution.

zackees, 2023-05-02:
Almost there...

```C++
/******************************************************************************

                              Online C++ Compiler.
               Code, Compile, Run and Debug C++ program online.
Write your code in this editor and press "Run" button to compile and execute it.

*******************************************************************************/

#include <iostream>

using namespace std;

#include <cstdint>
#include <tuple>

struct RGB {
    uint8_t r;
    uint8_t g;
    uint8_t b;
};

struct RGBW {
    uint8_t r;
    uint8_t g;
    uint8_t b;
    uint8_t w;
};

uint8_t mymin(uint8_t a, uint8_t b, uint8_t c) {
    return std::min(a, min(b, c));
}

inline RGBW rgb_to_rgbw(const RGB rgb) {
    uint8_t min_component = mymin(rgb.r, rgb.g, rgb.b);
    uint8_t w;
    bool is_min = true;
    if (min_component <= 84) {
        w = 3 * min_component;
    } else {
        w = 255;
        is_min = false;
    }
    uint8_t r_prime;
    uint8_t g_prime;
    uint8_t b_prime;
    if (is_min) {
        r_prime = rgb.r - min_component;
        g_prime = rgb.g - min_component;
        b_prime = rgb.b - min_component;
    }
    else {
        uint8_t w3 = w / 3;
        r_prime = rgb.r - w3;
        g_prime = rgb.g - w3;
        b_prime = rgb.b - w3;
    }
    // return std::make_tuple(r_prime, g_prime, b_prime, w);
    RGBW out = { r_prime, g_prime, b_prime, w };
    return out;
}

void test_rgb_to_rgbw(const RGB& input) {
    RGBW output = rgb_to_rgbw(input);
    cout << "Input " << int(input.r) << ", " << int(input.g) << ", " << int(input.b) << "\n";
    cout << "output " << int(output.r) << ", " << int(output.g) << ", " << int(output.b) << ", " << int(output.w) << "\n";
    cout << "\n";
}

int main()
{
    test_rgb_to_rgbw(RGB({ 85, 170, 255 }));
    test_rgb_to_rgbw(RGB({ 255, 255, 255 }));
    test_rgb_to_rgbw(RGB({ 254, 255, 255 }));
    return 0;
}
```

zackees, 2023-05-03:
Okay, the algorithm is done.

The attached unit tests show that the mapping back from RGBW -> RGB has a 0% error. I expect that this is off a little bit due to int truncation (I used floating point to at least calculate the error values, though the RGB is in uint8_t format).

Therefore I expect the real color error to be < 0.4%, given that 0.4% ~= 1/255.f. Additionally, my algorithm makes the following assumptions:

  * The white color component brightness is equal to the brightness of the other components.
  * The white color component is perfectly neutral and could be represented by 3 color component LEDs mixing together.
  
  I've also provided an escape hatch so the user can override my default implementation by using a C++ template trick. If the user defines rgb_2_rgbw() in their source code then the provided example will bind to that instead of this default implementation.
  
```C++

/////////////////////////// rgb_2_rgbw ///////////////////////////

// This is a working algorithm of the rgb_2_rgbw() conversion.

#include <assert.h>
#include <cstdint>
#include <iomanip>
#include <iostream>
#include <random>
#include <tuple>

using namespace std;

struct RGB {
    uint8_t r;
    uint8_t g;
    uint8_t b;
};

struct RGBW {
    uint8_t r;
    uint8_t g;
    uint8_t b;
    uint8_t w;
};

inline uint8_t min3(uint8_t a, uint8_t b, uint8_t c) {
    return std::min(a, min(b, c));
}

// Note that the unused template parameter here is used purely so that the
// user can define their own non-template rgb_2_rgbw() function and have it
// override this function. This works because the C++ compiler will always
// prefer non-template functions to template functions. In this case, the compiler
// will first attempt to bind to a non-template rgb_2_rgbw() function and if it
// does not find it, will default to the one specified here. This is similar to
// the __attribute__((weak)) linker directive.
//
// This function assumes that the white component can be represented by a three
// color LED (rgb) mixing together to produce neutral white. It's also assumed that
// the power output of the white color component is equal to that of one color
// component led.
//
// For RGBW strips that have non-neutral white color components, or white components
// which exceed or are less than the brightness of other color components then this
// function may produce colors that are slightly off.
template <int WHITE_TEMPERATURE = 4000>
inline RGBW rgb_2_rgbw(const RGB rgb) {
    uint8_t min_component = min3(rgb.r, rgb.g, rgb.b);
    uint8_t w;
    bool is_min = true;
    if (min_component <= 84) {
        w = 3 * min_component;
    } else {
        w = 255;
        is_min = false;
    }
    uint8_t r_prime;
    uint8_t g_prime;
    uint8_t b_prime;
    if (is_min) {
        assert(rgb.r >= min_component);
        assert(rgb.g >= min_component);
        assert(rgb.b >= min_component);
        r_prime = rgb.r - min_component;
        g_prime = rgb.g - min_component;
        b_prime = rgb.b - min_component;
    } else {
        uint8_t w3 = w / 3;
        assert(rgb.r >= w3);
        assert(rgb.g >= w3);
        assert(rgb.b >= w3);
        r_prime = rgb.r - w3;
        g_prime = rgb.g - w3;
        b_prime = rgb.b - w3;
    }
    RGBW out = { r_prime, g_prime, b_prime, w };
    return out;
}

/////////////////////////// BEGIN TEST ///////////////////////////

int calc_r(const RGBW input) {
    return int(input.r) + int(input.w) / 3;
}

int calc_g(const RGBW input) {
    return int(input.g) + int(input.w) / 3;
}

int calc_b(const RGBW input) {
    return int(input.b) + int(input.w) / 3;
}

float calc_error(const RGB input_rgb, const RGBW input_rgbw) {
    int r_val = calc_r(input_rgbw);
    int g_val = calc_g(input_rgbw);
    int b_val = calc_b(input_rgbw);
    int r_error = abs(r_val - int(input_rgb.r));
    int g_error = abs(g_val - int(input_rgb.g));
    int b_error = abs(b_val - int(input_rgb.b));
    int rgb_error = r_error + g_error + b_error;
    float error = float(rgb_error) / (float(255) * 3);
    return error;
}

void test_rgb_2_rgbw(const RGB input) {
    RGBW output = rgb_2_rgbw(input);
    cout << "Input " << int(input.r) << ", " << int(input.g) << ", " << int(input.b) << "\n";
    cout << "output " << int(output.r) << ", " << int(output.g) << ", " << int(output.b) << ", " << int(output.w) << "\n";
    int r_val = calc_r(output);
    int g_val = calc_g(output);
    int b_val = calc_b(output);
    int r_error = abs(r_val - int(input.r));
    int g_error = abs(g_val - int(input.g));
    int b_error = abs(b_val - int(input.b));
    int rgb_error = r_error + g_error + b_error;
    float error = float(rgb_error) / (255 * 3);
    cout << "error: " << error << "\n";
    cout << "\n";
}

void test_max_error() {
    float max_error = 0.f;
    RGB max_error_rgb = { 0, 0, 0 };
    for (int r = 0; r < 256; ++r) {
        for (int g = 0; g < 256; g++) {
            for (int b = 0; b < 256; b++) {
                RGB input = { uint8_t(r), uint8_t(g), uint8_t(b) };
                RGBW output = rgb_2_rgbw(input);
                float error = calc_error(input, output);
                if (error > max_error) {
                    max_error = error;
                    max_error_rgb = input;
                }
            }
        }
    }
    std::cout << std::fixed << std::setprecision(2);
    std::cout << "max error is: " << max_error * 100 << "%\n";
    if (max_error > 0.f) {
        std::cout << "max error rgb is: " << int(max_error_rgb.r) << ", " << int(max_error_rgb.g) << ", " << int(max_error_rgb.b) << "\n";
    }
}

int main()
{
    test_max_error();
    return 0;
}
```

toine512, 2023-05-03:
> Happy to help with testing if needed. I've got a couple of these sitting around not being very useful if anyone has a PR they want me to try?

Which MCU(s) do you have?

sammcj, 2023-05-03:
> @toine512 
> 
> Which MCU(s) do you have?

I've got a Sensory Bridge (ESP32-S2 I believe?), ESP32-S3, RockPi 4SE and a Pi Zero.

What I don't have is any C skills, but happy to run up someone's fork / branch etc!

toine512, 2023-05-03:
FastLED has issues with ESP32-S and ESP32-C #1438 #1189. First get it working with a RGB LED.
Then :
> Hello
> 
> I added white channel support retaining compatibility with existing code (thank you C++): https://github.com/toine512/FastLED-W However **only ESP32 RMT is supported**. Just use SK6812W controller. Colours are defined by 0xWWRRGGBB and there's a CRGB::WhiteLight constant. FASTLED_RMT_BUILTIN_DRIVER doesn't work since 3.4 and I'm not tooled to debug it.
> 
> I am just starting to test this with a real strip so keep in mind that things can be broken. Currently there is no white channel interpolation mode like WLED does. You're juste given the white channel to control. I hope someone can have fun with that.

FastLED runs on MCUs, you'll need something else for Linux.

zackees, 2023-05-03:
From what I can see, it looks like the SK6812 simply has an extra LED with no change in the protocol. But I could be wrong. So testing could be as easy as adding +25% to the RGB led array and assume every 1st (or is it 4th?) element is the white component.

If this is true, then a proper color test would involve having an RGB array, applying the color conversion test to it, then writing out the resulting RGBW array as an CRGB array to fastled.

sammcj, 2023-05-03:
Interesting - that would explain why it 'sort of' works - but looks very washed out / over white.

marmilicious, 2023-05-03:
> the RGB led array and assume every 1st (or is it 4th?) element is the white component

Several data sheets I've looked for SK6812RGBW pixels show White as 4th element.

Balb6, 2023-05-18:
Hello, I have the same issue with my LED stripe and don't have the skills to understand all that discussion, and I don't know if it will add something but in case it helps :

As said above, the add of a white LED can only change the saturation of the apparent color : 
- the aparent color is the sum of the light emited by all the LEDs
- the RGB values define the hue of the color
- the white LED emit a uniform (to simplify) amount of light in all colors, thus its level only changes the saturation of the resulting color.

Thus the use of a white LED don't add more "colors" to those attainable by the RGB LEDs.

I think the main use of the fourth white LED is to replace gray levels produced by the RGB triplets : for example convert 

RGB(255,255,255) into RGBW(0,0,0,255)
RGB(10,10,10) -> RGBW(0,0,0,10)
... etc ...

Or to add luminance to the gray levels : 
RGB(255,255,255) -> RGBW(255,255,255,255)
RGB(10,10,10) -> RGBW(10,10,10,10)

The RGB / W ratio could also be tuned to change the "warmth" of the gray levels.

Thus, with a RGBW there is a RGB palette, which is unchanged, and a gray levels palette, which is brighter and can be tuned 

Does it helps ?

zackees, 2023-05-19:
Typically each led outputs 20 mA of light.

So this means that RGB(1,1,1) = RGBW(0,0,0,3)

A different algorithm than mine could be implemented. But mine gives accurate color representation and has been tested before by other people that came to the same conclusion. It's also simple to understand and scales nicely.

It's also entirely in integer space and doesn't need to be converted into say HSV as an intermediary step.

You can also use my algorithm today. Allocate two CRGB arrays, one representing a normal pixel array of CRGB array, then a second with the same size but +25% for the white led. Feed the second led array to fastled.

On each draw step, convert the first CRGB array that you drew in to, to the second CRGB array that is 25% large because of the extra white component, using my code. Then call show(). FastLED should draw this correctly now.

If someone finds a better algorithm than mine than that's great. My algorithm was designed to get a reference implementation out there for discussion. It also allows pure white led as long as r=g=b and each is under 85, which is actually really bright given the perception curves our eye is used to.

theicfire, 2023-08-12:
Two thoughts about the SK6812 strips:
- The timing is similar enough to the WS2811 strips, that I used that. It's more similar to that than the WS2812 strips.
- You need to change the color ordering. The WS2811 takes `FastLED.addLeds<WS2811, LED_PIN, GRB>`, whereas I'm using `FastLED.addLeds<WS2811, LED_PIN, RGB>` for the SK6812 strips. 

Other than that, something like this will work:

```
#define NUM_LEDS_RGBW NUM_LEDS * 1.25 ... watch out for floating point issues
CRGB leds_rgbw[NUM_LEDS_RGBW];

...

  RGBW *leds_rgbw_pointer = (RGBW *)leds_rgbw;
  for (int i = 0; i < NUM_LEDS; i++) {
    leds_rgbw_pointer[i] = rgb_2_rgbw(leds[i]);
  }
```

polykoma, 2023-08-25:
maybe out of context but I stumbled over this piece of code. Saying its has working RGBW for the SK. maybe of use maybe not.
https://gist.github.com/tuxuser/45d06bfc2eb8b92de2f8c6fe0dded28d

kriegsman, 2023-09-13:
Cross-referencing this here:  I've (finally) started work on the low-level FastLED driver for four-channel (RGBW) WS281x LED strips.  I'll be tracking that work over on this Issue:  https://github.com/FastLED/FastLED/issues/1540

rovi59, 2023-11-16:
This worked for me:

https://www.partsnotincluded.com/fastled-rgbw-neopixels-sk6812

trullock, 2024-09-09:
> This worked for me:
> 
> https://www.partsnotincluded.com/fastled-rgbw-neopixels-sk6812

This worked for me with SK6812 RGBWs

zackees, 2024-09-09:
RGBW is now supported.

https://www.reddit.com/r/FastLED/s/I7xv02O2Yc

closed 2024-09-09 by zackees

trullock, 2024-09-09:
Timely! Thanks