FastLED/FastLED #1282: How to globally adjust saturation to zero for the whole sketch?

twoporylyj, 2021-08-18:
Greetings. I am new to FastLED, and I have a question. Is there a simple way to globally adyust saturation to zero for the whole sketch? So that I download any sketch that uses FastLED and displays an image or an effect, add or change only a few lines there, and the sketch starts displaying absolutely the same, but in graysacle? Thanks.

marmilicious, 2021-08-18:
Hello @twoporylyj   Give this a try:
https://pastebin.com/4CbH8T3g

We use github "issues" here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED where you might get some other ideas on how to do this.

twoporylyj, 2021-08-18:
Thanks, I will try this.

twoporylyj, 2021-08-19:
Tried your method, and it works. Thanks again.

marmilicious, 2021-08-25:
@twoporylyj  Feel free to close this issue if you have a working solution now.

twoporylyj, 2021-08-25:
Greetings. Even though now i have a perfectly working solution, it would be much greater to include global saturation adjustment into the library itself, like global brightness adjustment that already exists in the library. So i don't know whether to close this issue.