FastLED/FastLED #1290: Increase the BPM rate?

JacksonAudio, 2021-08-30:
The sine functions in this library are incredible. My only issue is that I need the BPM to be about 5 times faster. I need an LFO around 20Hz and the fastest I can get is 256 BPM / 60 = 4.26667Hz

Any idea how I can do this?

kriegsman, 2021-09-23:
Hi and great question and idea.  The easiest way is probably going to be for you to copy some of the source out of the library files and modified it.  The definition of beat88 says this:

```
LIB8STATIC uint16_t beat88( accum88 beats_per_minute_88, uint32_t timebase = 0)
{
    // BPM is 'beats per minute', or 'beats per 60000ms'.
    // To avoid using the (slower) division operator, we
    // want to convert 'beats per 60000ms' to 'beats per 65536ms',
    // and then use a simple, fast bit-shift to divide by 65536.
    //
    // The ratio 65536:60000 is 279.620266667:256; we'll call it 280:256.
    // The conversion is accurate to about 0.05%, more or less,
    // e.g. if you ask for "120 BPM", you'll get about "119.93".
    return (((GET_MILLIS()) - timebase) * beats_per_minute_88 * 280) >> 16;
}
```

You could probably make a similar sawtooth generator with a 16-bit integer BPM value like this (warning: untested!)
```
uint16_t beat168( uint16_t beats_per_minute_16, uint32_t timebase = 0)
{
    return (((GET_MILLIS()) - timebase) * beats_per_minute_16 * 280) >> 8;
}
```

Let me know how that works for you.

JacksonAudio, 2021-09-24:
Thanks for the info! I found that I can get what I need by just change the 280 to 1315. That gives me a range of 0-20Hz using 0-255. There are some ugly transients on the waveform as the values are shifting so I still need to work through that but thanks again for the awesome info!