FastLED/FastLED #1291: fill_rainbow resets Arduino Every...

emccarron, 2021-09-05:
Using fill_rainbow causes a reset on the Arduino Every (ATMega4809 based.)

Upgraded to latest Fastled rev (previous wouldn't compile.)

Code compiled but glitches when calling fill_rainbow causing device to reset.

ngyl88, 2021-09-23:
Sorry for the noise. Referenced to the wrong issue.

ngyl88, 2021-09-23:
This issue instead related to issue #716 which is a bigger work in progress. The ATmega4809 boards aren't fully supported yet.
- SPI support for Arduino Nano Every was added in PR #1166 . This PR probably fixed the compilation issue that you faced previously.

Is fill_rainbow dependent on SPI? I think reset is related to pin and the interface support. 

Sharing this PR because I think the pin definition linked in the PR will help for checking. And further looking into support on Arduino Nano Every.