FastLED/FastLED #1295: WS2812B controlled by analog input of Arduino Uno Wifi rev2

AlbertoAlosS, 2021-09-08:
Hi!

I need to control 18 WS2812b led stripes with an Arduino Uno Wifi. For the 13 digital pins is working fine but I need to use the 5 analog pins to achieve my requirements. Can FastLed use analog in as if they were digital ones to control de WS2812b?

Thanks a lot!

Alberto

marmilicious, 2021-09-11:
Did you try it? It will either work or it won't. Typically analog pins should be able to be digital pins, but digital pins can not be analog pins.

AlbertoAlosS, 2021-09-15:
Yes, I tried by naming them as A0,A1, etc and also as 14/15.. and I was not able to make it work. This is the code I used:

`#include <FastLED.h>
#define LED_PIN     A0,A1,etc or 14,15,16, etc
#define NUM_LEDS    1
CRGB leds[NUM_LEDS];
void setup() {

  FastLED.addLeds<WS2812, LED_PIN, GRB>(leds, NUM_LEDS);
  
}
void loop() {
  
  leds[0] = CRGB(255, 0, 0);
  FastLED.show();

}`

I achieved it by using the Neopixel library which allows working with analog pins as they where digital pins.

Note that I am not trying to use digital pins as analog, I am just trying to use analog pins as digital ones.

Regards!