FastLED/FastLED #1296: FastLED 3.4.0 don"t work with esp8266 whith git clone

marcelinDelcour, 2021-09-12:
Hello

I encounter this issue with the 3.4.0 version. when I get the library from github with _git clone https://github.com/FastLED/FastLED.git_ , even the blink exemple don't work on my wemos D1 mini or D1 mini pro

If I delete all FastLed files from Arduino/library and use arduino IDE library manger to get FastLED lib, it works fine
reproduce this issue on a windows 10 OS and a linux elementary 6 odin OS
arduino IDe is version 1.8.16
esp8266 arduino core is version 3.0.2 
no matter which pin i use

the sketch compile but it seem that delay() don't work anymore when followed by a FastLED.show()
with the blink exemple, the led stay off or on, but does not blink

regards 
Marcelin

samguyer, 2021-09-23:
Hi @marcelinDelcour 
Thanks for submitting this issue. We are aware that the latest ESP8266 core broke some/all of FastLED, and we are working on it! You can see some of the discussion on other FastLED issues related to ESP8266.

Jpipe001, 2021-09-26:
Try: Use Previous Board Manager 1.0.6.

[image]

kriegsman, 2021-11-01:
Just posting here to let folks know that @samguyer and I are looking at the ESP 8266 issue some more right now.

I just pulled down the ESP8266 3.0.2 core, and "DemoReel100" is working okay on a 30-pixel strip. We're investigating further.

kriegsman, 2021-11-02:
As of https://github.com/FastLED/FastLED/commit/ccd2c7b426179ad50e4e2c0d3f132954fc6aa72c, the problem with the ESP8266 core 3.0.x may be fixed; please give it a try and let us know if you can.  I tested it after a git-clone and it worked on my test setup, but your feedback would be helpful, too.

Jpipe001, 2021-11-03:
Will this fix my ESP32 issue too.

samguyer, 2021-11-03:
@Jpipe001 This fix is only for ESP8266 (the ESP32 driver is completely different code). Do you have an issue submitted for your ESP32 problem?

marcelinDelcour, 2021-11-04:
Try the last version as required and I confirm my issue is now resolved
Thank for your work !
Marcelin

closed 2021-11-04 by marcelinDelcour