FastLED/FastLED #1300: FastLED Not working with New Board Manager 2.0.0

Jpipe001, 2021-09-26:
With the latest version of Board Manger, FastLED 3.4.0 does not compile correctly and LEDs don't work.
My solution is to go back to previous version.

[image]

podaen, 2021-09-28:
Are You also using neopixel and fastled together? If yes which versions do have installed? I have rolleback my IDF like you did to 1.0.6... but it doesn't work for me.

Jpipe001, 2021-09-28:
Actually, although I rolled back the board manager and got the Fastled working with WS2812 leds. I still seem to be having issues.  I am just using Fastled and not Adafruit Neopixels.  These are my libraries:

#include <Arduino.h>
#include <FastLED.h>
#include <WiFi.h>
#include <WiFiMulti.h>
WiFiMulti wifiMulti;
#include <HTTPClient.h>
#include <WiFiClientSecure.h>
#include "time.h"
#include "ESPmDNS.h"

I am pretty sure that the problem is in the Fastled Library and this issue needs to be looked into.

ATLTVHEAD, 2021-09-28:
I'm having similar problems. When you say the board manager are you referring to the Esp32 package within the Arduino Boards Manager, or the Arduino Boards Manager itself. If that is the case, how does one do that? 

Sorry for all the questions!

Jpipe001, 2021-09-28:
A picture speaks 1000 words:
[image]

ATLTVHEAD, 2021-09-28:
Thank you so much! @Jpipe001

podaen, 2021-09-28:
It is a bit strang for me, because I had it working on IDF 2.0.0 but I had neopixel also running and I had to downgrade it, Soo I looked to use the newest versions of the librarys. Now I got IDF 1.0.6 and not the lastest neopixel. (see my topic) Fastled is the newest.

podaen, 2021-09-28:
Like I said, I used the all latest... except the Neopixel library, and it worked for me. Note, that I am not yet implemented the Wi-Fi for now. You need a task manager to accomplish that.

podaen, 2021-09-28:
Quot: _You need a task manager to accomplish Wi-Fi._ 

At least that is what I read on the internet. But I had a project on the SDK 1.0.6 running without taskmanager.

alorbach, 2021-09-29:
I am experiencing very strange problems with the default RMT driver on new SDK 2.0.0
Sometimes it works, sometimes after rebuild LEDs do not respond - FastLED.Show calls work  . but performance appears to be slower. When it works I have like 100 LED FPS and when the problem appears, its about 50 FPS.

I even had crashes which disappeared after I switched to FastLED.Show() 

Somwething is not right.

alorbach, 2021-09-30:
So I got FastLED to work again but with I2S driver!

> # define FASTLED_ESP32_I2S true

However the I2S driver does random flickering on not controlled LEDs, which might be a problem. I would more like to use the RMT driver again which was working flawless in 1.0.6 but not anymore in 2.0.0 which means there must be a bug somewhere.

podaen, 2021-09-30:
Did you unregistered the RMT?

alorbach, 2021-09-30:
> Did you unregistered the RMT?

How? Btw. RMT driver did work for some time and then it stopped working again while  I was improving unrelated code.

alorbach, 2021-09-30:
Here I made a video from I2S driver, not used LEDs blink from time to time, it does not happen with RMT:
https://user-images.githubusercontent.com/1675286/135405935-cb94f000-a1a5-4a03-a1c4-ce08c95ad7a6.mp4

urish, 2021-10-20:
> I don't know why it wasn't happening on 1.0.6, all memory was properly initialized to 0 after reboot?

As far as I can tell, the `rmt_tx` variable lives on the stack, so it just gets whatever value that was there before. 

A possible solution (which should also be future-proof) would probably be to initialize the variable when declaring it, e.g.

```cpp
rmt_config_t rmt_tx = {}:
```

closed 2024-09-08 by zackees