FastLED/FastLED #1301: Fastled Stops Working After Upgrading The Neopixel Library

podaen, 2021-09-26:
I am working on an esp32 vroom with Arduino IDE and Visual Micro and before upgrading the Neopixel library  I could work with neopixel and fastled in one project. After updating the Neopixel library, the Fastled libary does not work anymore. I have made a new project to test it out if there where no other reasons but I have the same results. Here Is my test project. 

```
//#include <rp2040_pio.h>
#include <FastLED.h>
#include <Adafruit_NeoPixel.h>

#define  MASTER_BRIGHTNESS 60

#define LED_PIN    18  // NeoPixel LED strand is connected to this pin
#define LED_PIN_2  19  // NeoPixel LED strand is connected to this pin

#define N_PIXELS 136//51
#define NUM_STRIPS 2//51 

#define LED_TYPE WS2813//WS2813 voor 12v ws2815      
#define COLOR_ORDER GRB
CRGB ledsLeft[N_PIXELS];
CRGB ledsRight[N_PIXELS];

Adafruit_NeoPixel strip = Adafruit_NeoPixel(N_PIXELS, LED_PIN, NEO_GRB + NEO_KHZ800);
Adafruit_NeoPixel strip1 = Adafruit_NeoPixel(N_PIXELS, LED_PIN_2, NEO_GRB + NEO_KHZ800);

CLEDController* controllers[NUM_STRIPS];

void setup() {
	Serial.begin(115200);
	while (!Serial);
	strip.begin();
	strip.setBrightness(MASTER_BRIGHTNESS);
	strip.show(); // all pixels to 'off'
	strip1.begin();
	strip.setBrightness(MASTER_BRIGHTNESS);
	strip1.show(); // all pixels to 'off'

	controllers[0] = &FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(ledsLeft, N_PIXELS);// .setCorrection(TypicalLEDStrip);
	controllers[1] = &FastLED.addLeds<LED_TYPE, LED_PIN_2, COLOR_ORDER>(ledsRight, N_PIXELS);// .setCorrectio

	Serial.println("End Setup");
}
void loop() {
	//delay(50);
	AllToBlack();
	delay(3000);
	TestColorsNeoPixel();
	TestColorsFastLed();
}
void AllToBlack() {
	strip.clear();
	strip.show();
	for (int i = 0; i < N_PIXELS; i++) {
		strip.setPixelColor(i, strip.Color(0, 0, 0));
		strip.setPixelColor(i, strip.Color(0, 0, 0));
		//strip1.setPixelColor(i, strip1.Color(0, 0, 0));
	}
	strip.show();
	strip1.show();
	strip.setBrightness(0);
	strip1.setBrightness(0);
	strip.show();
	strip1.show();
	//strip1.show();
}
void TestColorsNeoPixel() {
	Serial.println("Yellow");
	strip.setBrightness(MASTER_BRIGHTNESS);
	strip1.setBrightness(MASTER_BRIGHTNESS);
	strip.clear();
	strip.show();
	for (int i = 0; i < N_PIXELS; i++) {
		strip.setPixelColor(i, strip.Color(255, 255, 0));// 138, 255, 255
		strip1.setPixelColor(i, strip.Color(255, 255, 0));// 138, 255, 255
	}

	strip.show();
	strip1.show();
	delay(1000);

	Serial.println("red");
	strip.clear();
	strip.show();
	for (int i = 0; i < N_PIXELS; i++) {
		strip.setPixelColor(i, strip.Color(255, 0, 0));// 138, 255, 255
		strip1.setPixelColor(i, strip.Color(255, 0, 0));// 138, 255, 255
	}
	strip.show();
	strip1.show();
	delay(1000);
	Serial.println("blue");
	strip.clear();
	strip.show();
	for (int i = 0; i < N_PIXELS; i++) {
		strip.setPixelColor(i, strip.Color(0, 0, 255));// 138, 255, 255
		strip1.setPixelColor(i, strip.Color(0, 0, 255));// 138, 255, 255
	}
	strip.show();
	strip1.show();
	delay(1000);
}
void TestColorsFastLed() {
	Serial.println("yellow");
	//FastLED.showColor(CHSV(map(60, 0, 360, 0, 255), 255, 255);)
	for (int i = 0; i < N_PIXELS; i++) {
		ledsLeft[i] = ledsRight[i] = CHSV(map(60, 0, 360, 0, 255), 255, 255); //0xFFFF00;//
	}
	controllers[0]->showLeds(80);
	controllers[1]->showLeds(80);
	delay(1000);

	Serial.println("red");
	for (int i = 0; i < N_PIXELS; i++) {
		ledsLeft[i] = ledsRight[i] = CHSV(map(0, 0, 360, 0, 255), 255, 255); //0xFFFF00;//
	}
	////FastLED.show();
	//FastLED[0].showLeds(MASTER_BRIGHTNESS);
	//FastLED[1].showLeds(MASTER_BRIGHTNESS);
	controllers[0]->showLeds(80);
	controllers[1]->showLeds(80);

	delay(1000);
	Serial.println("pink");
	for (int i = 0; i < N_PIXELS; i++) {
		ledsLeft[i] = ledsRight[i] = CHSV(map(300, 0, 360, 0, 255), 255, 255); //0xFFFF00;//
	}
	//FastLED.show();
	controllers[0]->showLeds(80);
	controllers[1]->showLeds(80);
	delay(5000);
	Serial.println("blue");
	for (int i = 0; i < N_PIXELS; i++) {
		ledsLeft[i] = ledsRight[i] = CHSV(map(240, 0, 360, 0, 255), 255, 255); //0xFFFF00;//
	}
	//FastLED.show();
	controllers[0]->showLeds(80);
	controllers[1]->showLeds(80);
	delay(5000);

}
```

Jpipe001, 2021-09-27:
See Issue #1300

podaen, 2021-09-27:
Thanks, did some roleback to day. But not of the IDF. I did the neopixel library to see what was the cause of this problem and I think I found it. 

[See here what I did](https://github.com/adafruit/Adafruit_NeoPixel/issues/294)

You are right. It happens after an update of the IDF AND neopixel, but It can be avoided by unregister the RMT.

samguyer, 2021-11-04:
@podaen Thanks for digging into this issue. I was recently looking at the ESP32 support in the Adafruit library and noticed that they are using the built-in RMT driver, which takes over the whole device, so it doesn't play nice-nice with our driver.

podaen, 2021-11-15:
We are now two month further and still no reply from them on my topic there. I will close this topic because it is out of your concern.

closed 2021-11-15 by podaen