FastLED/FastLED #1302: FastLED Intermittent Pixel Issue

94Soarer, 2021-09-29:
Having an intermittent WS2812 led/leds issue. I have built 4 of these pcb boards prior and no issues, recent board (1 month after the last working one) I am having the following problem. 

The problem I am having is that at certain times only 1 of the 8 led's turn on and other times 3 of the 8 leds turn on randomly and stay on.

My code in the beginning cycles through the following colors, green, yellow, red. When it functions normally all led's turn on and display the color at 400ms intervals, led's after red turn off after 400ms. 
When I have the intermittent issue only the first led turns on and cycles through the color order at a lower brightness, at the end of displaying red, a group of 3 led's randomly turn on displaying random colors. (could be led2-led4 or led5-led7, totally random but always grouped). 

Snippets of Code:
`#include <FastLED.h>                                          // WS2812 Library
// Shift light Config
// How many leds are connected?
#define NUM_LEDS 8
// Define the Pins for WS2812
#define DATA_PIN 10
// Define the array of leds
CRGB leds[NUM_LEDS];`
`void setup()  {
  FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);
  FastLED.clear();
  FastLED.show();
  cycleLED();
}`
`void cycleLED()
{
  FastLED.setBrightness(50);
  int i;

  for (i = 0; i < 8; i++)
  {
    leds[i] = CRGB::Green;
  }
  FastLED.show();
  delay(400);

  for (i = 0; i < 8; i++)
  {
    leds[i] = CRGB::Yellow;
  }
  FastLED.show();
  delay(400);

  for (i = 0; i < 8; i++)
  {
    leds[i] = CRGB::Red;
  }
  FastLED.show();
  delay(400);
  FastLED.clear();
  FastLED.show();
}`

Setup:
8X WS2812B Pixels, Datasheet shown here with a minimum DIN of 2.7V. 
https://datasheet.lcsc.com/lcsc/2006151006_Worldsemi-WS2812B-B_C114586.pdf
Teensy 3.2 Microcontroller, for the data signal I am using pin 10 with a 390 ohm resistor. 
The pixel strip pcb is composed of the following. 
[image]

The entire setup is powered via a Pololu (D36V28F5) DC-DC step-down voltage regulator which drops 12V down to 5V. The 5V output is then used to power the teensy 3.2 and ws2812 LEDs.

94Soarer, 2021-10-15:
I have attached videos of the problem. As you can see only a few led's from the entire strip turn on and work as intended. Sometimes the strip turns on and gets stuck in the color blue. 

Unplugging the power and reconnecting power has always fixed this issue currently. 

FYI these are WS2812B-V5 Leds.
https://user-images.githubusercontent.com/91585966/137435480-a34d81cd-df78-4319-8078-243f80c8bb9d.mov

https://user-images.githubusercontent.com/91585966/137435488-18fe53db-21cc-4382-a66f-78b1e01d536d.mov

fbrkt, 2023-08-07:
I have the same problem with the WS2812B V5, unfortunately, I haven't fixed it yet, but what I can say about your schematic is that in the Datasheet on page 4 it says that the V5 don't need the filter capacitor : 
[image]

zackees, 2026-08-20:
Deferred during the bug burn-down: this intermittent hardware report has no deterministic current-version reproduction or signal/power evidence. Timing-specific work is tracked more narrowly in #1311; revisit with a minimal sketch, exact versions, voltage measurements, and logic-analyzer capture.