FastLED/FastLED #1311: Change Timing to match WS2812B-V5 datasheet.

94Soarer, 2021-10-19:
What would be the appropriate steps to change the timing values of the ws2812b clockless LEDs?

Currently using ws2812B version 5 where the transfer times for T1L and RES vary quite a bit. 

I believe that upon bootup this causes the led's to freeze and not send data out. 

Datasheet can be found here: https://datasheet.lcsc.com/lcsc/2006151006_Worldsemi-WS2812B-B_C114586.pdf

henrygab, 2021-11-04:
Here's a walkthrough of steps, when the clockless controller you thought you should use doesn't work (and you've done other troubleshooting, such as trying other pins, etc.).

<details><summary>Step 1: Decode the datasheet to into T1, T2, T3 requirements</summary><P/>

The datasheets are notoriously unreliable as to what values actually work well in real-world implementations.   The datasheet you linked to reports the following:

| Phase | Tmin | Tmax |
|-----|-----|-----|
| T0H | 220 | 380 |
| T0L | 580 | 1000 |
| T1H | 580 | 1000 |
| T1L | 580 | 1000 |

(all times in nanoseconds)

The trick now is to convert this into T1, T2, T3 values that, through testing, will actually work well for your configuration and setup.

</details><HR/>

Thus, if you trust the datasheet numbers:

```
T1       == T0H (220 .. 380)
   T2+T3 == T1L (580 .. 1000)
T1+T2    == T1H (580 .. 1000)
      T3 == T1L (580 .. 1000)
```

<details><summary>Step 2: Try existing controllers with similar timings</summary><P/>

The easiest way is to just look at chipsets.h (clockless section).
Filtering out by "close" values for T1 and T3 gives a small list:

```
// WS2811 - 320ns, 320ns, 640ns
// WS2813 - 320ns, 320ns, 640ns
// SK6812 - 300ns, 300ns, 600ns
// not quite in spec, but worth trying (might be close enough)
// TM1809 - 350ns, 350ns, 550ns
// TM1829 - 340ns, 340ns, 550ns
```

That's a pretty small subset to try... may as well try them all!

</details><HR/>
<details><summary>Step 3: determine custom values for T1, T2, T3</summary><P/>

Maybe the existing controllers aren't quite reliable enough.  Maybe you heard that consistent / balanced timings should work.  Maybe you just like making up new numbers.  Or, maybe the phase of the moon matters.  Whatever the case, find the values you want to test.  For the sake of example:

```
T1       == T0H (220 .. 380)
   T2+T3 == T1L (580 .. 1000)
T1+T2    == T1H (580 .. 1000)
      T3 == T1L (580 .. 1000)
```

Thus, I decide on a whim to try:
```
T1 = 250, T3 = 625

With T1 = 250,  330 <= T2 <= 750
With T3 = 625,  -45 <= T2 <= 375

T1 = 250, T2 = 325, T3 = 625 // close enough to test...
```

I want numbers fully within spec, so I adjust T1 upward:
```
T1 = 375, T3 = 625

With T1 = 375,  205 <= T2 <= 625
With T3 = 625,  -45 <= T2 <= 375

T1 = 375, T2 = 250, T3 = 625 // in-spec numbers
```

Note that I've stuck with numbers that are each multiples of 125ns.  This ensures that, if they work, they (should) work on 8MHz, 16MHz, and 24MHz CPUs.

</details><HR/>
<details><summary>Step 4: Try your custom values with self-defined controller</summary><P/>

Put the following at the top of your sketch, to test 

```
// Testing values of T1 = 375ns, T2 = 250ns, T3 = 625ns
template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class CustomTiming1Controller : public ClocklessController<
    DATA_PIN, C_NS(375), C_NS(250), C_NS(625), RGB_ORDER, 0, false, 5
    > {};
template<uint8_t DATA_PIN, EOrder RGB_ORDER = GRB> class WS2812B_v5A :
 public CustomTiming1Controller<DATA_PIN, RGB_ORDER> {};

// Testing values of T1 = 250ns, T2 = 325ns, T3 = 625ns
template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class CustomTiming2Controller : public ClocklessController<
    DATA_PIN, C_NS(250), C_NS(325), C_NS(625), RGB_ORDER, 0, false, 5
> {};
template<uint8_t DATA_PIN, EOrder RGB_ORDER = GRB> class WS2812B_v5B :
 public CustomTiming2Controller<DATA_PIN, RGB_ORDER> {};
```

Then just replace `WS2812B` with `WS2812B_v5A` when declaring your LED type.
If that doesn't work, try `WS2812B_v5B` to try the slightly-out-of-spec timings.

Rinse, lather, repeat, until you find values that work reliably for you.

**IMPORTANT:**  Try numbers that are NOT within specification.  You might be surprised by what works.

</details><HR/>

<details><summary>Step 5: If all else fails</summary><P/>

If you tried all the above, and can't find timings that work for you, then you'll likely need to move to a different LED control scheme.  Specifically, it's possible that some types of chips can't reliably be driven with the T1/T2/T3 scheme, such as when T1 and T3 vary so much as to make a compatible value for T2 non-existent.

In such a case, you'll need to find a different software, such as one where you can independently specify T0H, T0L, T1H, T1L.  _(I'd still try all the listed controllers first, "just in case".)_

</details><HR/>

Post your results here, as it will undoubtedly help others.

**_And good luck!_**

fbrkt, 2023-08-09:
What if the board is an Esp32, so using an AVR architecture ? can you still use C_NS() ?

zackees, 2024-10-12:
This has been fixed in FastLED 3.8 for the esp. We now use the WS2812-v5b timings of 250 uS for reset. We more than made up for it with async drawing now.