FastLED/FastLED #1312: ESP32 based boards supported, but fail to compile.

AndKe, 2021-10-21:
```
Arduino: 1.8.16 (Linux), Board: "Generic STM32F1 series, BluePill F103C8, STM32CubeProgrammer (SWD), Enabled (generic 'Serial'), CDC (generic 'Serial' supersede U(S)ART), Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"

/home/andre/Arduino/arduino-1.8.16/arduino-builder -dump-prefs -logger=machine -hardware /home/andre/Arduino/arduino-1.8.16/hardware -hardware /home/andre/.arduino15/packages -hardware /home/andre/Arduino/projects/hardware -tools /home/andre/Arduino/arduino-1.8.16/tools-builder -tools /home/andre/Arduino/arduino-1.8.16/hardware/tools/avr -tools /home/andre/.arduino15/packages -built-in-libraries /home/andre/Arduino/arduino-1.8.16/libraries -libraries /home/andre/Arduino/projects/libraries -fqbn=STM32:stm32:GenF1:pnum=BLUEPILL_F103C8,upload_method=swdMethod,xserial=generic,usb=CDCgen,xusb=FS,opt=osstd,rtlib=nano -ide-version=10816 -build-path /tmp/arduino -warnings=default -build-cache /tmp/arduino_cache_831661 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM32Tools.path=/home/andre/.arduino15/packages/STM32/tools/STM32Tools/1.4.0 -prefs=runtime.tools.STM32Tools-1.4.0.path=/home/andre/.arduino15/packages/STM32/tools/STM32Tools/1.4.0 -prefs=runtime.tools.CMSIS.path=/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=/home/andre/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.2.1-1.1.path=/home/andre/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1 -verbose /tmp/arduino_modified_sketch_671312/Blink.ino
/home/andre/Arduino/arduino-1.8.16/arduino-builder -compile -logger=machine -hardware /home/andre/Arduino/arduino-1.8.16/hardware -hardware /home/andre/.arduino15/packages -hardware /home/andre/Arduino/projects/hardware -tools /home/andre/Arduino/arduino-1.8.16/tools-builder -tools /home/andre/Arduino/arduino-1.8.16/hardware/tools/avr -tools /home/andre/.arduino15/packages -built-in-libraries /home/andre/Arduino/arduino-1.8.16/libraries -libraries /home/andre/Arduino/projects/libraries -fqbn=STM32:stm32:GenF1:pnum=BLUEPILL_F103C8,upload_method=swdMethod,xserial=generic,usb=CDCgen,xusb=FS,opt=osstd,rtlib=nano -ide-version=10816 -build-path /tmp/arduino -warnings=default -build-cache /tmp/arduino_cache_831661 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.STM32Tools.path=/home/andre/.arduino15/packages/STM32/tools/STM32Tools/1.4.0 -prefs=runtime.tools.STM32Tools-1.4.0.path=/home/andre/.arduino15/packages/STM32/tools/STM32Tools/1.4.0 -prefs=runtime.tools.CMSIS.path=/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=/home/andre/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.2.1-1.1.path=/home/andre/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1 -verbose /tmp/arduino_modified_sketch_671312/Blink.ino
Using board 'GenF1' from platform in folder: /home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0
Using core 'arduino' from platform in folder: /home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0
/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/extras/prebuild.sh /tmp/arduino /tmp/arduino_modified_sketch_671312
Detecting libraries used...
/home/andre/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb @/tmp/arduino/sketch/build_opt.h -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -I/tmp/arduino_modified_sketch_671312 -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/avr -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32 -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/LL -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/usb -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/OpenAMP -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/usb/hid -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/usb/cdc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Drivers/STM32F1xx_HAL_Driver/Inc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Drivers/STM32F1xx_HAL_Driver/Src -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/STM32F1xx -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/OpenAMP -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/OpenAMP/open-amp/lib/include -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/OpenAMP/libmetal/lib/include -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/OpenAMP/virtual_driver -w -x c++ -E -CC -DSTM32F1xx -DARDUINO=10816 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLUEPILL_F103C8\"" -DSTM32F103xB -DUSBCON -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_UART_MODULE_ENABLED -I/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/Core/Include/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/ -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/ -I/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/DSP/Include -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/variants/PILL_F103XX /tmp/arduino/sketch/Blink.ino.cpp -o /dev/null
Alternatives for FastLED.h: [FastLED@3.4.0]
ResolveLibrary(FastLED.h)
  -> candidates: [FastLED@3.4.0]
/home/andre/.arduino15/packages/STM32/tools/xpack-arm-none-eabi-gcc/9.2.1-1.1/bin/arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb @/tmp/arduino/sketch/build_opt.h -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -I/tmp/arduino_modified_sketch_671312 -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/avr -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32 -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/LL -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/usb -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/OpenAMP -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/usb/hid -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/cores/arduino/stm32/usb/cdc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Drivers/STM32F1xx_HAL_Driver/Inc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Drivers/STM32F1xx_HAL_Driver/Src -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/STM32F1xx -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/OpenAMP -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/OpenAMP/open-amp/lib/include -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/OpenAMP/libmetal/lib/include -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/OpenAMP/virtual_driver -w -x c++ -E -CC -DSTM32F1xx -DARDUINO=10816 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"BLUEPILL_F103C8\"" -DSTM32F103xB -DUSBCON -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_UART_MODULE_ENABLED -I/home/and [... cut ...] /stm32/1.9.0/cores/arduino/stm32/usb/cdc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Drivers/STM32F1xx_HAL_Driver/Inc -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Drivers/STM32F1xx_HAL_Driver/Src -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/STM32F1xx -I/home/andre/.arduino15/packages/STM32/hardware/stm32/1.9.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -
 ... 
Error compiling for board Generic STM32F1 series.

```

samguyer, 2021-11-02:
@AndKe I don't see the actual compile error here. Can you turn on verbose compiler output?

maddogjt, 2021-11-21:
@AndKe your log appears to reference an STM32F1 board.  Did you mean STM32 board rather than ESP32 board in the title?

AndKe, 2021-11-21:
@maddogjt  - yes, sorry for the confusion, STM32F1 is correct (I've corrected the title now)