FastLED/FastLED #1322: Flickering on ESP8266 with Core >= 3.0.0

enwi, 2021-11-20:
I am having trouble with FastLED on the ESP8266 and Arduino Core versions starting at `3.0.0`.

What I tried so far:
- Add resistor in line of data signal (tried 100 and 220 Ohm)
- Use different library versions (FastLED 2.9 - latest)
- Add `#define FASTLED_ALLOW_INTERRUPTS 0`
- Add `#define FASTLED_INTERRUPT_RETRY_COUNT 1`
- Add `FastLED.setMaxRefreshRate(100);`
- Tried different power supplies including lab bench supply
- Add this 'fix' https://github.com/stnkl/FastLEDManager/commit/54bbaf1d12c89977501e1dea95f9347ae0a954ed
- Set ESP8266 to 160mhz

I also tried almost all combinations of the list above with the same result. 
The only thing that worked so far was downgrading the Core to `2.7.4`.

This might as well be an issue on ESP8266 Arduino Core side of things and not directly related to FastLED, but this issue is present when using FastLED.

DmytroKorniienko, 2021-11-23:
@enwi 
Check actual master, or this commit: https://github.com/FastLED/FastLED/commit/ccd2c7b426179ad50e4e2c0d3f132954fc6aa72c

enwi, 2021-11-23:
I tried the master branch already, did not fix the issue hence I opened the issue.

AronHetLam, 2021-11-27:
I also tried with latest commit: ccd2c7b, and had mixed results. 
With release 3.4.0 through Arduino Library manager the first pixel is the wrong color, but the rest is fine.
With commit ccd2c7b It works as it should if I add `#define FASTLED_ALLOW_INTERRUPTS 0`. However, if I allow interrupts most of the time only the first pixel  updates.

skorokithakis, 2021-12-11:
master solved the flickering issue for me, where setting only the green LED would make the red one flicker.

rjr136, 2022-03-15:
Hello,

Use framework 2.7.4                     How?
In Home Assistant, ESPHome nothing to find which is called framework or Arduino core or "number" 3.0.0 or 2.7.4 or anything resembling.

So please can any tell how to solve problem, instead of writing things which are not to be found in Home Assistant, ESPhome or Home Assistant, Add-ons, ESPhome.

Thanx and regards,

ssieb, 2022-03-15:
@rjr136 https://esphome.io/components/esp8266.html#configuration-variables

sgurgul, 2022-03-15:
Does anybody can explain an impact of using this 2.7.4 framework in long-term ? 
Is there a risk it will not be supported anymore soon so we should already start looking for FastLED alternatives ?

pika28, 2022-10-20:
Hi, Like @sgurgul, I'm wondering about using the long term version?

BKneer, 2023-05-03:
Hi,
with the latest 3.4.x update from ESPHome I get tons of errors compiling the fastled for ESP8266 and framework version 2.7.4 as mentioned in the guideline. With earlier ESPHome version it still works.
The update to 3.4.x works fine on FastLED projects using ESP32.
would be appreciated to get a solution for that.

Thanks
Bernhard

anm, 2023-07-31:
I also have this problem, and it seems to have got worse with recent versions.

With FastLED 3.6.0 and Arduino 3.0.0, the flicker can be cured by disabling Wifi. With newer Arduino versions, I cannot stop it. It is extreme - constant fast flickering on most (all?) LEDs. It makes the library unusable.

I have noted though that the problem only occurs when there are more than 60 (or possibly 61/62, can't remember) LEDs programmed in.

This is not an LED hardware or signal integrity problem.  I have tested with different ESP boards and LEDs. The output waveform from the ESP is clearly wrong on the oscilloscope. (It switches off for short periods when it should not.)

I read in some other issues that it might be caused by non maskable interrupts (nmi) which were enabled with version 3.0.0 of Arduino but I have not investigated or confirmed this.