FastLED/FastLED #1326: ESP32: RMT parallel output takes twice the time as expected

n1kelodeon, 2021-12-09:
Hi,

first of all, thank you very much for this great library!

Today, I tried to use parallel output on ESP32 for the first time and encountered an issue.

Let's say we have `LEDS_PER_PIN * PIN_COUNT` (WS2812) LEDs connected and writing to them using a single pin takes `SINGLE_PIN_TIME` milliseconds.
Using `PIN_COUNT` pins where each pin has `LEDS_PER_PIN` connected should then take `SINGLE_PIN_TIME  / PIN_COUNT` milliseconds.

However, what I achieve is only `(SINGLE_PIN_TIME / PIN_COUNT) * 2` milliseconds (It takes twice the time it should in theory).

In my example, writing to 600 LEDs using a single pin takes 20 ms, but writing to them using 4 pins with 150 LEDs per pin takes 10 ms instead of the desired 5 ms (20 ms / 4).

When I use the I2S driver the desired 5 ms are achieved.
For now I see no reason not to use I2S, but still I think RMT is not working properly.

Here is my code. I'm using FastLED 3.4.0.

Would be great if someone could conffirm my observations.
Also, I might be missing something that I need to configure in code?

Thank you in advance!

```cpp
#include <Arduino.h>

// #define FASTLED_ESP32_I2S // This achieves the desired degree of parallelization
#include <FastLED.h>

const unsigned LEDS_PER_PIN = 150;
const unsigned PIN_COUNT = 4;

CRGB leds[PIN_COUNT * LEDS_PER_PIN];

void setup() {
    Serial.begin(115200);

    FastLED.addLeds<WS2812B, 16, EOrder::BGR>(leds, 0 * LEDS_PER_PIN, LEDS_PER_PIN);
    FastLED.addLeds<WS2812B, 17, EOrder::BGR>(leds, 1 * LEDS_PER_PIN, LEDS_PER_PIN);
    FastLED.addLeds<WS2812B, 18, EOrder::BGR>(leds, 2 * LEDS_PER_PIN, LEDS_PER_PIN);
    FastLED.addLeds<WS2812B, 19, EOrder::BGR>(leds, 3 * LEDS_PER_PIN, LEDS_PER_PIN);
}

void loop() {
    auto timeBefore = millis();
    FastLED.show();
    auto passedTime = millis() - timeBefore;

    Serial.print("show() took ");
    Serial.print(passedTime);
    Serial.println(" ms");

    delay(1000);
}
```

adityathakekar, 2021-12-14:
Hey thanks for sharing this. Same results as you observed need to find the bug.

samguyer, 2021-12-15:
@n1kelodeon Thanks for the report. There have been some changes to the RMT driver that reduce parallelism, but the big change went from 8-way to 4-way. If you want to know the reasons, I can elaborate. BUT you should definitely be able to get 4-way parallelism. I'll look into it!

samguyer, 2022-01-02:
@n1kelodeon When I run your code, I get the expected 5ms for the show time. I'm using the current development head, which isn't significantly different from the 3.4.0 release. Can you tell me what hardware you're using?

n1kelodeon, 2022-02-16:
[samguyer](https://github.com/samguyer) Thanks for looking into it and please excuse the late reply.

I think this issue can be closed, since the bug has been fixed somewhere after it was introduced with the 3.4.0 release. The current master works fine for me, too.

I'm using ESP32 DevKit C V4.

closed 2022-02-16 by n1kelodeon