FastLED/FastLED #1329: ESP32: 200ms flicker every ~17.9 seconds with RMT driver

Jueff, 2021-12-14:
I discovered an issue that LEDs have short ~200ms flicker every ~17.9 seconds.
I investigated the problem and it turned out, the it has to do with __clock_cycles(). __clock_cycles() returns a 32bit value, each time the 32bit overrun occurs you see a short LED flicker at all LEDs which values change more then 10 times per second.

The sample project below blinks one LED very fast, if you watch the LED you will see the flickering every 17,9 seconds.

I think the cause of the problem is in the ESP32RMTController::fillNext method.

the line 
`if (mLastFill != 0 and now > mLastFill) {`
doesn't handle overrun correctly and can be replaced by
`if (mLastFill != 0) {`

as calculating the delta time still is OK (only 1 CPU tick difference in case of overrun).

**Code sample for problem reproduction**

```
#include <Arduino.h>

#define FASTLED_INTERNAL       // Disable version number message in FastLED library (looks like an error)
#include <FastLED.h>

const unsigned NUM_OF_LEDS = 256;
CRGB leds[NUM_OF_LEDS];

TaskHandle_t  FASTLedTaskHnd;

void setup() {
    Serial.begin(921600);
    //FastLED.addLeds<WS2812B, 27, EOrder::RBG>(leds, 0, NUM_OF_LEDS);
    CLEDController& controller0 = FastLED.addLeds<NEOPIXEL, 27>(leds+  0, NUM_OF_LEDS);
    controller0.clearLeds(256);
    FastLED.setDither(DISABLE_DITHER);
    
    xTaskCreatePinnedToCore(FASTLedTask,"FASTLed",4096,NULL,1,&FASTLedTaskHnd,0);
}

void FASTLedTask( void * parameter ) {
//--------------------------------
  while(1) {
    yield();
    loopFASTLed();
    delay (1);
  }
}

void loop() 
{
}

auto lastSecond = millis()/1000;

void loopFASTLed()
{
    auto timeBefore = millis();
    leds[0].r = ((timeBefore/30)&0x01)*128;
    FastLED.show();
    auto passedTime = millis() - timeBefore;

    if ((passedTime!=8&&passedTime!=7) || timeBefore/1000!=lastSecond)
    {
      lastSecond = timeBefore/1000;
      Serial.print("millis=");
      Serial.print(timeBefore);
      Serial.print(" cycles=");
      Serial.print(__clock_cycles());
      Serial.print(" show() took ");
      Serial.print(passedTime);
      Serial.println(" ms");
    }
}

```

Jueff, 2021-12-14:
could also be caused by the fact, that Fastled.show() is called from a a second core task.
I'm not sure, but it seams that __clock_cycles() returns different values when called from different core.

Maybe xthal_get_ccount() returns same value for both cores.

Jueff, 2021-12-14:
I can confirm that problem doesn't occur when calling FASTLed.shop inside loop() (on the default core).

Jueff, 2021-12-15:
changing the code in ESP32RMTController::fillNext to 

```
    uint32_t now = __clock_cycles();
    if (check_time) {
        if (mLastFill != 0 /*and now > mLastFill*/) {
            //uint32_t delta = (now - mLastFill);
            //if (delta > mMaxCyclesPerFill) {
            int32_t delta = (now - mLastFill);
            if (delta > (int32_t)mMaxCyclesPerFill) {
```

fixes the problem in my environment.

samguyer, 2021-12-15:
@Jueff Interesting -- thanks for tracking this problem down! Clearly, I need to spend more time staring at my LED displays. ;-)
Do you want to submit a PR for this change?

samguyer, 2022-04-10:
@Jueff I'm going to close this issue. Thanks for helping fix it!

closed 2022-04-10 by samguyer