FastLED/FastLED #1340: Bug in SM16703 timings

makaronnik13, 2021-12-30:
It seems that there is a bug in one of the declarations of SM16703Controller. That bug certainly messes up colors.

Controller in first declaration extends 
`ClocklessController<DATA_PIN, 3 * FMUL, 4 * FMUL, 3 * FMUL, RGB_ORDER>`
And in second declaration it extends
`ClocklessController<DATA_PIN, C_NS(300), C_NS(600), C_NS(300), RGB_ORDER>`

It's clear that time ratios are different. And according to SM16703 datasheet the second one is correct. So probably controller in first declaration should extend 
`ClocklessController<DATA_PIN, 3 * FMUL, 6 * FMUL, 3 * FMUL, RGB_ORDER>`

This modification instantly fixed color issues in my case

andig, 2022-01-12:
A PR could help speed up fixing the code

zackees, 2026-04-19:
The duplicate-declaration issue is resolved — SM16703 timing is now centralized in [`src/fl/chipsets/led_timing.h:319`](https://github.com/FastLED/FastLED/blob/master/src/fl/chipsets/led_timing.h#L319) as `TIMING_SM16703` with a single source of truth: T1=300, T2=600, T3=300 (T0H=300 ns, T0L=900 ns, T1H=900 ns, T1L=300 ns, ~833 kHz).

If the canonical 300/600/300 still gives wrong colors on real hardware, that would be a datasheet-vs-chip-reality question — please file a follow-up with a scope capture so we can adjust. Closing this particular "two declarations disagree" bug as fixed by the centralization; happy to reopen if there's still a mismatch.